keirf/greaseweazle #235: Errno 2

Andy111S, 2022-11-04:
Hi,

Probably a real NOOB issue here, whenever I attempt a read, I get:

** FATAL ERROR:
[Errno 2] No such file or directory: 'C:\\Users\\andyl\\Documents\\Computers\\Disk Images\\imation.ima'

Directory is good, file is what should be created, so does not exist yet.

Don't know why its lists double back slashes - they were not in the provided path.

Happens directly via CMD or FluxMyFluffyFloppy (FMFF referenced as background info only).

Command being attempted is:

gw read --device=COM3 --drive=A --format ibm.1440 --tracks=c=0-83 "C:\Users\andyl\Documents\Computers\Disk Images\imation.ima"

(FMFF generates "C:\Users\andyl\Documents\Computers\FluxMyFluffyFloppy_3.09_(win32bit_win64) (1)\greaseweazle\gw.exe" read --device=COM3 --drive=A --format ibm.1440 --tracks=c=0-83 "C:\Users\andyl\Documents\Computers\Disk Images\imation.ima")

GW version is Windows 1.3
FMFF version is 3.09

Regards,
Andy

keirf, 2022-11-04:
Add `--bt` to the command line and paste the error backtrace here (ie. command line starts `gw --bt read --device=COM3`)

Andy111S, 2022-11-04:
This is odd. When I paste the below into this post, the "\" are doubled - i.e. "\\", and this is what it output in the CMD GW error:

** FATAL ERROR:
[Errno 2] No such file or directory: 'C:\\Users\\andyl\\Documents\\Computers\\Disk Images\\imation.ima'

But, in my post above the double back slashes have been replaced by single ones - I didnt make this change!

I am posting this again to see if it posts correctly...

keirf, 2022-11-04:
If you wish to paste verbatim, place the multi-line paste between triple backticks (' ''' ').

Andy111S, 2022-11-04:
You may have figured that i am a little new to this...
```
** FATAL ERROR:
[Errno 2] No such file or directory: 'C:\\Users\\andyl\\Documents\\Computers\\Disk Images\\imation.ima'
```

keirf, 2022-11-04:
So gw has doubled up your slashes and then compains the directory doesn't exist?

keirf, 2022-11-04:
Are you running the v1.3 Windows binary release downloaded from the greaseweazle wiki?

Andy111S, 2022-11-04:
> So gw has doubled up your slashes and then complains the directory doesn't exist?

Basically seems to have...

> Are you running the v1.3 Windows binary release downloaded from the greaseweazle wiki?

Yes...

Please bear in mind that I am new to this, so may be making a school boy error...

Going to do the --bt now...

Thanks!

Andy111S, 2022-11-04:
The back trace:
```
C:\Users\andyl\Documents\Computers\FluxMyFluffyFloppy_3.09_(win32bit_win64) (1)\greaseweazle>gw --bt read --device=COM3 --drive=A --format ibm.1440 --tracks=c=0-83 "C:\Users\andyl\Documents\Computers\Disk Images\imation.ima"
Reading c=0-83:h=0-1 revs=2
Format ibm.1440
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 120, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\Console.py", line 16, in run
  File "gw.py", line 3, in <module>
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\cli.py", line 69, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\tools\read.py", line 234, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\image\image.py", line 20, in __enter__
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\andyl\\Documents\\Computers\\Disk Images\\imation.ima'

C:\Users\andyl\Documents\Computers\FluxMyFluffyFloppy_3.09_(win32bit_win64) (1)\greaseweazle>
```

keirf, 2022-11-04:
The double backslash thing seems to be a red herring. I get that in my error reports too, on Windows. Possibly Windows escapes the backslashes when passing the path to gw, but I'm not sure. Anyway, it is harmless and not the cause of your problem.

Your problem really has to be that the path is incorrect. That is the only way you will get `FileNotFoundError` when creating a new file. So you need to check that the `Disk Images` folder really does exist where you think it does. Because it clearly does not.

Andy111S, 2022-11-04:
Thanks for you assistance with this...

Copied from my Explorer after having browsed to the folder in question:
```
C:\Users\andyl\Documents\Computers\Disk Images
```
Now, I am pretty dyslexic, but the folder is definitely at that path...

keirf, 2022-11-04:
Start off just trying to create an image in the current folder, no path specified at all.

Andy111S, 2022-11-04:
I have tried that:
```
C:\Users\andyl\Documents\Computers\FluxMyFluffyFloppy_3.09_(win32bit_win64) (1)\greaseweazle>gw --bt read --device=COM3 --drive=A --format ibm.1440 --tracks=c=0-83 "imation.ima"
Reading c=0-83:h=0-1 revs=2
Format ibm.1440
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 120, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\Console.py", line 16, in run
  File "gw.py", line 3, in <module>
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\cli.py", line 69, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\tools\read.py", line 234, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\image\image.py", line 20, in __enter__
FileNotFoundError: [Errno 2] No such file or directory: 'imation.ima'
```

keirf, 2022-11-04:
What Windows version is this? And this is just a normal CMD prompt?

Andy111S, 2022-11-04:
Could it be a Reputation Based Protection thing preventing the app from creating a file?

I'll turn it off and try again...

Andy111S, 2022-11-04:
Sorry, its Windows 10, and its standard CMD prompt.

Andy111S, 2022-11-04:
> Could it be a Reputation Based Protection thing preventing the app from creating a file?
> 
> I'll turn it off and try again...

No difference...

keirf, 2022-11-04:
Duplicating your command line as close as I can. Seems to work:

```
C:\Users\keirf\Downloads\greaseweazle-1.3-win\greaseweazle-1.3>gw --bt read --device=COM6 --drive=A --format ibm.1440 --tracks=c=0-83 imation.ima
Reading c=0:h=0 revs=2
Format ibm.1440
T0.0: IBM MFM (18/18 sectors) from Raw Flux (155702 flux in 400.50ms)
...
```

keirf, 2022-11-04:
What if you create the `imation.ima` file by some means (eg. copy another file to that name) so that it exists, and then run your gw command?

Andy111S, 2022-11-04:
Worth a go, give me 10 minutes...

keirf, 2022-11-04:
Currently I'm bamboozled. Lots (most) users use the Windows binary release, with or without a GUI wrapper, and this is the first time I've heard of this error.

andrewbird, 2022-11-04:
> C:\Users\andyl\Documents\Computers\FluxMyFluffyFloppy_3.09_(win32bit_win64) (1)\greaseweazle>

I wonder if this funky current directory (brackets etc) is upsetting things?

Andy111S, 2022-11-04:
> > C:\Users\andyl\Documents\Computers\FluxMyFluffyFloppy_3.09_(win32bit_win64) (1)\greaseweazle>
> 
> I wonder if this funky current directory (brackets etc) is upsetting things?

Fair point, ill de-funk it...

keirf, 2022-11-04:
You could always copy the directory somewhere else and run it there. But that doesn't really make a lot of sense: in the current command line you're not specifying any path at all.

Perhaps download and unzip v1.3 on a completely fresh simpler path?

Do you have any other Windows system you can try on?

Andy111S, 2022-11-04:
> 

I'm going to try a few things:
De-funking the path broke some things in FMFF so I will re-install both at a simpler path.
If that does not get me any further, I will try my old faithful XP machine - I do miss XP...

Andy111S, 2022-11-04:
OK, so its resolved.
It was a combination of Windows 10 Defender having an issue with gw.exe being untrusted, then the Ransomware function preventing untrusted apps from having access to 'protected folders', and the Protected Folders function considering user document folders as protected...
All working now - thanks for your patience and input. Off to mess with floppies now...

keirf, 2022-11-04:
Well worked out! It isn't helpful that a permissions error gets turned into file not found. The reasoning is probably that otherwise information is leaked about whether or not a folder exists.

closed 2022-11-04 by keirf