keirf/greaseweazle #237: D88 format read support

tdaede, 2022-11-10:
D88 is a format used primarily by the PC-88, but also by many other Japanese computers with similar floppy hardware. It is similar to IMD in that it provides track layout images too. I'm adding support for D88 to gw.

Because this image format has a flexible sector layout, I am wondering what I should use for default_format. I could add a generic IBM_Format that doesn't assume anything about the sector layout, for example. Or, I could add a function to the Image class that can optionally return a custom Format which overrides default_format.

Alternately, I could manually call the mfm code right in the D88 reader and treat it as a flux format. Though, I would also like to add support for IMD, NFD, and other extremely similar formats that could reuse the same code.

tdaede, 2022-11-14:
Partially done with #246. I just reused IBM_MFM_Formatted in this case, but I'll probably need a separate class for D88 writing. The verify_track and get_img_track functions work as-is.

keirf, 2022-11-15:
That said the EDSK format is much abused to fit in all sorts of protected tracks, so it's a pain to handle. D88 could be much cleaner.

keirf, 2023-02-01:
I strongly suggest D88 gets rewritten along the lines of the new IMD support. The idea is that you define each track's layout in an `IBMTrackFormat` object, and then construct a track object (`mk_track()`) and stuff data into the resulting track which has been nicely laid out for you with appropriate gaps. This means we are defining gap heuristics in just one place and the image format handlers can concentrate on just handling image parsing rather than doing low-level track layout.

keirf, 2023-02-03:
I have taken a swing at this as part of reworking the IBM track handling and adding static type hints.

The code is already simpler! But it may not work. I have no D88 image files.

keirf, 2023-02-03:
I found one. Code is now fixed. Seems to work!

closed 2023-02-03 by keirf