keirf/greaseweazle #242: Writing IBM floppies takes twice as long with gw as with fluxengine

tdaede, 2022-11-11:
For pc98.dd format:

fluxengine: 1m18s
gw: 2m39s

tdaede, 2022-11-14:
It seems that one reason for this is that gw does index-aligned reads during verify, as the verify checks the TrackArea start and end times.

keirf, 2022-11-15:
I should implement a quick version of the decoder which can take unaligned data and simply scan it for sectors not caring about positions. This would be fine for stuffing an `IBM_MFM_Formatted`. It would also make `gw read` of these IBM formats faster too.

Still this would probably be 3 revolutions per track on verified write: Aligned write, unaligned read of just over a track. Then next aligned write waits most of a revolution to start. That would be 600ms * 160 = 1m36s best case. Although a bit better than that if pc98-dd is less than 160 tracks or you are spinning at 360rpm.

It's difficult to see how you do better than 3 revolutions per track... One trick would be to write both sides and read back both sides all in one command: Write side 0, Write side 1, Read side 0, Read side 1. Then waste a revolution stepping to next cylinder. That would be 5 revolutions per cylinder. Or 1s * 80 = 1m20s.

I don't think fluxengine does anything as crazy as that. Is it verifying at all, in your test?

keirf, 2023-04-24:
Any further updates before I close this?

closed 2023-05-11 by keirf