keirf/greaseweazle #247: Track overruns should error

tdaede, 2022-11-14:
The current pre-index padding code ultimately results in skewing the clock rate if the sector data does not fit in the track length. It should instead exit with an error.

keirf, 2022-11-14:
Which pre-index padding code?

tdaede, 2022-11-14:
mfm.py:246

```
        # Add the pre-index gap.
        tlen = int((self.time_per_rev / self.clock) // 16)
        gap = max(tlen - len(t)//2, 0)
        t += encode(bytes([self.gapbyte] * gap))
```

The max() function results in it not padding if the track is too long, which then potentially results in the RPM correction code doing the wrong thing.

keirf, 2022-11-15:
Yes, so a long track squishes the bits to fit. This can happen. Perhaps not in current uses of `IBM_MFM` and `IBM_FM` though.

keirf, 2023-02-01:
Perhaps there should be an error or warning for gross misfits (eg clock gets squished by 20% or more)

tdaede, 2023-02-01:
The current easiest way to trigger it is with D88 files that either need a smaller gap3, or were dumped from floppies with overlapping sectors (D88 can't represent the overlap, but it works on emulators that don't care about track length).

keirf, 2023-02-01:
If D88 changes to use the IBMTrackFormat logic (same as IMD) then at least the disks with short gap3 will work. Overlapping sectors likely have a massively over large track and we could definitely warn on that.

commit 1de102bb (2023-05-11): ibm: Warn on oversized track layouts
Fixes #247