keirf/greaseweazle #260: macOS install fails

jluke1, 2022-12-12:
macOS install fails with error
"       File "<string>", line 2, in <module>
      ModuleNotFoundError: No module named 'setuptools_scm'"

macOS 10.15.7
Python 3.10.8
pip 22.3.1 from /usr/local/lib/python3.10/site-packages/pip (python 3.10)
brew update ; brew upgrade has been run.

keirf, 2022-12-12:
I've heard of occasional problems on Mac before, in just this way. I would expect the setuptools_scm dependency to be dealt with automatically but it seems not always to be. Try
`python -m pip install setuptools_scm`

jluke1, 2022-12-12:
Thanks that fixed it. Fast response.

Thanks for all the excellent work.

Regards
John Luke

keirf, 2022-12-12:
Thanks. I have updated the wiki to include this additional step. It should be harmless where it's not needed (already installed).

closed 2022-12-12 by keirf