keirf/greaseweazle #261: Support for ZX spectrum (clone) CP/M disk format or just how to make custom format

kaunomedis, 2022-12-14:
I want to archive some 5.25" disks from ZX spectrum Soviet era clone (CP/M ir TRDOS). It is DD disk (40 tracks), double sided and 5 sectors, 1024 bytes per sector. ISO MFM format.
I can read flux but I can not convert it to binary image. I tested all available formats and akai.800 can read track 0. So it is possible.

And I want to write back some disks.

Maybe it is possible to add some option for "custom formats"? As I am planing to read some other ancient CP/M computers (like compucorp).

keirf, 2022-12-16:
Yes custom formats will be added. Other options include using third party tooling such as RIDE (https://github.com/tomas-nestorovic/RIDE) or dump to SCP using greaseweazle and then convert using https://github.com/keirf/disk-utilities or HxC software.

kaunomedis, 2022-12-19:
RIDE looks very promising, but it is not easy to compile (my visual studio is too new for the source) , will search for binaries somewhere.

Disk utilities may help a bit, but somehow ZX emulators do not like the result.

keirf, 2022-12-19:
Then try hxc tools. Load the raw image and export as IMG or whatever.

keirf, 2023-01-02:
There is now an alpha test build which allows configuration of new disk formats. Here is the tools download link:
https://github.com/keirf/greaseweazle/suites/10143755085/artifacts/495424734

And here is a link to the default config file, which explains the layout:
https://github.com/keirf/greaseweazle/blob/master/src/greaseweazle/data/diskdefs.cfg

You override the file using the new `--diskdefs` command-line option.

Let me know how it goes!

kaunomedis, 2023-01-02:
Nice feature. As for my disks, I tested it with 40 track drive and had only track 0 data. All other tracks were unreadable. Then I decided that my disk drive is bad and switched to 80 tracks drive. With "step=2" there was garbage. Then I decided to test disk with "step=1" and got lots of data... Maybe the disk is HD? (Label on the disk is "Doule side, Double density, 360kb"... They formated it to HD?

"working" config lines:
disk zx.cpm.800
    cyls = 80
    heads = 2
    tracks *:ibm.mfm
        secs = 5
        bps = 1024
        rate = 250
		interleave = 2
    end
end

Will check some other disks and share ideas. Now I am comparing disk images "by hand". At last, I have "retry" on bad sectors. I can "grind" the FD to salvage data.

...
T79.0: Giving up: 1 sectors missing
T79.1: IBM MFM (5/5 sectors) from Raw Flux (75783 flux in 332.99ms)
Cyl-> 0         1         2         3         4         5         6         7
H. S: 01234567890123456789012345678901234567890123456789012345678901234567890123456789
0. 0: .X.....X..........................X............X................................
0. 1: .X.....X.............X.............................................X............
0. 2: XX.X................X........X.........X..................X.....................
0. 3: .X................................................X........X....................
0. 4: .X..........................................X.................X...............XX
1. 0: .X..............................................................................
1. 1: .X..............................................................................
1. 2: .X..............................................................................
1. 3: .X..............................................................................
1. 4: .X..............................................................................
Found 772 sectors of 800 (96%)

keirf, 2023-01-02:
It's hardly tested so there could also be bugs. Can you do a raw dump of a disk, perhaps like this:
`gw read a.scp`
And zip and attach the resulting image file? I can take a look and see if it's got good data and if so I can recommend a diskdef stanza.

kaunomedis, 2023-01-02:
Looks like it was DD/DD disks formated as DS/HD :(

Here is same, "system" disk.
[U00.zip](https://github.com/keirf/greaseweazle/files/10333027/U00.zip)

I am very happy, "disk grinding" is recovering much more data!
T70.1: IBM MFM (5/5 sectors) from Raw Flux (75240 flux in 332.99ms)
T71.0: IBM MFM (4/5 sectors) from Raw Flux (75265 flux in 332.98ms)
T71.0: IBM MFM (4/5 sectors) from Raw Flux (188106 flux in 832.21ms) (Retry #1.1)
T71.0: IBM MFM (4/5 sectors) from Raw Flux (300947 flux in 1331.43ms) (Retry #1.2)
T71.0: IBM MFM (4/5 sectors) from Raw Flux (413788 flux in 1830.66ms) (Retry #1.3)
T71.0: IBM MFM (5/5 sectors) from Raw Flux (526630 flux in 2329.90ms) (Retry #2.1)
T71.1: IBM MFM (5/5 sectors) from Raw Flux (75240 flux in 332.98ms)
T72.0: IBM MFM (5/5 sectors) from Raw Flux (75267 flux in 332.99ms)
T72.1: IBM MFM (5/5 sectors) from Raw Flux (75256 flux in 332.99ms)
T73.0: IBM MFM (5/5 sectors) from Raw Flux (75256 flux in 332.99ms)
...

keirf, 2023-01-02:
Oh okay, yeah I see. I glanced at your initial response and thought things were worse than they actually are. By the way, if you have DD on HD disks then it can improve the read-out if you cover the HD hole.

Anyway seems you are on the right track and the new configuration support is working well. Glad to hear you're merging IMG files okay -- even though gw doesn't do it for you, it's pretty easy to script up for sector image files. Perhaps I should put a pattern for missing sectors other than 00 bytes? Eg on AmigaDOS ADF files I repeat `-=[BAD SECTOR]=-` throughout the sector.

kaunomedis, 2023-01-02:
I don't know how bad sector is currently marked. And how it is detected. Sometimes (especially for text files) the data is recoverable in the sector regardless some garbage.
I recovered first ZX program! It is very very first smart electrical meter design ideas! That later become quite a big company!
Still don't know the disk format, but managed to recover software. I am happy, thanks.

keirf, 2023-01-02:
Great to hear this is working well!

closed 2023-01-02 by keirf

kaunomedis, 2023-01-14:
CP/M for ZX spectrum disk image reading confirmed using SteinBlume CP/M disk browser. Files, both binary and text were extracted. Binary files tested on "generic" CP/M emulator.

bbbradsmith, 2023-09-06:
Just wanted to leave a thumbs up for this. I couldn't find information about custom disk definitions on the wiki, but some digging around I found this thread and copying data/diskdefs.cfg file as an example to start from it worked great. Very helpful for Atari ST images in various permutations of tracks/sectors.