keirf/greaseweazle #265: Custom format enabled, TR-DOS 5.25 disk can read only one side

kaunomedis, 2023-01-05:
Testing new beta feature- custom format. Can not force to read some hacked ZX Spectrum TR-DOS disk. It is 16 sectors, 256 bytes/sector format. Somehow I can read only one side. Looks like there is data on other side too.

disk zx.trdos.640
    cyls = 80
    heads = 2
    tracks *:ibm.mfm
        secs = 16
		bps = 256
    end
end
(other params are unknown for me)

Please check attached flux image.
[TRDOS_D35.zip](https://github.com/keirf/greaseweazle/files/10354600/TRDOS_D35.zip)

keirf, 2023-01-05:
Looks like the head (H) field of sectors is 0 on both sides?

I assume this is two single-sided images. Perhaps someone used a side-select switch to manually select side 1?

You should be able to dump as follows:
1. Change to `heads = 1` in your config.
2. Dump Side A
3. Dump Side B by adding `--tracks=hswap` to your command line.

kaunomedis, 2023-01-05:
I created TR-DOS disk in emulator. Both "sides" have id=0...
[TRDOS.zip](https://github.com/keirf/greaseweazle/files/10355141/TRDOS.zip)

keirf, 2023-01-05:
If it is actually a two-sided disk image with both sides having head fields with H=0, then add `h=0` inside your `tracks` definition block.

You can define a disk with multiple different track formats by declaring multiple `tracks` blocks within a `disk`. You can specify exactly which tracks each block applies to by changing the `*` to an explicit track list. For example `1-5.0` would be cylinders 1-5, head 0. If you get stuck send me a disk and I will make a disk definition for you as an example.

keirf, 2023-01-05:
Then you want
```
disk zx.trdos.640
cyls = 80
heads = 2
tracks *:ibm.mfm
secs = 16
bps = 256
h = 0
end
end
```

kaunomedis, 2023-01-05:
Yes! It works. At least it is reading. As TR-DOS is "normal" OS for ZX spectrum, I hope it will work with fuse emulator (this emulator has more options compared to others). Will report the result.

keirf, 2023-01-05:
It is interesting if TR-DOS always marks both sides as 0. For example, FlashFloppy TRD image handler does not do this. I wonder how well tested FlashFloppy is with TRD images? Do you know of anyone using FF with TRD images?

closed 2023-01-06 by keirf

kaunomedis, 2023-01-14:
TR-DOS reading confirmed using FUSE emulator. It can boot/run games from disk image if BETA disk controller is enabled in 80 track mode.

GW can not write directly to .trd format, so the only way is to use HxC to convert from .hfe to .trd or .dsk

commit 4d833010 (2023-01-06): diskdefs: Add zx.trdos.640 definition

Fixes #265
commit 4d8519df (2023-01-09): diskdefs: Add zx.trdos.640 definition

Fixes #265