keirf/greaseweazle #266: Correct settings for Atari 8-bit 32K 5.25 floppy?

bleggett, 2023-01-05:
I'm trying to read an Atari 8-bit 5.25 floppy with a generic PC 5.25 drive I have successfully used with GW in the past.

The floppy says it's a "32K disk" on the label  - trying to read it to `.scp` format with this CLI `gw.exe read --format=atari.90 --raw "C:\Users\xxx\Desktop\derp.scp"` 

gives me this:

```
Reading c=0-39:h=0:step=2 revs=2
Format atari.90
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: IBM FM (17/18 sectors) from Raw Flux (79336 flux in 400.60ms)
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: IBM FM (17/18 sectors) from Raw Flux (198293 flux in 1001.20ms) (Retry #1.1)
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: IBM FM (17/18 sectors) from Raw Flux (317250 flux in 1601.80ms) (Retry #1.2)
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: IBM FM (17/18 sectors) from Raw Flux (436207 flux in 2202.40ms) (Retry #1.3)
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: IBM FM (17/18 sectors) from Raw Flux (555163 flux in 2803.00ms) (Retry #2.1)
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
T0.0: IBM FM (17/18 sectors) from Raw Flux (674119 flux in 3403.60ms) (Retry #2.2)
T0.0: Ignoring unexpected sector C:0 H:0 R:18 N:0
..snip
```

keirf, 2023-01-05:
The format specifier is wrong in Greaseweazle. The sector numbering is 1..18, not 0..17. Hence sector 0 is not found, and a "spurious" sector 18 is found instead!

keirf, 2023-01-05:
With latest Greaseweazle tools (v1.5), download the attached config and run as
```gw.exe read --diskdefs=atari.90.cfg --format=atari.90 --raw "C:\Users\xxx\Desktop\derp.scp"```

Let me know how it goes!

[atari.90.cfg.zip](https://github.com/keirf/greaseweazle/files/10356272/atari.90.cfg.zip)

bleggett, 2023-01-06:
@keirf Nice! That worked (I also had to add `--tracks=step=2`). Thank you!

commit 2b56d29b (2023-01-06): atari.90: Sector IDs are 1..18 not 0..17.

Leave the gap definitions as is. These quite neatly centre the
track gap across the index.

Fixes #266
closed 2023-01-06 by keirf