keirf/greaseweazle #267: Another weird Atari 8-bit 5.25 read issue

bleggett, 2023-01-09:
I was able to dump some Atari 8-bit floppies using the fix outlined in #266 - but I have some others that give the following with the same drive/same settings, but a different 8-bit 5.25 disk?

I don't think this is a media error.

```
C:\Users\algos\Downloads\greaseweazle-1.5-win\greaseweazle-1.5>gw.exe read --diskdefs=atari.90.cfg --format=atari.90 --tracks=step=2 --raw "C:\Users\algos\Desktop\derp.scp"
Reading c=0-39:h=0:step=2 revs=2
Format atari.90
T0.0: IBM FM (18/18 sectors) from Raw Flux (88872 flux in 333.86ms)
T1.0: IBM FM (18/18 sectors) from Raw Flux (87228 flux in 333.86ms)
T2.0: IBM FM (18/18 sectors) from Raw Flux (86152 flux in 333.85ms)
T3.0: IBM FM (18/18 sectors) from Raw Flux (84594 flux in 333.84ms)
T4.0: IBM FM (0/18 sectors) from Raw Flux (92545 flux in 333.84ms)
T4.0: IBM FM (0/18 sectors) from Raw Flux (231974 flux in 834.36ms) (Retry #1.1)
T4.0: IBM FM (0/18 sectors) from Raw Flux (371028 flux in 1334.90ms) (Retry #1.2)
T4.0: IBM FM (0/18 sectors) from Raw Flux (510603 flux in 1835.41ms) (Retry #1.3)
T4.0: IBM FM (0/18 sectors) from Raw Flux (649913 flux in 2335.91ms) (Retry #2.1)
T4.0: IBM FM (0/18 sectors) from Raw Flux (789546 flux in 2836.43ms) (Retry #2.2)
T4.0: IBM FM (0/18 sectors) from Raw Flux (929194 flux in 3336.95ms) (Retry #2.3)
T4.0: IBM FM (0/18 sectors) from Raw Flux (1068080 flux in 3837.46ms) (Retry #3.1)
T4.0: IBM FM (0/18 sectors) from Raw Flux (1207734 flux in 4337.97ms) (Retry #3.2)
T4.0: IBM FM (0/18 sectors) from Raw Flux (1347597 flux in 4838.48ms) (Retry #3.3)
T4.0: Giving up: 18 sectors missing
T5.0: IBM FM (0/18 sectors) from Raw Flux (93520 flux in 333.83ms)
T5.0: IBM FM (0/18 sectors) from Raw Flux (234012 flux in 834.36ms) (Retry #1.1)
T5.0: IBM FM (0/18 sectors) from Raw Flux (374006 flux in 1334.86ms) (Retry #1.2)
T5.0: IBM FM (0/18 sectors) from Raw Flux (514763 flux in 1835.37ms) (Retry #1.3)
T5.0: IBM FM (0/18 sectors) from Raw Flux (655251 flux in 2335.87ms) (Retry #2.1)
T5.0: IBM FM (0/18 sectors) from Raw Flux (796091 flux in 2836.38ms) (Retry #2.2)
T5.0: IBM FM (0/18 sectors) from Raw Flux (936500 flux in 3336.87ms) (Retry #2.3)
T5.0: IBM FM (0/18 sectors) from Raw Flux (1076932 flux in 3837.38ms) (Retry #3.1)
T5.0: IBM FM (0/18 sectors) from Raw Flux (1217424 flux in 4337.89ms) (Retry #3.2)
T5.0: IBM FM (0/18 sectors) from Raw Flux (1358420 flux in 4838.41ms) (Retry #3.3)
T5.0: Giving up: 18 sectors missing
T6.0: IBM FM (0/18 sectors) from Raw Flux (93752 flux in 333.83ms)
T6.0: IBM FM (0/18 sectors) from Raw Flux (234428 flux in 834.33ms) (Retry #1.1)
T6.0: IBM FM (0/18 sectors) from Raw Flux (375105 flux in 1334.83ms) (Retry #1.2)
T6.0: IBM FM (0/18 sectors) from Raw Flux (516360 flux in 1835.34ms) (Retry #1.3)
T6.0: IBM FM (0/18 sectors) from Raw Flux (657218 flux in 2335.83ms) (Retry #2.1)
T6.0: IBM FM (0/18 sectors) from Raw Flux (797697 flux in 2836.34ms) (Retry #2.2)
T6.0: IBM FM (0/18 sectors) from Raw Flux (938290 flux in 3336.86ms) (Retry #2.3)
T6.0: IBM FM (0/18 sectors) from Raw Flux (1079354 flux in 3837.36ms) (Retry #3.1)
T6.0: IBM FM (0/18 sectors) from Raw Flux (1220033 flux in 4337.86ms) (Retry #3.2)
T6.0: IBM FM (0/18 sectors) from Raw Flux (1360721 flux in 4838.38ms) (Retry #3.3)
T6.0: Giving up: 18 sectors missing
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: IBM FM (0/18 sectors) from Raw Flux (77896 flux in 333.84ms)
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: IBM FM (0/18 sectors) from Raw Flux (194676 flux in 834.33ms) (Retry #1.1)
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: IBM FM (0/18 sectors) from Raw Flux (311456 flux in 1334.83ms) (Retry #1.2)
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: IBM FM (0/18 sectors) from Raw Flux (428236 flux in 1835.34ms) (Retry #1.3)
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: IBM FM (0/18 sectors) from Raw Flux (545016 flux in 2335.83ms) (Retry #2.1)
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: Ignoring unexpected sector C:7 H:0 R:255 N:0
T7.0: IBM FM (0/18 sectors) from Raw Flux (661795 flux in 2836.33ms) (Retry #2.2)
```

keirf, 2023-01-09:
Perhaps they aren't atari dos format. Copy protection? Take a raw dump and view it using hxc software.

bleggett, 2023-01-10:
All right - 2 of the 3, when dumped to a raw `.scp`, work just fine in the Altirra emulator when converted to ATX format via `a8rawconv`.

3rd still doesn't, but it appears that the errors are intentional, I guess?

closed 2023-01-10 by bleggett