keirf/greaseweazle #268: Blue Pill Direct - Not write

AidenShaw2020, 2023-01-11:
Hi all, i have issue with Blue Pill "Direct", when i try to write img to physical media:
[image]

Reading working OK:
[image]
[image]

Wirings is ok (without shortages), and i also tryied with dupont cables, and six other FDD's, but i have same experience.
[image]
[image]
[image]

I use FW 1.2 and GW 1.4 (1.5 and Flux Engine have same issues).
Can you help me? Thanks.

keirf, 2023-01-11:
The only extra lines that really matter for writing are WGATE (floppy pin 24, mcu pin B14) and WDATA (floppy pin 22, mcu pin B4). So make extra sure those are okay. Even better if you can put a scope or logic analyser on them. Also are you sure the disk you are testing is good and is high density?

AidenShaw2020, 2023-01-11:
Hi @keirf, wirings for both pins Is OK, but thank you, for tapping me with logic analyser.
When I check both pins at same time, I have TTL only for B14, not B4:
[image]

For sure, I tried STM32 in Arduino IDE, and really, pin PB4 is not working.
I changed pin B4 to pin B5 in src/mcu/stm32f1/floppy.c:

`
/* WDATA: Pin B5, Timer 3 Channel 1, DMA1 Channel 3. */
#define gpio_wdata  gpiob
#define pin_wdata   5
#define tim_wdata   (tim3)
#define dma_wdata   (dma1->ch3)
`
, and now I have TTL on both pins, but writing still not working.
[image]

Do I need edit pin somewhere else?

Thanks.

keirf, 2023-01-11:
You can't just change pin. The chosen timer peripheral is only on the original pin.

First port of call would be the blinky test firmware to confirm that your blue pill is not a bad clone.

keirf, 2023-01-11:
https://github.com/keirf/Greaseweazle/wiki/STM32-Fakes

And if you take the blinky firmware from an old Greaseweazle release that's better as the serial baud rate is incorrect in the version in most recent Greaseweazle releases. This version would do: https://github.com/keirf/greaseweazle/releases/download/v0.20/Greaseweazle-v0.20.zip

AidenShaw2020, 2023-01-11:
I test blinky as first, and passed it. This version (0.20), works too.
https://user-images.githubusercontent.com/62829872/211920431-ee0a5c4b-0449-4d35-9152-6af17e096083.MOV

I also tried Bluepill Diagnostics, with pass:
[image]

keirf, 2023-01-11:
Guess you need a new mcu or blue pill all the same

AidenShaw2020, 2023-01-11:
I was afraid of that. I was glad to get a "working" one :(
Thanks for your support.

closed 2023-01-11 by AidenShaw2020

AidenShaw2020, 2023-02-16:
Hi @keirf, after chip replacement, all is ok :)
For all thats have same problem as i, this is seller, where i bought replacements chips: https://a.aliexpress.com/_msFQ1sa

[image]
[image]