keirf/greaseweazle #270: Multiple format on same disk - gw ignores custom format (BETA)

kaunomedis, 2023-01-14:
I have two strange disks for homebrew ZX spectrum. Looks like one format overwritten on another. Disk is readable, but I can not define tracks. Here is just testing format description:
```
disk zx.xaos.000
    cyls = 15
    heads = 2
	step =1
    tracks 0.1,1,3,5,7,9:ibm.mfm
        secs = 5
        bps = 1024
        rate = 250
		interleave = 2
    end
	tracks 0.0,2,4,6,8:ibm.mfm
        secs = 9
        bps = 512
        rate = 250
    end
end
```
Data in the disk:
0.0 - 9 sectors
0.1 - 5 sectors
1,3,5... (even) - 5 sectors
2,4,6... (odd) - 9 sectors.

With configuration above, gw reads. But it reads only 0.0, 0.1, 1.0, 1.1, 3.0, 3.1...
It ignores 2,4 and etc. track format.

T0.0: IBM MFM (9/9 sectors) from Raw Flux (94401 flux in 332.99ms)
HFE: Data bitrate detected: 250 kbit/s
T0.1: IBM MFM (5/5 sectors) from Raw Flux (80005 flux in 332.99ms)
T1.0: IBM MFM (5/5 sectors) from Raw Flux (76961 flux in 333.00ms)
T1.1: IBM MFM (5/5 sectors) from Raw Flux (75578 flux in 333.00ms)
T2.0: Ignoring unexpected sector C:1 H:0 R:9 N:2
...
T2.0: IBM MFM (0/9 sectors) from Raw Flux (79653 flux in 333.00ms)
...
T2.0: Giving up: 9 sectors missing
T2.1: Ignoring unexpected sector C:1 H:1 R:1 N:2
T2.1: Ignoring unexpected sector C:1 H:1 R:2 N:2
...
T2.1: Ignoring unexpected sector C:1 H:1 R:9 N:2
T2.1: IBM MFM (0/9 sectors) from Raw Flux (1140312 flux in 4826.30ms) (Retry #3.3)
T2.1: Giving up: 9 sectors missing
T3.0: IBM MFM (5/5 sectors) from Raw Flux (79682 flux in 333.00ms)
T3.1: IBM MFM (5/5 sectors) from Raw Flux (78406 flux in 333.01ms)
T4.0: Ignoring unexpected sector C:2 H:0 R:4 N:2
T4.0: Ignoring unexpected sector C:2 H:0 R:2 N:2
...

it is oddity, but maybe it is interesting.

Disk image:
[XAOS1.zip](https://github.com/keirf/greaseweazle/files/10418777/XAOS1.zip)

keirf, 2023-01-15:
The cylinder number (C) is half what it should be on the even cylinders. Probably this is a different disk image written on a 40-track drive, or with double stepping?

You might want to define that as a separate disk image type and dump it with (`--tracks=step=2`)? Or perhaps it is a copy-protected disk, or something?

kaunomedis, 2023-01-15:
I tested with step setting, but somehow I can read only one half of the disk. The `h0.off=+1` options do not work? Or it steps "2" steps when `--tracks=step=2` enabled?

I also wondering about customs settings (BETA), why I can read only track 0, ant other (9 sector) tracks are not readable. If I have spare time, I will play with it. Are the 9 sectors tracks in higher tracks unreadable or just GW ignores the settings?

keirf, 2023-01-15:
GW expects the cylinder number (C) in the sector header to match the actual cylinder number, else it won't accept the sector. There is no way to override this currently.

`--tracks=step=2` will step twice per cylinder. So when reading cylinder `N`, the heads will positioned at `2*N` on your drive.

keirf, 2023-01-15:
We could allow cylinder override but:
1. You would probably need a track config per track, hardcoding the cylinder number for each track
2. It's not clear this is even a single coherent disk image that you are creating here? It could be two disk images, or a ghost of an old disk image, subsequently overwritten only partially by a new image, or...?

keirf, 2023-01-15:
Also `h0.off` is unlikely what you want. It was introduced mainly for allowing reading of C64 flippy disks using a modified double-sided drive.

kaunomedis, 2023-01-15:
I will try to make some investigation about it. Is it "ghost" or protection. Person who give me that disks is not very informative- he do not remember...

I am interested if disk config file format for tracks can be written:
```

disk zx.xaos.000
    cyls = 10
    heads = 2
	step =1
    tracks 0.1,1,3,5,7,9:ibm.mfm
        secs = 5
        bps = 1024
        rate = 250
	interleave = 2
    end
    tracks 0.0,2,4,6,8:ibm.mfm
        secs = 9
        bps = 512
        rate = 250
    end
end
```
Is it valid config? The 10 cyls/tracks is only for faster testing. In real it is 80.

keirf, 2023-01-15:
A later `tracks` config matching on a particular track will override the config for that track.

Your config is valid. 

By the way, this new configuration format is in the latest greaseweazle release. There was a small syntax change: there is now no colon in the `tracks` specifier immediately before the track format name (eg. `ibm.mfm`)

kaunomedis, 2023-01-15:
Something is wrong. It can not read 9 sectors from that "magic" disk.
I made config only consisting 512/9 per cyl.

Only 0.0 is read. 

T0.0: IBM MFM (9/9 sectors) from Raw Flux (94401 flux in 332.98ms)
HFE: Data bitrate detected: 250 kbit/s
T0.1: Ignoring unexpected sector C:0 H:1 R:5 N:3 - it is 5 sectors. OK
T1.0: Ignoring unexpected sector C:1 H:0 R:3 N:3 - it is 5 sectors. OK
T1.1: Ignoring unexpected sector C:1 H:1 R:1 N:3 - it is 5 sectors. OK
T2.0: Ignoring unexpected sector C:1 H:0 R:5 N:2 - it is 9 sectors BAD!
T2.1: Ignoring unexpected sector C:1 H:1 R:6 N:2 - it is 9 sectors BAD!

I can export binary from HxC... the sectors are good. But gw can't read 9 sectors track. Meanwhile 5 sectors do not cause any problems.
Any ideas?

kaunomedis, 2023-01-15:
I tested disk with 40 track drive...

0.0 - is OK, 9 sectors read.
0.1 - ignoring unexpected.
1.0 - and to the end (40 track) all cils read with 9 sectors.

Somehow gw can not read this disks 9 sectors track in 80 track drive, but can read in 40. But in 40 there is no way to read ghosting 5 sectors track.

It is OK, maybe it is ghosting. But why gw can not read 9 sectors tracks in 80 track drive? HxC can extract...
I've met that person who gave disks for me. These disk may be formated and used on FDD from some Soviet mainframe (Bulgarian made ИЗОТ)...

I will try to analize binary data. Maybe there are tracks from mainframe and from ZX spectrum on same disk.

keirf, 2023-01-15:
You should be able to read on 80t drive with `--tracks=step=2` on the command line. This should be equivalent to a 40t drive .

My guess is disk was written in a 40t drive and the other tracks are a ghost of a previous use with an 80t drive .

kaunomedis, 2023-01-15:
Disk images have similarities and have very different data.
The only way to do this disk:
format 40 track drive, 9 sectors.
reformat also 40 track disk using 80 track drive using 5 sectors.

OR

using 80 drive format 5 sectors whole disk. Then using 80 drive format as DD - 9 sectors, 40 tracks, every second track. :)

Both versions have CP/M!

5 sectors version have ZX Spectrum game.
9 sectors version have CP/M, some documents... and last sectors have IBM PC data- "c:\vc\viewers\dbview !.!"
I think this disk was in 3 different computers.

Thanks for help. Closing this case of digital archeology.

closed 2023-01-15 by kaunomedis