keirf/greaseweazle #272: drive letter in windows?

orcutt989, 2023-01-20:
Apologies is this is a dumb question, but I feel like I've been googling and I'm not understanding.  Is there a way to mount the drives connected to the greaseweazle as drive letters in windows?

keirf, 2023-01-20:
No Greaseweazle is not visible directly as drives to Windows. You can only access it through the gw tool or other compatible utilities.

closed 2023-01-20 by keirf

orcutt989, 2023-01-20:
Thanks for responding so quickly. I assume there's a good reason this is not impleme ted.  Seems like it has everything it needs from a hardware interfacing standpoint.

Is it out of scope or something? What is the reason? Seems like it could be a feasible feature.

keirf, 2023-01-20:
A driver would need to be implemented for Windows. There are no plans to do that.

keirf, 2023-01-20:
Or the Greaseweazle would need to implement USB floppy protocol. There's no plan to do that either as USB floppy drives are readily available.

orcutt989, 2023-01-20:
Sure 3.5 inch floppies are readily available but I'm sure you're aware the vintage computing scene is exploding. 

People are acquiring 5.25" and 8" floppies for repair and hobbying at a staggering rate. It would be really nice if the best platform (greaseweazle) could support it. 

As far as I'm aware theres no USB option for this. The conveniences of this would be monumental. Having to boot a motherboard with FDD on it when tweaking these drives over and over again is nuts. 

For now I'm going to buy an [8$ FDD to us adapter on eBay](https://www.ebay.com/itm/394315950779?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=eqevvt2hs8y&sssrc=2349624&ssuid=M4O4J6kfQCC&var=&widget_ver=artemis&media=COPY),  and replace the female FDD connector with a male so it can take a combo edge connector/fdd female dual cable.

I would highly encourage ya'll discuss the possibility of this. I would love to help and I'll do some research and see if I can contribute.