keirf/greaseweazle #274: Crash when using disk definition with different number of sectors per track

hharte, 2023-01-26:
While reading some 8" floppies with Greaseweazle, I noticed that when reading CompuPro CP/M disks that have a different number of sectors for track 0 than the rest of the tracks, Greaseweazle crashes while printing the track/sector summary.

Using the following disk definition:
```
disk compupro.ssdd
    cyls = 77
    heads = 1
    tracks 0 ibm.fm
        secs = 26
        bps = 128
        rate = 250
        rpm = 360
    end
    tracks * ibm.mfm
        secs = 8
        bps = 1024
        rate = 500
        rpm = 360
    end
end
```

results in:

```
gw.exe read --format compupro.ssdd --raw compupro_ssdd_r4.scp
Reading c=0-76:h=0 revs=2
Format compupro.ssdd
T0.0: IBM FM (26/26 sectors) from Raw Flux (118386 flux in 351.82ms)
T1.0: IBM MFM (8/8 sectors) from Raw Flux (137411 flux in 345.81ms)
T2.0: IBM MFM (8/8 sectors) from Raw Flux (128417 flux in 341.94ms)
T3.0: IBM MFM (8/8 sectors) from Raw Flux (129938 flux in 339.41ms)
T4.0: IBM MFM (8/8 sectors) from Raw Flux (128459 flux in 337.71ms)
T5.0: IBM MFM (8/8 sectors) from Raw Flux (131963 flux in 336.57ms)
T6.0: IBM MFM (8/8 sectors) from Raw Flux (129083 flux in 335.78ms)
T7.0: IBM MFM (8/8 sectors) from Raw Flux (133464 flux in 335.23ms)
T8.0: IBM MFM (8/8 sectors) from Raw Flux (126072 flux in 334.82ms)
T9.0: IBM MFM (8/8 sectors) from Raw Flux (127988 flux in 334.52ms)
T10.0: IBM MFM (8/8 sectors) from Raw Flux (132520 flux in 334.29ms)
T11.0: IBM MFM (8/8 sectors) from Raw Flux (132079 flux in 334.12ms)
T12.0: IBM MFM (8/8 sectors) from Raw Flux (131752 flux in 333.99ms)
T13.0: IBM MFM (8/8 sectors) from Raw Flux (129208 flux in 333.88ms)
T14.0: IBM MFM (8/8 sectors) from Raw Flux (126055 flux in 333.81ms)
T15.0: IBM MFM (8/8 sectors) from Raw Flux (126872 flux in 333.76ms)
T16.0: IBM MFM (8/8 sectors) from Raw Flux (127735 flux in 333.71ms)
T17.0: IBM MFM (8/8 sectors) from Raw Flux (126828 flux in 333.63ms)
T18.0: IBM MFM (8/8 sectors) from Raw Flux (125992 flux in 333.61ms)
T19.0: IBM MFM (8/8 sectors) from Raw Flux (127251 flux in 333.57ms)
T20.0: IBM MFM (8/8 sectors) from Raw Flux (123632 flux in 333.53ms)
T21.0: IBM MFM (8/8 sectors) from Raw Flux (124881 flux in 333.50ms)
T22.0: IBM MFM (8/8 sectors) from Raw Flux (125069 flux in 333.47ms)
T23.0: IBM MFM (8/8 sectors) from Raw Flux (125239 flux in 333.44ms)
T24.0: IBM MFM (8/8 sectors) from Raw Flux (125147 flux in 333.42ms)
T25.0: IBM MFM (8/8 sectors) from Raw Flux (125320 flux in 333.38ms)
T26.0: IBM MFM (8/8 sectors) from Raw Flux (124748 flux in 333.38ms)
T27.0: IBM MFM (8/8 sectors) from Raw Flux (124867 flux in 333.36ms)
T28.0: IBM MFM (8/8 sectors) from Raw Flux (124725 flux in 333.32ms)
T29.0: IBM MFM (8/8 sectors) from Raw Flux (125378 flux in 333.33ms)
T30.0: IBM MFM (8/8 sectors) from Raw Flux (125357 flux in 333.30ms)
T31.0: IBM MFM (8/8 sectors) from Raw Flux (125573 flux in 333.31ms)
T32.0: IBM MFM (8/8 sectors) from Raw Flux (128527 flux in 333.27ms)
T33.0: IBM MFM (8/8 sectors) from Raw Flux (130728 flux in 333.25ms)
T34.0: IBM MFM (8/8 sectors) from Raw Flux (125849 flux in 333.24ms)
T35.0: IBM MFM (8/8 sectors) from Raw Flux (125829 flux in 333.24ms)
T36.0: IBM MFM (8/8 sectors) from Raw Flux (125781 flux in 333.21ms)
T37.0: IBM MFM (8/8 sectors) from Raw Flux (125757 flux in 333.24ms)
T38.0: IBM MFM (8/8 sectors) from Raw Flux (125773 flux in 333.21ms)
T39.0: IBM MFM (8/8 sectors) from Raw Flux (125783 flux in 333.19ms)
T40.0: IBM MFM (8/8 sectors) from Raw Flux (125805 flux in 333.19ms)
T41.0: IBM MFM (8/8 sectors) from Raw Flux (125825 flux in 333.17ms)
T42.0: IBM MFM (8/8 sectors) from Raw Flux (125847 flux in 333.17ms)
T43.0: IBM MFM (8/8 sectors) from Raw Flux (125857 flux in 333.16ms)
T44.0: IBM MFM (8/8 sectors) from Raw Flux (125904 flux in 333.16ms)
T45.0: IBM MFM (8/8 sectors) from Raw Flux (125936 flux in 333.13ms)
T46.0: IBM MFM (8/8 sectors) from Raw Flux (125937 flux in 333.14ms)
T47.0: IBM MFM (8/8 sectors) from Raw Flux (125897 flux in 333.13ms)
T48.0: IBM MFM (8/8 sectors) from Raw Flux (125925 flux in 333.12ms)
T49.0: IBM MFM (8/8 sectors) from Raw Flux (125929 flux in 333.12ms)
T50.0: IBM MFM (8/8 sectors) from Raw Flux (125888 flux in 333.11ms)
T51.0: IBM MFM (8/8 sectors) from Raw Flux (125860 flux in 333.11ms)
T52.0: IBM MFM (8/8 sectors) from Raw Flux (125836 flux in 333.12ms)
T53.0: IBM MFM (8/8 sectors) from Raw Flux (125800 flux in 333.08ms)
T54.0: IBM MFM (8/8 sectors) from Raw Flux (125772 flux in 333.08ms)
T55.0: IBM MFM (8/8 sectors) from Raw Flux (125777 flux in 333.09ms)
T56.0: IBM MFM (8/8 sectors) from Raw Flux (125814 flux in 333.07ms)
T57.0: IBM MFM (8/8 sectors) from Raw Flux (125804 flux in 333.07ms)
T58.0: IBM MFM (8/8 sectors) from Raw Flux (125836 flux in 333.05ms)
T59.0: IBM MFM (8/8 sectors) from Raw Flux (125864 flux in 333.07ms)
T60.0: IBM MFM (8/8 sectors) from Raw Flux (125897 flux in 333.05ms)
T61.0: IBM MFM (8/8 sectors) from Raw Flux (125889 flux in 333.04ms)
T62.0: IBM MFM (8/8 sectors) from Raw Flux (125958 flux in 333.04ms)
T63.0: IBM MFM (8/8 sectors) from Raw Flux (125966 flux in 333.05ms)
T64.0: IBM MFM (8/8 sectors) from Raw Flux (125921 flux in 333.02ms)
T65.0: IBM MFM (8/8 sectors) from Raw Flux (125910 flux in 333.03ms)
T66.0: IBM MFM (8/8 sectors) from Raw Flux (125932 flux in 333.01ms)
T67.0: IBM MFM (8/8 sectors) from Raw Flux (125896 flux in 333.02ms)
T68.0: IBM MFM (8/8 sectors) from Raw Flux (125860 flux in 333.01ms)
T69.0: IBM MFM (8/8 sectors) from Raw Flux (125821 flux in 333.01ms)
T70.0: IBM MFM (8/8 sectors) from Raw Flux (125807 flux in 333.00ms)
T71.0: IBM MFM (8/8 sectors) from Raw Flux (125768 flux in 333.00ms)
T72.0: IBM MFM (8/8 sectors) from Raw Flux (125747 flux in 333.01ms)
T73.0: IBM MFM (8/8 sectors) from Raw Flux (125753 flux in 332.98ms)
T74.0: IBM MFM (8/8 sectors) from Raw Flux (125764 flux in 333.00ms)
T75.0: IBM MFM (8/8 sectors) from Raw Flux (125780 flux in 332.99ms)
T76.0: IBM MFM (8/8 sectors) from Raw Flux (125833 flux in 333.02ms)
Cyl-> 0         1         2         3         4         5         6         7
H. S: 01234567890123456789012345678901234567890123456789012345678901234567890123456
0. 0: .............................................................................
0. 1: .............................................................................
0. 2: .............................................................................
0. 3: .............................................................................
0. 4: .............................................................................
0. 5: .............................................................................
0. 6: .............................................................................
0. 7: .............................................................................
0. 8: .Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 138, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\console.py", line 16, in run
  File "gw.py", line 3, in <module>
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\cli.py", line 69, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\tools\read.py", line 236, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\tools\util.py", line 272, in with_drive_selected
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\tools\read.py", line 164, in read_to_image
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\tools\read.py", line 125, in print_summary
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\codec\ibm\mfm.py", line 124, in has_sec
IndexError: list index out of range
```

Thanks for your help!

keirf, 2023-01-26:
This was indeed a silly off-by-one bug. You can try out a fix by downloading the build artifact below. This also includes a tonne of cleanups since the last official release so you will be doing me a favour by kicking the tyres. Note you need to be logged in to GitHub for the download link to work.

https://github.com/keirf/greaseweazle/suites/10593572230/artifacts/527953270

hharte, 2023-01-28:
Thanks, @keirf this solves the issue I observed.

closed 2023-01-28 by hharte