keirf/greaseweazle #275: Diskdef: Option for sector and/or track padding in IMG file

keirf, 2023-01-29:
As discussed in #143 Olivetti M20 (and others) may pad sectors in single-density track0 so that all sectors are addressed uniformly. In this case, padding the 128b sectors on track 0 to 256b in the IMG file.

Some options:
1. A file-scoped `file_sec_size` and/or `file_track_size` option applied to seeking to any sector or track.
2. Per track `file_size`
3. Per sector `file_size`

Possibly some requirements gathering needed to see what actually is needed and makes most sense. Also some consideration to what is most easily achieved to support the same in FlashFloppy firmware's IMG.CFG feature.

eberhab, 2023-01-29:
Summarizing, for the M20 both approaches would be identical since on the floppy all but the first sector are empty. There is an argument for using a per-sector approach, when writing mame-created imgs, since mame uses (non-zero) sector padding.
I don't see a preference for the scope, based on a gut feeling I would probably have gone with a target size per sector/ track, rather than a file size/ file based option.

galibert, 2023-01-30:
Amusingly, mame does not use non-zero padding explicitely, it just doesn't write the second half in the sector and skips 128 bytes to the next one.  Go figure what it is that really does the padding (other parts of mame, the OS, who knows).

keirf, 2023-01-30:
Please try the followinmg firmware. The `olivetti.m20` diskdef has been updated to include the padding. Let me know how it goes! By the way, you need to be logged in to GitHub for the download link to work.

https://github.com/keirf/greaseweazle/suites/10656233061/artifacts/532600461

eberhab, 2023-01-30:
Wow great! Happy to try, but I am having trouble to extract the downloaded archive.

keirf, 2023-01-30:
I'll have a look in a bit. What happens and where are you trying to extract?

eberhab, 2023-01-30:
With unzip under bash/ ubuntu:
```
$ unzip greaseweazle.ci.1.8.dev1+g0d61ebb.zip 
Archive:  greaseweazle.ci.1.8.dev1+g0d61ebb.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
```

keirf, 2023-01-30:
I think your download was bad for some reason. Perhaps you didn't get the zip file but a small error text file or somesuch? I have downloaded and confirmed all okay on my Ubuntu desktop. Details of the zip file:
```
viper:~$ ls -l ~/Downloads/greaseweazle.ci.1.8.dev1+g0d61ebb.zip 
-rw-rw-r-- 1 keir keir 7423999 Jan 30 11:41 /home/keir/Downloads/greaseweazle.ci.1.8.dev1+g0d61ebb.zip
viper:~$ md5sum ~/Downloads/greaseweazle.ci.1.8.dev1+g0d61ebb.zip 
69f90284ad102125bb28e4b3c15ee353  /home/keir/Downloads/greaseweazle.ci.1.8.dev1+g0d61ebb.zip
viper:~$ unzip -l ~/Downloads/greaseweazle.ci.1.8.dev1+g0d61ebb.zip 
Archive:  /home/keir/Downloads/greaseweazle.ci.1.8.dev1+g0d61ebb.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
  7462364  2023-01-30 09:05   greaseweazle-1.8.dev1+g0d61ebb-win.zip
   110174  2023-01-30 09:01   greaseweazle-1.8.dev1+g0d61ebb.zip
---------                     -------
  7572538                     2 files
```

keirf, 2023-01-30:
Remember you must be logged in for the download to work. Here is an alternative link that should work without login:
https://nightly.link/keirf/greaseweazle/actions/artifacts/532600461.zip

eberhab, 2023-01-30:
Thanks! Indeed the first download seems to have failed at 4 MB for some reason. Will make some tests.

Also thanks for exploring the track-scoped sector-padding option. It seemed more readable and versatile than the file scope, though I don't know if there are disks where one would want to define different padding settings in different track scopes. As an analog for a potential track padding, I guess bytes-per-track/ `img_bpt` could be an option.

keirf, 2023-01-30:
It simply was the most convenient scope for the diskdef handler, as it makes the parameter readily available to the MFM and FM handlers. For FlashFloppy, the more natural scope will likely be file-wide, but perhaps I will make it track-wide there too. I will see.

eberhab, 2023-01-30:
Just tested it, writing and reading back a couple of images. It works perfectly both with public archived images, as well as with mame created/ converted images. This is great! Thanks a lot!

This has come a long way, from "most available M20 images are missing the FM track because few people can read them" to "you can now do in a single command". Very neat!

I do not own a FlashFloppy device yet, but if I can be of any help testing the padding on that side (providing empty sample imgs or so) let me know.

eberhab, 2023-01-31:
> Amusingly, mame does not use non-zero padding explicitely, it just doesn't write the second half in the sector and skips 128 bytes to the next one. Go figure what it is that really does the padding (other parts of mame, the OS, who knows).

I also did some digging but could not find it either inside the the M20 related files. Unfortunately my knowledge of the MAME guts is not sufficient. Thinking that it could be the OS responsible doesn't help pinning it down either :P Mh, I really wonder if it's just "undefined" and then propagates to an unknown place, and if so, if one could actually define it to be zero in the m20_dsk.c. Let's see if someone replies to the Bannister-Forum thread. 

> Please try the following build. The olivetti.m20 diskdef has been updated to include the padding. Let me know how it goes!

I also tried with a 1.2 MB drive, combining with the step parameter, works like a charm.

keirf, 2023-01-31:
MAME has wrappers around file accesses, which allow filler value to be specified https://github.com/mamedev/mame/blob/master/src/lib/util/ioprocsfill.h

This filler seems to default to 0xff for image writes, quite possibly via https://github.com/mamedev/mame/blob/master/src/tools/image_handler.cpp#L311

eberhab, 2023-01-31:
Interesting, looks like it's on purpose then. Good to know. It can cause a bit of confusing, the padding with 0xff, because when converting between formats, you can end up with an image that has identical data but different checksum.

This would also have implications on people's image files if the m20 implementation ever switched to track padding. let's hope it doesn't.

This is probably where the M20 related function "skips" over to the next 256 Bytes:
https://github.com/mamedev/mame/blob/master/src/lib/formats/m20_dsk.cpp#L102

What's interesting is the comment of "//128" behind that line, which makes me think that whoever put it there might have been thinking about sector vs. track padding as well? Hopefully not to change it again in the future. If you do change the file_offset top 128 and then add another file_offset += 2048 _after_ the loop, I guess that's really all it would take to change it.

keirf, 2023-01-31:
The comment is probably to indicate that the actual sector size was considered, and that the 256 is not a bug/typo.

These M20 files always seem to have had sector padding and it seems likely this de facto standard pre-dates MAME?

eberhab, 2023-01-31:
Unfortunately MAME is afik the only reference. Your interpretation of the comment makes sense to me, it is also described this way in the file header.

If you only look at originally imaged img files, you just cannot tell the difference, because all but the first sector are empty. So you in both padding cases you are looking at 128 Bytes of data, followed by 3968 Bytes of zeros (summing up to 4 kiB). (There are other formats around, imd and td0, but I do not know how if/ they do padding. Mame supports them r/o so there could be a possibility to look to learn more if those give a hint.)

Historically, when imaging into img's, people often used the "track method", just because it is easier. (In a pre-GW world) you read FM and MFM separately, then join them with a single `dd` command. The result is the same. Based on the file itself, you cannot tell in which order the zeros were written. (It just starts to make a difference if you write non-zero data like mame does.)

keirf, 2023-01-31:
IMD and TD0 have per-sector metadata, and are self-describing. The issue of padding really doesn't arise for those. They are superior formats, really.

Ah well, anyway, if the need for track padding does arise in future then I can easily add it.

eberhab, 2023-01-31:
Thanks for adding this and for all the insights!

eberhab, 2023-01-31:
Is it planned for GW to support one of those IMD/ TD0?

keirf, 2023-01-31:
I think TD0 is proprietary and/or no public spec. IMD alreayd support in Disk-Utilities so yes, I have C code to do IMD and I will port it to Python for Greasseweazle tools.

keirf, 2023-01-31:
New issue #276

keirf, 2023-01-31:
If you have any example IMD images link/point at a few in #276 and I will add GW support and test against them.

eberhab, 2023-01-31:
Great! I have linked some. Adding here: Those IMDs can already be converted with MAME's `floptool flopconvert imd m20`. Resulting IMGs then shows the 0xff sectors padding (if someone ever needs a sample of this).

eberhab, 2023-07-19:
> For FlashFloppy, the more natural scope will likely be file-wide, but perhaps I will make it track-wide there too. I will see.

Hi @keirf did you realize supporting track padded/ "Olivetti-m20" IMGs in FlashFloppy yet? I briefly looked through the issues but couldn't see it.