keirf/greaseweazle #276: Support IMD image files

keirf, 2023-01-31:
As per title. There are a few flexible formats like this which might also share code.

keirf, 2023-01-31:
Can someone attach or link some interesting IMD files to this ticket for testing purposes?

eberhab, 2023-01-31:
Some IMDs for the M20 can be found here: http://www.z80ne.com/m20/index.php?argument=sections/download/wrm20/wrm20.inc

keirf, 2023-02-01:
Please test the following build. It supports reading and writing IMD image files:
https://github.com/keirf/greaseweazle/suites/10718125599/artifacts/537123881

eberhab, 2023-02-01:
Wow that was quick!

I did a few tests and noticed that the checksum of the IMD file read back from floppy differs from the original IMD file written to it, which I guess is probably fine? Maybe just some flexible ordering of things inside the IMD which produces different checksum with same userdata? (Haven't read the specs.). In fact every-time I read the floppy again and again the resulting IMD has a different checksum.

Converting the resulting IMD to IMG produces an expected output/ identical sector content, which is what matters. So I think it looks great. 

Some details:

Files used: [IMD](http://www.z80ne.com/m20/sections/download/wrm20/OLITEST.IMD) source (f5e7509b) and its [IMG](https://archive.org/download/OlivettiM20/olitest.img) version (7519913c).

Testing:
* IMD -> Floppy -> IMG (OK 7519913c)
* IMD -> Floppy -> IMD (BAD 9178f98c != f5e7509b)
* IMD -> Floppy -> IMD -> IMG (OK 7519913c)

The mismatching IMDs all boot fine in MAME and are identical to known IMG when converted. Conversion IMD -> IMG was done using MAME's floptool. As discussed [here](https://github.com/keirf/greaseweazle/issues/275) one needs to remove the 0xff padding to compare. For M20 images (only) I can use the following commands:

```
floptool flopconvert imd m20 floppy.imd tmp.img
dd if=tmp.img of=floppy.img bs=128 count=1 >/dev/null
dd if=tmp.img of=floppy.img bs=4096 skip=1 seek=1 >/dev/null
```
Thanks a lot for adding this!

keirf, 2023-02-01:
No problem! Regenerated IMD will always differ from the original because the image header contains the time of creation. Also there is a variable sized comment section in the same header.

eberhab, 2023-02-01:
Interesting! Did you come across any tools/ an easy way to read these extra info (time/ comment) from the files under e.g. bash?

Any chance I can use gw tools to convert between formats?

keirf, 2023-02-01:
The header is ASCII text so you can `strings a.imd | head -1` or `hexdump -C a.imd | less` or various other things.

keirf, 2023-02-01:
You can use `gw convert` to convert between formats. You can also specify a `--format` for the conversion but it's not always needed

eberhab, 2023-02-01:
Ok that was all far easier than I thought. Looking at some creating dates now. That's pretty neat. I find it's an interesting alternative to IMGs. Is the C code that you ported available somewere? Maybe at some point it can be added to enable write support also on the MAME side.

ibm5110, 2023-02-03:
> Interesting! Did you come across any tools/ an easy way to read these extra info (time/ comment) from the files under e.g. bash?

The original ImageDisk tool asks for the information much like AnaDisk.