keirf/greaseweazle #277: SCP files created with extra tracks

ibm5110, 2023-02-03:
When I create an SCP file with gw, either by reading a disk or by converting an image to SCP, the tools from HxC (like the visual floppy) show me a disk with 101 tracks instead of 77 (8" floppy disk image) and won't recognise it as 8" disk with the error "This image doesn't fit on a 8" disk". The tracks 77-100 are empty, though. Is that a bug in gw or in the HxC tools?

ibm5110, 2023-02-03:
Also it says "end track 152" with only 77 tracks; additional info:

```
# hxcfe.exe -infos -finput:/tmp/x/x.scp
HxC Floppy Emulator : Floppy image file converter v2.5.5.1
Copyright (C) 2006-2021 Jean-Francois DEL NERO
This program comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions;

libhxcfe version : 2.13.6.6

Input file : /tmp/x/x.scp
---------------------------------------------------------------------------
-                        File informations                                -
---------------------------------------------------------------------------
File: /tmp/x/x.scp
Checking /tmp/x/x.scp
File loader found : SCP_FLUX_STREAM (SCP Stream Loader)
Loading /tmp/x/x.scp
file loader found!
Can't open config.script !
Loading SCP file...
Version : 0x24
Disk Type : 0x80
Start track : 0
End track : 152
Number of revolution(s) : 1
Flags : 0x03
File Checksum : 0x1EED21D6
Bit Cell width : 0
Number of heads : 1
Resolution factor : 0
Revolution 0 track generation... First valid index position : 0
[...]
Revolution 0 track generation... First valid index position : 0
track file successfully loaded and encoded!
Sanity Checker :  Track 77/Side 0 not allocated ?!?
Sanity Checker :  Track 78/Side 0 not allocated ?!?
Sanity Checker :  Track 79/Side 0 not allocated ?!?
Sanity Checker :  Track 80/Side 0 not allocated ?!?
Sanity Checker :  Track 81/Side 0 not allocated ?!?
Sanity Checker :  Track 82/Side 0 not allocated ?!?
Sanity Checker :  Track 83/Side 0 not allocated ?!?
Sanity Checker :  Track 84/Side 0 not allocated ?!?
Sanity Checker :  Track 85/Side 0 not allocated ?!?
Sanity Checker :  Track 86/Side 0 not allocated ?!?
Sanity Checker :  Track 87/Side 0 not allocated ?!?
Sanity Checker :  Track 88/Side 0 not allocated ?!?
Sanity Checker :  Track 89/Side 0 not allocated ?!?
Sanity Checker :  Track 90/Side 0 not allocated ?!?
Sanity Checker :  Track 91/Side 0 not allocated ?!?
Sanity Checker :  Track 92/Side 0 not allocated ?!?
Sanity Checker :  Track 93/Side 0 not allocated ?!?
Sanity Checker :  Track 94/Side 0 not allocated ?!?
Sanity Checker :  Track 95/Side 0 not allocated ?!?
Sanity Checker :  Track 96/Side 0 not allocated ?!?
Sanity Checker :  Track 97/Side 0 not allocated ?!?
Sanity Checker :  Track 98/Side 0 not allocated ?!?
Sanity Checker :  Track 99/Side 0 not allocated ?!?
Sanity Checker :  Track 100/Side 0 not allocated ?!?

File type : SCP_FLUX_STREAM - SCP Stream Loader
Floppy interface mode : GENERIC_SHUGART_DD_FLOPPYMODE - Shugart Interface
Number of Track : 101
Number of Side : 1
Total Size : 256256 Bytes, Number of sectors : 2002
Stopping HxCFloppyEmulator...
```

keirf, 2023-02-03:
Your HxC tools are probably old. Try the beta version.

ibm5110, 2023-02-03:
Almost the same result, but with 99 instead of 101 tracks:

```
HxC Floppy Emulator : Floppy image file converter v2.5.5.1
Copyright (C) 2006-2022 Jean-Francois DEL NERO
This program comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions;

libhxcfe version : 2.13.19.2

Input file : /tmp/x/x.scp
---------------------------------------------------------------------------
-                        File informations                                -
---------------------------------------------------------------------------
File: /tmp/x/x.scp
Checking /tmp/x/x.scp
File loader found : SCP_FLUX_STREAM (SCP Stream Loader)
Loading /tmp/x/x.scp
file loader found!
Can't open config.script !
Loading SCP file...
Version : 0x24
Disk Type : 0x80
Start track : 0
End track : 152
Number of revolution(s) : 1
Flags : 0x03
File Checksum : 0x1EED21D6
Bit Cell width : 0
Number of heads : 1
Resolution factor : 0
Revolution 0 track generation... First valid index position : 0
[...]
Revolution 0 track generation... First valid index position : 0
track file successfully loaded and encoded!
Sanity Checker :  Track 77/Side 0 not allocated ?!?
Sanity Checker :  Track 78/Side 0 not allocated ?!?
Sanity Checker :  Track 79/Side 0 not allocated ?!?
Sanity Checker :  Track 80/Side 0 not allocated ?!?
Sanity Checker :  Track 81/Side 0 not allocated ?!?
Sanity Checker :  Track 82/Side 0 not allocated ?!?
Sanity Checker :  Track 83/Side 0 not allocated ?!?
Sanity Checker :  Track 84/Side 0 not allocated ?!?
Sanity Checker :  Track 85/Side 0 not allocated ?!?
Sanity Checker :  Track 86/Side 0 not allocated ?!?
Sanity Checker :  Track 87/Side 0 not allocated ?!?
Sanity Checker :  Track 88/Side 0 not allocated ?!?
Sanity Checker :  Track 89/Side 0 not allocated ?!?
Sanity Checker :  Track 90/Side 0 not allocated ?!?
Sanity Checker :  Track 91/Side 0 not allocated ?!?
Sanity Checker :  Track 92/Side 0 not allocated ?!?
Sanity Checker :  Track 93/Side 0 not allocated ?!?
Sanity Checker :  Track 94/Side 0 not allocated ?!?
Sanity Checker :  Track 95/Side 0 not allocated ?!?
Sanity Checker :  Track 96/Side 0 not allocated ?!?
Sanity Checker :  Track 97/Side 0 not allocated ?!?
Sanity Checker :  Track 98/Side 0 not allocated ?!?

File type : SCP_FLUX_STREAM - SCP Stream Loader
Floppy interface mode : GENERIC_SHUGART_DD_FLOPPYMODE - Shugart Interface
Number of Track : 99
Number of Side : 1
Total Size : 256256 Bytes, Number of sectors : 2002
Stopping HxCFloppyEmulator...
```

ibm5110, 2023-02-03:
This is the start of the gw generated SCP file:

```
0000 0000: 53 43 50 24 80 01 00 98  03 00 01 00 D6 21 ED 1E  SCP$.... .....!..
0000 0010: 64 05 00 00 00 00 00 00  28 E6 01 00 00 00 00 00  d....... (.......
0000 0020: DE B8 03 00 00 00 00 00  36 85 05 00 00 00 00 00  ........ 6.......
```

It clearly states $98=152 tracks instead of 77, although only 1 head is used. Therefore, the offsets to the track data is strange, every other offset is zero (for the missing second head).

keirf, 2023-02-03:
It's a HxC bug. It can't even load its *own* single-sided SCP files. The SCP image spec says that every other offset should be zero for a single-sided image. It's unclear whether track start/end/number values should be doubled or not. I double them. HxC doubles them. But HxC then seems to get confused about end track on import. Blame the SCP spec and lack of example image files -- HxC author is just as confused as I am.

A simple workaround would be to convert to a format that doesn't anger tools authors. EG. kryoflux:
```
gw convert my.scp my_kf/00.0.raw
```

And import *that* to HxC.

keirf, 2023-02-03:
Also raise the bug with HxC. In the HxC gui I generated a single-sided IBM disk. 80 cyls. I exported as SCP. I then loaded the SCP. The disk now had 160(!) cyls.

closed 2023-02-05 by keirf