keirf/greaseweazle #279: 'NoneType' object has no attribute 'decode_raw'

Bzzz, 2023-02-05:
Hi,

new greaseweazle v4 user here - I've read a couple of 5.25" HD disks without problems (amazing!) and still have a couple of lower density ones left. Those will read fine as well, but there's no img file produced at the end:

> Reading c=0-79:h=0-1:step=2 revs=2 (MD2D double sided double density 48TPI disk)
Format ibm.360
T0.0: IBM MFM (9/9 sectors) from Raw Flux (88102 flux in 333.71ms)
...and so on, all the same...
T39.1: IBM MFM (9/9 sectors) from Raw Flux (75989 flux in 333.73ms)
** FATAL ERROR:
'NoneType' object has no attribute 'decode_raw'

Is this a bug or am I doing something wrong? Same Newtronics/Mitsumi D509V2 drive, same 1.8 software, using the GreaseweazleGUI as indicated on the wiki page.

keirf, 2023-02-05:
Almost certainly a new bug. Please can you try v1.7?

keirf, 2023-02-05:
Also please get a backtrace by adding `--bt` to your command line between `gw` and `read`. `gw --bt read ...`

keirf, 2023-02-05:
I think it's been broken in this way since v1.5 (released 4 Jan) and it's because your trackspec is wrong. You need this on the command line:
```
--tracks=step=2
```

Don't change the cylinders! These are automatically set correctly for the format (`ibm.360`) and are in "logical"/disk cylinders, not drive cylinders! You are actually trying to read cylinders 40-79 off the disk, and of course they don't exist and, due to a fairly recent bug, it actually crashes the `gw read` command. I will get that fixed but it's less critical than I first thought.

Bzzz, 2023-02-06:
So that worked...
```
Reading c=0-39:h=0-1:step=2 revs=2
Format ibm.360
T0.0: IBM MFM (9/9 sectors) from Raw Flux (88103 flux in 333.62ms)
T0.1: IBM MFM (9/9 sectors) from Raw Flux (87951 flux in 333.63ms)
T1.0: IBM MFM (9/9 sectors) from Raw Flux (79221 flux in 333.62ms)
T1.1: IBM MFM (9/9 sectors) from Raw Flux (83898 flux in 333.63ms)
T2.0: IBM MFM (9/9 sectors) from Raw Flux (80675 flux in 333.64ms)
T2.1: IBM MFM (9/9 sectors) from Raw Flux (82293 flux in 333.65ms)
T3.0: IBM MFM (9/9 sectors) from Raw Flux (79138 flux in 333.65ms)
T3.1: IBM MFM (9/9 sectors) from Raw Flux (79309 flux in 333.63ms)
T4.0: IBM MFM (9/9 sectors) from Raw Flux (84154 flux in 333.65ms)
T4.1: IBM MFM (9/9 sectors) from Raw Flux (87398 flux in 333.66ms)
T5.0: IBM MFM (9/9 sectors) from Raw Flux (78857 flux in 333.65ms)
T5.1: IBM MFM (9/9 sectors) from Raw Flux (78353 flux in 333.65ms)
T6.0: IBM MFM (9/9 sectors) from Raw Flux (78602 flux in 333.66ms)
T6.1: IBM MFM (9/9 sectors) from Raw Flux (78160 flux in 333.65ms)
T7.0: IBM MFM (9/9 sectors) from Raw Flux (80365 flux in 333.65ms)
T7.1: IBM MFM (9/9 sectors) from Raw Flux (78301 flux in 333.65ms)
T8.0: IBM MFM (9/9 sectors) from Raw Flux (80122 flux in 333.65ms)
T8.1: IBM MFM (9/9 sectors) from Raw Flux (78773 flux in 333.65ms)
T9.0: IBM MFM (9/9 sectors) from Raw Flux (79819 flux in 333.66ms)
T9.1: IBM MFM (9/9 sectors) from Raw Flux (79673 flux in 333.65ms)
T10.0: IBM MFM (9/9 sectors) from Raw Flux (78189 flux in 333.65ms)
T10.1: IBM MFM (9/9 sectors) from Raw Flux (78053 flux in 333.66ms)
T11.0: IBM MFM (9/9 sectors) from Raw Flux (78412 flux in 333.65ms)
T11.1: IBM MFM (9/9 sectors) from Raw Flux (78257 flux in 333.66ms)
T12.0: IBM MFM (9/9 sectors) from Raw Flux (79681 flux in 333.65ms)
T12.1: IBM MFM (9/9 sectors) from Raw Flux (92190 flux in 333.65ms)
T13.0: IBM MFM (9/9 sectors) from Raw Flux (83124 flux in 333.66ms)
T13.1: IBM MFM (9/9 sectors) from Raw Flux (75933 flux in 333.66ms)
T14.0: IBM MFM (9/9 sectors) from Raw Flux (83020 flux in 333.66ms)
T14.1: IBM MFM (9/9 sectors) from Raw Flux (80166 flux in 333.66ms)
T15.0: IBM MFM (9/9 sectors) from Raw Flux (78849 flux in 333.66ms)
T15.1: IBM MFM (9/9 sectors) from Raw Flux (79013 flux in 333.66ms)
T16.0: IBM MFM (9/9 sectors) from Raw Flux (82323 flux in 333.65ms)
T16.1: IBM MFM (9/9 sectors) from Raw Flux (79430 flux in 333.66ms)
T17.0: IBM MFM (9/9 sectors) from Raw Flux (82634 flux in 333.66ms)
T17.1: IBM MFM (9/9 sectors) from Raw Flux (88570 flux in 333.66ms)
T18.0: IBM MFM (9/9 sectors) from Raw Flux (77368 flux in 333.67ms)
T18.1: IBM MFM (9/9 sectors) from Raw Flux (78757 flux in 333.66ms)
T19.0: IBM MFM (9/9 sectors) from Raw Flux (79048 flux in 333.66ms)
T19.1: IBM MFM (9/9 sectors) from Raw Flux (79729 flux in 333.66ms)
T20.0: IBM MFM (9/9 sectors) from Raw Flux (79886 flux in 333.65ms)
T20.1: IBM MFM (9/9 sectors) from Raw Flux (80873 flux in 333.65ms)
T21.0: IBM MFM (9/9 sectors) from Raw Flux (79392 flux in 333.66ms)
T21.1: IBM MFM (9/9 sectors) from Raw Flux (81837 flux in 333.65ms)
T22.0: IBM MFM (9/9 sectors) from Raw Flux (83101 flux in 333.65ms)
T22.1: IBM MFM (9/9 sectors) from Raw Flux (83377 flux in 333.65ms)
T23.0: IBM MFM (9/9 sectors) from Raw Flux (83848 flux in 333.65ms)
T23.1: IBM MFM (9/9 sectors) from Raw Flux (83545 flux in 333.66ms)
T24.0: IBM MFM (9/9 sectors) from Raw Flux (83464 flux in 333.66ms)
T24.1: IBM MFM (9/9 sectors) from Raw Flux (79121 flux in 333.65ms)
T25.0: IBM MFM (9/9 sectors) from Raw Flux (78304 flux in 333.66ms)
T25.1: IBM MFM (9/9 sectors) from Raw Flux (78373 flux in 333.65ms)
T26.0: IBM MFM (9/9 sectors) from Raw Flux (77709 flux in 333.66ms)
T26.1: IBM MFM (9/9 sectors) from Raw Flux (77350 flux in 333.66ms)
T27.0: IBM MFM (9/9 sectors) from Raw Flux (79132 flux in 333.66ms)
T27.1: IBM MFM (9/9 sectors) from Raw Flux (78073 flux in 333.65ms)
T28.0: IBM MFM (9/9 sectors) from Raw Flux (79827 flux in 333.67ms)
T28.1: IBM MFM (9/9 sectors) from Raw Flux (78554 flux in 333.66ms)
T29.0: IBM MFM (9/9 sectors) from Raw Flux (77953 flux in 333.66ms)
T29.1: IBM MFM (9/9 sectors) from Raw Flux (77145 flux in 333.65ms)
T30.0: IBM MFM (9/9 sectors) from Raw Flux (76847 flux in 333.65ms)
T30.1: IBM MFM (9/9 sectors) from Raw Flux (77461 flux in 333.65ms)
T31.0: IBM MFM (9/9 sectors) from Raw Flux (81060 flux in 333.65ms)
T31.1: IBM MFM (9/9 sectors) from Raw Flux (78805 flux in 333.65ms)
T32.0: IBM MFM (9/9 sectors) from Raw Flux (77987 flux in 333.65ms)
T32.1: IBM MFM (9/9 sectors) from Raw Flux (86162 flux in 333.65ms)
[... 48 lines cut ...]
T1.1: IBM MFM (0/9 sectors) from Raw Flux (55906 flux in 333.69ms)
T1.1: IBM MFM (0/9 sectors) from Raw Flux (139493 flux in 833.97ms) (Retry #1.1)
T1.1: IBM MFM (0/9 sectors) from Raw Flux (223530 flux in 1334.25ms) (Retry #1.2)
T1.1: IBM MFM (0/9 sectors) from Raw Flux (306805 flux in 1834.54ms) (Retry #1.3)
T1.1: Giving up: 9 sectors missing
T2.0: Ignoring unexpected sector C:1 H:0 R:9 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:5 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:2 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:7 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:8 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:6 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:3 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:1 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:4 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:9 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:5 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:2 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:7 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:8 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:6 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:3 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:1 N:2
T2.0: Ignoring unexpected sector C:1 H:0 R:4 N:2
T2.0: IBM MFM (0/9 sectors) from Raw Flux (79220 flux in 333.70ms)
``

keirf, 2023-02-06:
Yes the docs are bad, they need a bit of TLC. New ticket #280

commit b0b65928 (2023-02-06): Fix handling of tracks out-of-range for format specified.
Fixes #279
closed 2023-02-06 by keirf