keirf/greaseweazle #280: Update the terrible wiki docs

keirf, 2023-02-06:
A lot of user questions are due to lack of good documentation. Not just on specifics of the command line, but general principles of imaging, and greaseweazle's approach to it. The "Getting Started" page https://github.com/keirf/greaseweazle/wiki/Getting-Started is particularly grim.

tdaede, 2023-07-05:
One downside about the github wiki is that it's impossible to open pull requests on it. It might be easier to contribute to if it was a sphinx/mdbook doc in the repo.

keirf, 2023-07-05:
Absolutely true. In fact there are many options better than the GitHub wiki, which is used only for historic reasons of immediate convenience.  For example, Docusaurus served up via Netlify, with the assets in a first-class proper GitHub repo? https://developer.redis.com/create/netlify/deploy-docusaurus-to-netlify/

keirf, 2023-07-05:
I'd be very happy to see the docs migrate :)

wohali, 2023-07-05:
If we need a multi-user-editable wiki, I'd recommend Wiki.JS as an alternative? It supports a wide variety of backends and lots of plugins. But largely only you edit the wiki 😉 

For static docs, and since this is Python, Sphinx / ReadTheDocs.org would be a logical choice, and since this is an open source project you could publish docs directly from this repo with a simple setup on their side. I've used this for many of my OSS projects, including Apache CouchDB, for over a decade with zero problems.

Or if you prefer a static site generator like docusaurus, jekyll, hugo, nikola are all roughly the same, and it doesn't really matter where it's hosted.