keirf/greaseweazle #281: Feature request: Support MAME's MFI disk image format

drencorxeen, 2023-02-10:
Would it be possible to request support for MAME/MESS's MFI disk image format.

So that it would make it easier to import floppy disks for use with MAME and then after using MAME can then write the disk image back to real floppies.

keirf, 2023-02-11:
This is possible. It sounds like a flux image format that could represent many types of disk format. Are there example MFI images available containing known formats, for validating MFI support?

drencorxeen, 2023-02-11:
I can at least format DD and HD versions in a emulation that I use commonly.  Plus I can try to find the source for image format.

keirf, 2023-02-11:
If you can format an MFI image, attach one or two to this ticket.

drencorxeen, 2023-02-11:
[MFI_Example_Images.zip](https://github.com/keirf/greaseweazle/files/10714598/MFI_Example_Images.zip)
Here is both DD and HD examples of the MFI disk images.

drencorxeen, 2023-02-11:
Here is also a link to MAME/MESS's code for the MFI disk image.

https://github.com/mamedev/mame/blob/master/src/lib/formats/mfi_dsk.cpp

drencorxeen, 2023-02-23:
[MFI_Examples_IBMPC.zip](https://github.com/keirf/greaseweazle/files/10817808/MFI_Examples_IBMPC.zip)

Here is a IBM Compatible formatted 720K and 1.44MB MFI disk images as well.