keirf/greaseweazle #283: Converting Amiga IPF to SCP - Splice level

merlinkv, 2023-02-19:
Hi Keirf,

I have tried converting Amiga games from IPF to SCP format several times and it works really well (much better than the HxC utility).

For example, the game Stormlord, the HxC utility converts it wrong resulting in an unplayable disc. However the GW converts it perfectly and you can play it on emulator or in real Amiga when writing the SCP to disk.

However, the resulting SCP files have a Splice Level of 2 and cannot be changed (or I don't know). Would it be possible to add an option to be able to convert the IPF's to a splice level of 3, 4 or 5?

Thanks and best regards.

keirf, 2023-02-19:
What is "splice level"?

merlinkv, 2023-02-19:
From the SuperCard Pro Manual

"Copy Mode

The Copy Mode is the method that will be used to copy or image a disk.
INDEX - uses the index pulse to capture a single revolution. Most all
commercially produced disks can be copied/imaged using this copy mode.
SPLICE - uses an intelligent method to determine where the track start/stop
is located. When imaging a disk, you can choose the number of revolutions
to capture. This handy for 3rd party programs that analyze the disk format
and create image files from the raw flux image. This copy mode is
substantially slower, but provides a solution for tracks that were not written
using the index mark when created."

Convert option of HxC make SCP's with Splice Level of 3, but many of them results on not working images/disks.

Convert option of GW make SCP's with Splice Level of 2 (two accuracy revolutions), from now all images/disks works fine.

Obviously the GW conversion algorithm is much better.

My question is if you could add the option Splice Level (Revs to read per track) in the conversion algorithm from Amiga IPF to Amiga SCP

Thanks and best regards.

keirf, 2023-02-19:
Ah, revolutions per track in the output SCP image! Hmmm well that makes sense. I'm not sure how easy it is for me to implement in my current pipeline. Let me have a look.

keirf, 2023-02-19:
Is there a particular specific reason to want more revolutions ("splice levels") by the way?

merlinkv, 2023-02-19:
With two revs the real disks works, but it is noticeable that the floppy drives have a harder time reading the disks (they become very noisy), although they read them without problems.

With more revs (disks from SuperCard Pro), it seems that the floppy drives suffer less, they are less noisy when reading the disks on real Amiga's.

Tests made with GW and SuperCard Pro, with DD disks and floppy drives in perfect condition.

keirf, 2023-02-19:
How is the Greaseweazle generated SCP written to real disk?

merlinkv, 2023-02-19:
Sorry for the delay to answer, I have been doing more tests with different games.

**Examples:**

**Stormlord**

Converted from IFF to SCP works perfect on emulators and can write back OK to real disk.

Readed real disk with GW with 2, 3 or 4 revs, works only on emulators, can't write back OK to disk (some time works, but 95% of times not).

**Spell Bound (Psygnosis), Speedball and Strike Force Harrier games**

Converted from IFF to SCP works perfect on emulators and can write back OK to real disks.

Readed real disks with GW with 3 or 4 revs, works on emulators and can write back OK to disks but, with 3 or 4 revs the floppy drive works less noisy compared with the 2 revs disks.

----- 

Not that it's very important, I just wondering if there was a possibility since floppy drives work a bit quieter with more revs per track.

keirf, 2023-02-19:
I don't think any of this has much to do with number of revs in the SCP image. In any case, writing to disk from SCP generated from IPF is not a recommended operation: Simply write the IPF directly to disk.

merlinkv, 2023-02-19:
The idea is to have SCP images that can be used in emulators and can be written to real disks like "SCP Preservation Images".

Anyway, if you think it's not useful or necessary, no problem.

Thank you very much as always.

keirf, 2023-02-19:
In general conversion does not produce a superior preservation image. I always recommend to keep original raw dumps for copy protected disks. Also IPF is good as it is generated from an awareness of the copy protection track formats. "Dumb" downstream conversions from these two are basically always inferior (to a first approximation) except for specific use cases: eg hfev3 to use in floppy emulator.

keirf, 2023-02-19:
I'm not actually sure that SCP as a downstream format is useful at all! It's useful for original raw dumps mainly. Perhaps it's also useful for downstream tools that accept say SCP and not IPF (for example). In that case it would be an intermediate format only, or for a single use case, rather than your "gold image".

merlinkv, 2023-02-19:
We have different opinions :) ... IPF files are small in size but you can't produce IPF files like the SPS even with your Disk Utilities. In the other hand you need the dll's to read IPF files in emulators.

SCP files are bigger but you can create your images/disk with cheap hardware (one SuperCard Pro is not needed, just the Greaseweazle) and free software and you don't need any dll to use the images in emulators like WinUAE.

keirf, 2023-02-19:
Yes an SCP raw dump is fine. That is quite a different thing than a generated SCP (from IPF or whatever). Also of course you can already create an SCP raw dump with however many revolutions you like.

merlinkv, 2023-02-19:
Yes but this not work fine on games like Stormlord ... just convert IPF to SCP and write the SCP to disk.

In can't create a perfect SCP image from real disk using Greaseweazle or SuperCard Pro ... 2, 3, 4, o 5 revs ... no way ...

keirf, 2023-02-19:
Well don't do that. Write the IPF to disk. No SCP .

The raw SCP dump is probably fine but without knowing where the write splice should be it can't be written back to disk unless the write splices are all at the index. On some Amiga disks this isn't the case. Also weak bits would not be treated correctly for writeback either. I guess SCP isn't a good writeback format usually.

merlinkv, 2023-02-19:
I love the SCP format :)

About the SCP format ...

[](https://www.cbmstuff.com/downloads/scp/scp_image_specs.txt)

closed 2023-03-11 by keirf