keirf/greaseweazle #284: Prevent double stepping when writing 40-track D88 image to 40-track drive

sdsnatcher73, 2023-02-23:
When writing a 40 track D88 image the host software will perform double stepping. While this is understandable if the image is written to an 80 track (4D or HD drive) it does not when writing the image to a 40 track drive. I think the double stepping might actually take place in converting the D88 image to something gw uses internally. I am using firmware 1.3, host software v1.8

Here is the output of the command on an 80 track drive (Panasonic JU-475, A: on IBM PC cable):

```
**gw write PC-8801FH\ system\ disk\ \(Demonstration\).d88**          
Writing c=0-81:h=0-1
T0.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T0.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T2.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T2.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T4.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T4.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T6.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T6.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
...
T74.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T74.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T76.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T76.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T78.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T78.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
All tracks verified
```
On a 40 track drive (Panasonic JU-455, B: on IBM PC cable) the output is:
```
**gw write PC-8801FH\ system\ disk\ \(Demonstration\).d88 --drive b**
Writing c=0-81:h=0-1
T0.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T0.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T2.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T2.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T4.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T4.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
...
T40.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T40.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T42.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T42.1: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T42.1: Writing Track (Verify Failure: Retry #1)
T42.1: Writing Track (Verify Failure: Retry #2)
T42.1: Writing Track (Verify Failure: Retry #3)
```

If I convert the image to SCP with HxC software the output on the 40 track drive is:
```
**gw write PC-8801FH\ system\ disk\ \(Demonstration\)_d88.scp --drive b**                                  
Writing c=0-81:h=0-1
T0.0: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T0.1: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T1.0: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T1.1: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T2.0: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T2.1: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
...
T38.0: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T38.1: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T39.0: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
T39.1: Writing Track (Flux: 200.1ms period, 220.1 ms total, Terminate at index)
No tracks verified (Reason: Verify unavailable)
```

sdsnatcher73, 2023-02-23:
Double stepping is done in d88.py irrespective of target drive:
`                if media_flag == 0x00:
                    physical_cyl = track_index // 2 * 2
                else:
                    physical_cyl = track_index // 2`

If I change it to
`                physical_cyl = track_index // 2`

keirf, 2023-02-23:
Agreed. Please confirm the fix checked in a moment ago.

sdsnatcher73, 2023-02-23:
This fixes it for writing to a 40 track drive. Writing to a 80 track drive fails now, because no double stepping is performed in the output fase yet. The writing is fine obviously (40 tracks are written) but the PC-88 will fail reading the tracks as track 1 is written in-between where track 0 and 1 are expected.

sdsnatcher73, 2023-02-23:
Indeed that works well. Issue fixed.

closed 2023-02-23 by sdsnatcher73