keirf/greaseweazle #286: Feature request: Specify IBM gap byte value(s)

keirf, 2023-03-11:
From David Ladd on FB: "There is a trick with the WD179x and WD279x that if gap2 and gap3 are reduced to 1 byte and then the value written is 0x00 rather than 0x4E you can manage to squeeze 20spt at 256bps without problem. "

Seems allowing gap2 and gap3 byte values of 0x00 would suffice, but probably makes sense to allow byte value for each of the gap types to be independently set to any value.

drencorxeen, 2023-03-12:
This indeed would be a very useful feature.  I agree!

keirf, 2023-05-11:
This is now implemented on `master`. Can you please test? You define your own diskdef and add new parameter `gapbyte = 0` to the track parameters.

If you want a Windows prebuilt you can find one here: https://nightly.link/keirf/greaseweazle/actions/artifacts/691428559.zip

Or install tip of master from source into your existing Python environment:
```python -m pip install git+https://github.com/keirf/greaseweazle```

drencorxeen, 2023-05-11:
Any way the gap#byte's could be changed independently?

In the CoCo's use with the 20spt layout gap1 and gap4a are still a byte value of 0x4E where as gap2 and gap3 are a bye value of 0x00.
Or does this make the process to complicated?

I know from the test on real hardware the WD17xx is having issues reading a disk that was created with all gap bytes being written as 0x00.

keirf, 2023-05-11:
Here's a build which forces the track gap to be 0x4e: https://nightly.link/keirf/greaseweazle/actions/artifacts/692053442.zip

Or to install from source:
```
python -m pip install git+https://github.com/keirf/greaseweazle@gapbyte
```

Also, please share the diskdef you are using to create these special tracks.

closed 2023-05-12 by keirf