keirf/greaseweazle #294: macOS - Could not find SPS/CAPS IPF decode library

nicktones, 2023-04-06:
I'm getting the could not find SPS/CAPS error after updating to 1.9 main branch. Had previously worked with the macos-caps branch.

Maybe 1.9 isn't searching for the framework in the right place?

keirf, 2023-04-06:
It hasn't changed since the original fix.

nicktones, 2023-04-06:
Strange. Nothing has changed on my end other than updating the tools.

keirf, 2023-04-07:
What do you have under /Library/Frameworks

nicktones, 2023-04-07:
[image]

CAPSImg.framework is from link in wiki.

keirf, 2023-04-07:
What is the Mac model and CPU? Can you show contents of CAPSImg.framework folder?

nicktones, 2023-04-07:
[image]

M2 MacBook Air 2022

keirf, 2023-04-07:
I added a bit fo debug logging to a `macos` branch. You can install in your Python environment as follows
```
python3 -m pip install git+https://github.com/keirf/greaseweazle@macos
```

When you try to read an IPF it will print lines `Trying ...` and list the library paths being attempted. `/Library/Frameworks` is expected to be implicitly tried when the name `CAPSImg.framework/CAPSImg` is passed. Perhaps it isn't tried now for some reason? There certainly isn't much in that folder on your system. Perhaps `/System/Library/Frameworks` is more popular? You could also try in your home folder: `~/Library/Frameworks`

nicktones, 2023-04-07:
Thanks for this. Output from debug build attached - doesn't seem like it's looking in the right place. Under macOS Ventura /System/Library/Frameworks isn't writable and there isn't a frameworks folder in the home library.
[image]

Looks like it's looking in the app folder maybe?

keirf, 2023-04-07:
Yes it's looking explicitly in the app folder. You could put it exactly there at least to see whether that works.

Alternatively try the other things that ought to be on the search path by default:
- `/System/Library/Frameworks`
- `~/Library/Frameworks`

keirf, 2023-04-07:
Oh I see `/System/Library/Frameworks` doesn't exist... You could try creating the one in your home folder. `mkdir -p ~/Library/Frameworks`

You can delete it if it doesn't work.

keirf, 2023-04-07:
You can also now reinstall the tools same as before:
```
python3 -m pip install git+https://github.com/keirf/greaseweazle@macos
```

I have *explicitly* added the `/Library/Frameworks` path.

nicktones, 2023-04-07:
Amazing - explicitly adding the folder worked with the new build. Thanks!

FYI - creating the Frameworks folder in the home folder and adding there didn't work.

All sorted with the latest debug build. Thanks again

keirf, 2023-04-07:
How weird. I'll add that for the next release. Does no harm.

nicktones, 2023-04-07:
Indeed. Could be a change in latest Ventura update I guess.

Thanks again.

closed 2023-04-07 by nicktones

keirf, 2023-04-07:
This is now in master branch:
```
python3 -m pip install git+https://github.com/keirf/greaseweazle
```

commit 4c0e90da (2023-04-07): macos: Search for SPS/CAPS explicitly under /Library/Frameworks

Fixes #294