keirf/greaseweazle #295: Question about dump low-density floppies with high-density floppy drives

OldMadMan, 2023-04-08:
Using the latest GW Tools version 1.10, I tried to dump IBM DOS 3.3 floppy (5.25 inch, 360KB) with my TEAC FD-55GFR (a high density 5.25" floppy drive). The command I used is:   **gw read --drive A --format ibm.360 DOS.img**

However GW only correctly dump tracks 0.0 and 0.1. All tracks since track 1.0 cannot be recognized. This made me very confused, and then I noticed that a line of GW prompts appeared on the screen during the crawl: Reading c=0-39:h=0-1 revs=2.

So I wonder if GW didn't add a parameter needed to read low-density floppy disks on high-density floppy drives, that is, the magnetic head moves two grids at a time. So I used this command: gw read --drive A --format ibm.360 --tracks="c=0-39:h=0-1:step=2" DOS.img and it works.

I would like to ask the reason why GW did not add the parameter step=xx by default. Is it because some users may be using low-density floppy drives? If yes, maybe you can consider adding a floppy drive density option at a later date? 

Thank you.

keirf, 2023-04-08:
It's not specifically a density issue but a 40-track Vs 80-track (96tpi Vs 48tpi) issue. And you have an option to deal with it. `step=2`. You found it. Well done. :-)

keirf, 2023-04-08:
It's not added by default because as you note some users may use a 48tpi drive. Also it's simply my model of usage that the tracks are specified in terms of "native" track numbering (be that 48 or 96 tpi) and then subsequently modified for physical track numbering via modifier parameters including step=

closed 2023-04-08 by keirf