keirf/greaseweazle #301: Add support for MSX disk formats

sdsnatcher73, 2023-04-23:
For MSX disk images I currently use atarist.360 and atarist.720. While this works MSX does support more formats (40 track based) and also I would suspect MSX format is more like IBM's (given Microsoft's involvement). I would propose the following to diskdefs.cfg. I am not sure about iam, cskew and hskew used in atarist formats. 

`disk msx.1d
    cyls = 40
    heads = 1
    tracks * ibm.mfm
        secs = 9
        bps = 512
        gap3 = 84
        rate = 250
    end
end

disk msx.1dd
    cyls = 80
    heads = 1
    tracks * ibm.mfm
        secs = 9
        bps = 512
        gap3 = 84
        rate = 250
    end
end

disk msx.2d
    cyls = 40
    heads = 2
    tracks * ibm.mfm
        secs = 9
        bps = 512
        gap3 = 84
        rate = 250
    end
end

disk msx.2dd
    cyls = 80
    heads = 2
    tracks * ibm.mfm
        secs = 9
        bps = 512
        gap3 = 84
        rate = 250
    end
end`

I will create some flux dumps of freshly formatted disks (I will gw erase them prior to formatting on real hardware). I am not sure I have a machine that supports the 40 track format but will check.

sdsnatcher73, 2023-04-23:
[msx1dd.zip](https://github.com/keirf/greaseweazle/files/11303119/msx1dd.zip)
[msx2dd.zip](https://github.com/keirf/greaseweazle/files/11303120/msx2dd.zip)

sdsnatcher73, 2023-04-23:
Found a National FS-5500 which supports the 40 track option during formatting (all MSX's can read such disks afterwards). When issueing the call fornat command the following is printed on screen:

[code]Drive name? (A,B)
1 - 1 side
2 - 2 sides
3 - 1 side,  double track
4 - 2 sides, double track[/code]

The following two flux dumps are from option 1 (msx1d.scp) and 2 (msx2d.scp). The 3.5" HD drive seems to double step automatically upon reading (so I dumped cylinders 0 to 41 for both). Or do you think the drives are not double stepping (and only cylinders 0-39 are used)? That would seem a bit odd though...

[msx1d.zip](https://github.com/keirf/greaseweazle/files/11303485/msx1d.zip)
[msx2dd.zip](https://github.com/keirf/greaseweazle/files/11303484/msx2dd.zip)

keirf, 2023-04-24:
There's no interleave or skew. So your config lines are correct.

commit f3f8da08 (2023-04-24): diskdefs: For MSX systems.

Fixes #301
closed 2023-04-24 by keirf