keirf/greaseweazle #307: PLL blanking for FM/MFM disks

tdaede, 2023-05-03:
The PLL described in the upd765a supports blanking, where the PLL state is locked immediately after index and data sections. This is to avoid the expected discontinuity from moving the PLL frequency too much while still allowing a fast PLL response.

http://bitsavers.org/components/nec/_appNotes/AN08_uPD765_preliminary_197902.pdf

This is presumably mostly an issue with rewritten disks, not factory written disks. In fact doing it may make decode of factory / duplicator created disks worse.

It would be nice to implement this in Greaseweazle for higher quality disk recovery. The main difficulty is that bit cell assignment is done in one pass, followed by decode by `import_flux_data`, this would have to be changed to be incremental.

keirf, 2023-05-03:
The Greaseweazle PLL isn't actually an authentic emulation of a real PLL. I doubt it has trouble locking back on across the length of a pre-sync header, or that it is greatly affected by the few illegal clock bits left by turning off the write gate.

Making the decode incremental, or teaching the decoder somehow about regions, would be pretty invasive!

tdaede, 2023-07-12:
I'm convinced this is probably not worth the effort. Closing.

closed 2023-07-12 by tdaede