keirf/greaseweazle #309: Write problem - 2.88 special formatting

Olivier-ADLER, 2023-05-05:
I did a test with a 2.88 floppy formatted near 4 MB with 2MGUI. This formatting utility use the full track to store data and reset the controller just before the start of the track to avoid overwriting this area. The default gap between the end of the track and the start is 20. I was able to read the floppy with a GW raw read to an SCP file, and write it back. But the resulting floppy is not readable anymore, on the same PC and same floppy drive. 

I'm not sure if the problem does come from the dump, or from the write.

Dumping and writing a normal 2.88 Dos 6.22 formatted floppy is working. This show that the 2.88 drive and PC bios works correctly.
[2mgui2881_of_1.zip](https://github.com/keirf/greaseweazle/files/11408801/2mgui2881_of_1.zip)

This archive is a 7z one, please change the extension after download. I was not able to get a small enough file size with .zip.

Olivier-ADLER, 2023-05-05:
Here is the 2M / 2MGUI documentation converted to pdf.
[Manuel.pdf](https://github.com/keirf/greaseweazle/files/11408836/Manuel.pdf)

And the 2mgui bios extender to support this enhanced formatting.

[2mgui19.zip](https://github.com/keirf/greaseweazle/files/11408861/2mgui19.zip)

keirf, 2023-05-05:
It could be due to data crossing the index boundary, depending on how the disk is formatted. Or the index sensor on the dump drive being slightly different position to on the drive that originally wrote the disk. Since SCP image is written index to index by "gw write".

keirf, 2023-05-05:
Since you have a fixed version of HxC try visualising the disk and see where the data lies on each track, in the dump.

Olivier-ADLER, 2023-05-06:
Here is what i get inside the HxC analyzer. The full track is not decoded because it's a special format without sectors but we can see the start of each track. The shift between each track start is normal, it is done like this by 2mgui formatter to speedup readings.

[image]

The same drive was used to format through 2mgui by a normal 2.88 compatible floppy controller, and to dump through GW.exe.

I'm going to make a try with a longer gap between the end and the start of each track to give more room for timing shifts.

Here is the modified HxC dll to read 2.88 archives.

[libhxcfe.zip](https://github.com/keirf/greaseweazle/files/11412346/libhxcfe.zip)

keirf, 2023-05-06:
The spiral pattern shows it is not index aligned and in fact that pattern is probably for efficient track to track access. Hence data will cross the index mark on some tracks and since Greaseweazle is ending the write at index, data at that point will be corrupted at that write-overlap point. This can only be fixed by analysing the format and writing with knowledge of the splice point. Am I likely to do that for this esoteric format? Perhaps not.

Olivier-ADLER, 2023-05-06:
I understand. This mean that the dump of such a disk should be correct. Only a write will be problematic ?

Is it really necessary to decode the format to be able to write it ? Would a modification that give the capability to write a track after the index (-WriteAfterIndex option) be sufficient ? After thinking to that a bit more it is probably necessary as well to detect the start position of each track during the dump process.

Writing them is probably only a curiosity, but decoding the format to get back files is perhaps more useful if such disks are found, specially for 1.44M or 720k format where 2MGUI has probably been used.

I would be curious to know if Pauline or another system is actually able to write this format.

keirf, 2023-05-06:
If we can read them to retrieve data then writing back isn't a problem as we know the track format. As you say the problem is to know where the track gap is in the dump (for writing back). Perhaps this could be done with tricks that stop short of full track decode. Some tools will attempt this but Greaseweazle currently does not. I don't know if it can be done reliably.

Olivier-ADLER, 2023-05-06:
As soon as the track gap is large enough to accommodate for the drive speed variations, this format seems reliable when writing it with a standard controller. I tried a few files copy in both directions without any error, on 2.88 floppy. On 1.44 and 720k it should be a bit more reliable. The default gap value is 20. There is a test routine in 2mgui that gives the minimum possible track gap for a given drive. It is 2 on my 2.88 drive.

closed 2023-05-10 by keirf