keirf/greaseweazle #310: Getting crash on ED disk image read trying to create HFE disk image

drencorxeen, 2023-05-05:
OS: Windows 11
Command used:  .\gw.exe read gotektest.hfe::bitrate=1000
Output:

T81.0: Raw Flux (535280 flux in 698.13ms)
T81.1: Raw Flux (543581 flux in 707.63ms)
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 135, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\console.py", line 18, in run
  File "gw.py", line 3, in <module>
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\cli.py", line 69, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\tools\read.py", line 245, in main
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\image\image.py", line 29, in __exit__
  File "C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\greaseweazle\image\hfe.py", line 304, in get_image
struct.error: ushort format requires 0 <= number <= 0xffff

keirf, 2023-05-06:
HFE track length field is too short to represent ED tracks. The error message could be nicer I agree!

drencorxeen, 2023-05-06:
Ah, well that explains it then.

What is the best disk image format to use to save all track data for ED based disks then?

Thank you

keirf, 2023-05-06:
Either SCP or Kryoflux raw files.

commit 6a58e7fb (2023-05-10): hfe: Better error message when a track is too long to fit in HFE.
Fixes #310
closed 2023-05-10 by keirf