keirf/greaseweazle #315: Handle duplicate sectors in D88 images

sdsnatcher73, 2023-05-10:
Several FM-77 D88 images have duplicated sectors (the origin seems to lie in the dumping tool used way back when). Currently gw cannot write these images directly to disk as it does not expect sectors to be duplicated (well that is what I assume is happening). The current workaround is to use Simon Owen's samdisk to strip the duplicated sectors and then write the image.

Here is an image that has this issue and a copy where the duplicate sectors have been removed:
[Ys II - Ancient Ys Vanished - The Final Chapter (1988)(Falcom) (Program Disk).d88.zip](https://github.com/keirf/greaseweazle/files/11445443/Ys.II.-.Ancient.Ys.Vanished.-.The.Final.Chapter.1988.Falcom.Program.Disk.d88.zip)
[Ys II - Ancient Ys Vanished - The Final Chapter (1988)(Falcom) (Program Disk)[Duplicate Sectors Removed].d88.zip](https://github.com/keirf/greaseweazle/files/11445444/Ys.II.-.Ancient.Ys.Vanished.-.The.Final.Chapter.1988.Falcom.Program.Disk.Duplicate.Sectors.Removed.d88.zip)

Here is the samdisk command used and it's output:
`samdisk copy Ys\ II\ -\ Ancient\ Ys\ Vanished\ -\ The\ Final\ Chapter\ \(1988\)\(Falcom\)\ \(Program\ Disk\).d88 Ys\ II\ -\ Ancient\ Ys\ Vanished\ -\ The\ Final\ Chapter\ \(1988\)\(Falcom\)\ \(Program\ Disk\)\[Duplicate\ Sectors\ Removed\].d88
Fixed: removed 12 duplicate sectors from oversized cyl 4 head 0
Fixed: removed 8 duplicate sectors from oversized cyl 8 head 0
Fixed: removed 4 duplicate sectors from oversized cyl 13 head 0
Fixed: removed 4 duplicate sectors from oversized cyl 15 head 0
Fixed: removed 8 duplicate sectors from oversized cyl 17 head 0
Fixed: removed 4 duplicate sectors from oversized cyl 21 head 0
Fixed: removed 4 duplicate sectors from oversized cyl 24 head 0
Fixed: removed 4 duplicate sectors from oversized cyl 28 head 0
Fixed: removed 8 duplicate sectors from oversized cyl 29 head 0
Fixed: removed 8 duplicate sectors from oversized cyl 33 head 0
Fixed: removed 8 duplicate sectors from oversized cyl 34 head 0
Fixed: removed 4 duplicate sectors from oversized cyl 35 head 0
Fixed: removed 1 duplicate sector from oversized cyl 39 head 0
Fixed: removed 7 duplicate sectors from oversized cyl 0 head 1
Fixed: removed 12 duplicate sectors from oversized cyl 1 head 1
Fixed: removed 8 duplicate sectors from oversized cyl 10 head 1
Fixed: removed 8 duplicate sectors from oversized cyl 11 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 16 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 17 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 18 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 20 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 22 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 27 head 1
Fixed: removed 7 duplicate sectors from oversized cyl 29 head 1
Fixed: removed 8 duplicate sectors from oversized cyl 30 head 1
Fixed: removed 8 duplicate sectors from oversized cyl 32 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 33 head 1
Fixed: removed 4 duplicate sectors from oversized cyl 34 head 1
`

keirf, 2023-05-11:
I have fixed this issue on branch `issue-315`. The result for this disk looks sane.

Can you please test?

keirf, 2023-05-11:
Well the logic in samdisk is only to remove duplicates if the track is oversized. So a warning and then de-duplication makes sense. I could then finally report the final track size I suppose, or report that it's no longer oversized?

sdsnatcher73, 2023-05-11:
Last one makes most sense, no need to warn about something that is remedied automatically. If the track is still oversized after deduplication a warning is of course in place.

keirf, 2023-05-11:
There you go! This is all now in `master`.

closed 2023-05-11 by keirf