keirf/greaseweazle #316: RAW dump not written back correctly

sdsnatcher73, 2023-05-11:
Now I am not entirely sure if this is an issue in gw host tools but am reporting it as it may be.
;:
I found a raw flux dump of the game 4th Unit 2:
[4th Unit 2, The (1988)(Datawest)(JP)(Disk 1 of 3)(A) (raw flux).zip](https://github.com/keirf/greaseweazle/files/11448507/4th.Unit.2.The.1988.Datawest.JP.Disk.1.of.3.A.raw.flux.zip)

The disks were dumped (with KryoFlux I assume) as 80 track media (82 tracks total dump).  FM77 media are mostly 40 tracks though, as is this game. If I write the flux image as is the game won't boot on real hardware. If I convert the raw image to scp (keeping the 82 tracks the result is the same when written back to disk, not booting (which makes sense as no data should be lost):

gw erase (erasing all 82 tracks on the media, just to be sure)
gw write 4thunit2_1_00_0.raw (does not boot)

gw convert 4thunit2_1_00_0.raw 4thunit2_1_00_0.scp
gw erase
gw write 4thunit2_1_00_0.scp (does not boot)

I then tried to remove the odd tracks by running:
gw convert 4thunit2_1_00_0.scp 4thunit2_1_40.scp --tracks c=0-39:step=2

If I then issue
gw erase
gw write 4thunit2_1_40.scp (still won't boot)

I then converted this 40 track scp to d88 using HxC Floppy Emulator tools:
[4thunit2_1.d88.zip](https://github.com/keirf/greaseweazle/files/11448860/4thunit2_1.d88.zip)

If I write this to disk it does boot (miraculously imho)...

keirf, 2023-05-11:
There's nothing that exciting about this image it's just that it's a raw dump of a user-written disk and perhaps it's not quite clean enough to reliably write back raw.

What perhaps GW needs is a "generic ibm" mode where it will scan and recover sectors. Or even a raw mode where it will scan for and regenerate bitcells at a specified bitrate (ie. basically clean up a raw track through Greaseweazle's PLL clock recovery). These are two things that HxC software does make available to users.

keirf, 2023-05-11:
It's very simple to add a "raw" codec so I'll do that and we can see if it helps here.

keirf, 2023-05-11:
I have added the codec to branch `issue-316`. Could you give it a go? For example:
```
gw convert 4thunit2_1_00_0.scp 4thunit2_1_40_clean.scp --format raw.250 --tracks c=0-39
```
And see if that boots!

sdsnatcher73, 2023-05-11:
This does seem to have a positive effect but not every write turns out in a positive boot (media age is an issue possibly). Will do some more tests with different disks to make sure.

keirf, 2023-05-11:
One issue is that there is of course no verification. So on marginal media there will be no retries on the write. It's another option now anyway. Better still will be to scrape the sectors and properly regenerate the track, with verification on write.

closed 2023-05-11 by keirf

sdsnatcher73, 2023-05-11:
> Better still will be to scrape the sectors and properly regenerate the track, with verification on write.
Would it make sense to add D88 image file output support to make this possible?

keirf, 2023-05-11:
Yes part of this would be adding D88 output support. It would also naturally be added to the existing IMD output support. It's a fairly different to how Greaseweazle tools currently work. Needs a bit of thought.

commit 8c4efc16 (2023-05-11): gw: New raw codec and diskdefs raw.{125,250,500}.

Can be used to clean up index-aligned dirty raw tracks.

Fixes #316