keirf/greaseweazle #318: Problems running gw on macOS Ventura

Burrito78, 2023-05-12:
Hi, 

i installed and it throws Python errors. I'm stumped, sorry.

Here is the installation:
```
alex@Minivonlexander ~ % python3 -m pip install git+https://github.com/keirf/greaseweazle@latest
Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/keirf/greaseweazle@latest
  Cloning https://github.com/keirf/greaseweazle (to revision latest) to /private/var/folders/y1/xqj4xsgx111dtsr194gjw3540000gn/T/pip-req-build-hb8e3dq6
  Running command git clone --filter=blob:none --quiet https://github.com/keirf/greaseweazle /private/var/folders/y1/xqj4xsgx111dtsr194gjw3540000gn/T/pip-req-build-hb8e3dq6
  Running command git checkout -q 03280074da38cd488238b20e2a2c0dce9aeda835
  Resolved https://github.com/keirf/greaseweazle to commit 03280074da38cd488238b20e2a2c0dce9aeda835
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: crcmod in ./Library/Python/3.9/lib/python/site-packages (from greaseweazle==1.11) (1.7)
Requirement already satisfied: bitarray in ./Library/Python/3.9/lib/python/site-packages (from greaseweazle==1.11) (2.7.3)
Requirement already satisfied: pyserial in ./Library/Python/3.9/lib/python/site-packages (from greaseweazle==1.11) (3.5)
Requirement already satisfied: requests in ./Library/Python/3.9/lib/python/site-packages (from greaseweazle==1.11) (2.30.0)
Requirement already satisfied: charset-normalizer<4,>=2 in ./Library/Python/3.9/lib/python/site-packages (from requests->greaseweazle==1.11) (3.1.0)
Requirement already satisfied: idna<4,>=2.5 in ./Library/Python/3.9/lib/python/site-packages (from requests->greaseweazle==1.11) (3.4)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./Library/Python/3.9/lib/python/site-packages (from requests->greaseweazle==1.11) (2.0.2)
Requirement already satisfied: certifi>=2017.4.17 in ./Library/Python/3.9/lib/python/site-packages (from requests->greaseweazle==1.11) (2023.5.7)
```

And here is the output when trying to run 'gw':
```
alex@Minivonlexander ~ % gw
Usage: /usr/local/bin/gw [--time] [action] [-h] ...
  --time      Print elapsed time after action is executed
  -h, --help  Show help message for specified action
Actions:
Traceback (most recent call last):
  File "/usr/local/bin/gw", line 8, in <module>
    sys.exit(main())
  File "/Users/alex/Library/Python/3.9/lib/python/site-packages/greaseweazle/cli.py", line 64, in main
    return usage(argv)
  File "/Users/alex/Library/Python/3.9/lib/python/site-packages/greaseweazle/cli.py", line 35, in usage
    mod = importlib.import_module('greaseweazle.tools.' + a)
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/alex/Library/Python/3.9/lib/python/site-packages/greaseweazle/tools/info.py", line 12, in <module>
    import requests, re
  File "/Users/alex/Library/Python/3.9/lib/python/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
  File "/Users/alex/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py", line 38, in <module>
    raise ImportError(
ImportError: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with LibreSSL 2.8.3. See: https://github.com/urllib3/urllib3/issues/2168
```

keirf, 2023-05-12:
I think this is a recent packaging problem with Python. I could find the error message in Stackoverflow, and there are a few suggestions there for how to work around it. https://stackoverflow.com/questions/76187256/importerror-urllib3-v2-0-only-supports-openssl-1-1-1-currently-the-ssl-modu

Another way to work around would be to install pyenv and from that install a clean Python env of a different version (eg. 3.8) and use that for Greaseweazle. But I think the Stackoverflow suggestions linked above will probably get you going.

keirf, 2023-05-12:
The suggested line to try is a downgrade of urllib as follows:
```
pip install urllib3==1.26.6 
```
You may need to run `pip3` instead of `pip`. LEt me know how it goes. I may need to add this to the software installation hints for MacOS for a while.

keirf, 2023-05-12:
I tested this on Linux and probably the correct way to downgrade to latest urllib 1.x is as follows:
```
pip install "urllib3<2.0"
```

Burrito78, 2023-05-12:
@keirf 

```
alex@Minivonlexander / % pip install urllib3==1.26.6 
zsh: command not found: pip
```
macOS is driving me crazy today, sorry.
I think i need so set a permanent path in the PATH variable to the location where gw and pip reside.

Could you poste the necessary command(s) please?

keirf, 2023-05-12:
Skip the stupid `pip` wrapper actually, and try this:
```
python3 -m pip install "urllib<2"
```

You should see it uninstall urllib 2.0.2 and replace it with urllib 1.26.15. The quotes are simply to escape the `<` character which otherwise gets interpreted by the shell.

keirf, 2023-05-12:
As for commands on your path, it looks from previous command lines you posted that `python3` and `gw` are both on your path? And you don't need `pip`. Actually it is safer to directly invoke `python3 -m pip` as you know you are invoking the correct Python env this way!

Python on Mac is known to be an experience. See https://www.explainxkcd.com/wiki/index.php/1987:_Python_Environment 

A cleaner setup could be pyenv + virtualenv. I use this for development so I have my own clean isolated Python install for Greaseweazle (on Ubuntu Linux). However, I'm pretty sure the urllib downgrade will get you going!

Burrito78, 2023-05-12:
@keirf 

I'm sorry, still not cooperating...

```
alex@Minivonlexander / % python3 -m pip install "urllib<2"
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement urllib<2 (from versions: none)
ERROR: No matching distribution found for urllib<2
alex@Minivonlexander / % 
```

keirf, 2023-05-12:
Argh, that's because I mistyped the command line. It should be `urllib3` not `urllib`.
Try this:
```
python3 -m pip install "urllib3<2"
```

Burrito78, 2023-05-12:
😃 

```
alex@Minivonlexander / % python3 -m pip install "urllib3<2"
Defaulting to user installation because normal site-packages is not writeable
Collecting urllib3<2
  Downloading urllib3-1.26.15-py2.py3-none-any.whl (140 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 140.9/140.9 kB 2.9 MB/s eta 0:00:00
Installing collected packages: urllib3
  Attempting uninstall: urllib3
    Found existing installation: urllib3 2.0.2
    Uninstalling urllib3-2.0.2:
      Successfully uninstalled urllib3-2.0.2
Successfully installed urllib3-1.26.15
alex@Minivonlexander / % gw
Usage: /usr/local/bin/gw [--time] [action] [-h] ...
  --time      Print elapsed time after action is executed
  -h, --help  Show help message for specified action
Actions:
  info        Display information about the Greaseweazle setup.
  read        Read a disk to the specified image file.
  write       Write a disk from the specified image file.
  convert     Convert between image formats.
  erase       Erase a disk.
  clean       Clean a drive in a zig-zag pattern using a cleaning disk.
  seek        Seek to the specified cylinder.
  delays      Display (and optionally modify) drive-delay parameters.
  update      Update the Greaseweazle device firmware to latest (or specified) version.
  pin         Change the setting of a user-modifiable interface pin.
  reset       Reset the Greaseweazle device to power-on default state.
  bandwidth   Report the available USB bandwidth for the Greaseweazle device.
  rpm         Measure RPM of drive spindle.
```

Thank you!

closed 2023-05-12 by keirf

keirf, 2023-05-12:
By the way I just moved the `latest` tag to the latest version v1.12. So you may want to install greaseweazle package again!