keirf/greaseweazle #324: Feature request:  Support for Commodore .d71 (1571) disks and images

TheRealMikeKnight, 2023-05-31:
Hello!  Just got my GreaseWeazle!  Awesome!

I was wondering if support for Commodore 1571 disk images (.d71) can be added please?  The 1571 GCR format is very similar to the 1541 GCR format, but it's double sided.

There is support in the GreaseWeazle for the .d64 to read and write both sides of a _flippy_ disk, surely it wouldn't be too much more complicated to add double-sided 1571 GCR (.d71) format to the awesome list of supported disk formats?  Pleaaasee???  Surely it would also bear some resemblance to the already available 1581 (.d81) format??  (Just guessing--what do I know?)

Thank you SO MUCH.  That would absolutely make GreaseWeazle the ultimate value to me.

TheRealMikeKnight, 2023-05-31:
WOW!  I can’t believe the rapid response and how eager you are to accommodate!  THANK YOU!

YES!  I would be happy to raw dump a 1571 GCR disk.  I am EXTREMELY new to greaseweazle, so for expedience, What command do I give the GreaseWeazle to do that for you (obviously gw read -–format commodore.1571 won’t work)?  What file extension should the output be?  Just .img?

Yes:  you are correct!  35 cylinders per side.  The disk layout is the same on both sides.

Of course, in the event I am wrong, here is the Commodore 1571 User Guide from Zimmers where it discusses the disk layout in detail:  `http://http://www.zimmers.net/anonftp/pub/cbm/manuals/drives/1571_users_Guide_252095-04_(1985_Aug).pdf` The technical stuff needed can be found on Pages 104-106.

Looking forward to hearing back from you! :)

THANK YOU!!

keirf, 2023-05-31:
Just do `gw read --tracks=c=40 a.scp` which will read 40 cylinders to an SCP image file. If you are reading on a 96tpi 80 cyl drive then you will need `--tracks=c=40:step=2`

TheRealMikeKnight, 2023-05-31:
OK I tried sending the attached SCP file via email.  Were you able to receive it?

Thanks :)

keirf, 2023-05-31:
I haven't received an email. Can you upload to a cloud drive such as Google Drive or Dropbox etc and share a link?

keirf, 2023-05-31:
GitHub strips attachments from emails but another alternative is to zip the file and then attach it here using the GitHub web interface.

TheRealMikeKnight, 2023-05-31:
[C1571SCP.zip](https://github.com/keirf/greaseweazle/files/11618169/C1571SCP.zip)

Ahhh very good!  Thanks! :)

TheRealMikeKnight, 2023-05-31:
I'm on it!

TheRealMikeKnight, 2023-05-31:
OMG!!  My old 360K PC Floppy disk drive can write Commodore 1571 double-sided diskettes and read them to .d71 format!!

WOW!!!  You are AMAZING!!

THANK YOU!!!!

I'll bet you are 100% the ONLY person in the world to have achieved this!

--Even WITHIN Commodore software (running on a C128) I can think of only TWO titles that can convert a .d71 to a full 1571 double sided GCR image, but you need a 1581 drive to hold the .d71 file AND a 1571 drive to accomplish this.  I have YET to see the REU used as a RAMdisk to achieve this.

WAY TO GO!  Congratulations!

If you wish to add more functionality, I can provide you many more CP/M MFM 5 1/4" disk images in various formats (I think 100+ e.g. Coleco ADAM, Microbee, SuperBrain, Epson, MAXI, NEC, Olivetti, TRS, Zenith ... MANY more)

Even if not, THANK YOU!!!!!  YOU ROCK!

keirf, 2023-06-01:
Great to hear! Adding more formats ia always good. Also I will add a generic IBM format handler soon which will scrape up all sectors it can find on a track, rather than necessarily needing to know the exact track layout ahead of time. Although the latter is better for error checking.

closed 2023-06-01 by keirf