keirf/greaseweazle #325: Decoder for Format commodore.1541 fails with some correct flux images

markusC64, 2023-06-03:
There are some valid commdore 1541 images that decode without any error using g64conv but have many decoding errors using greaseweazle tools.

I am attaching one:

[_scienceIII.7z.zip](https://github.com/keirf/greaseweazle/files/11643257/_scienceIII.7z.zip)

I will post a description how to decode that properly soon, I have to translate the document from german to english first, you see.

gw convert _scienceIII.scp 1.d64

Converting c=0-39:h=0 -> c=0-39:h=0
Format commodore.1541
T0.0: Commodore 1541 GCR (16/21 sectors) from Raw Flux (107826 flux in 500.00ms)
T1.0: Commodore 1541 GCR (19/21 sectors) from Raw Flux (109068 flux in 500.00ms)
T2.0: Commodore 1541 GCR (20/21 sectors) from Raw Flux (107364 flux in 500.00ms)
T3.0: Commodore 1541 GCR (17/21 sectors) from Raw Flux (107562 flux in 500.00ms)
T4.0: Commodore 1541 GCR (20/21 sectors) from Raw Flux (107076 flux in 500.00ms)
T5.0: Commodore 1541 GCR (20/21 sectors) from Raw Flux (106716 flux in 500.00ms)
T6.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (105624 flux in 500.00ms)
T7.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (105780 flux in 500.00ms)
T8.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (106896 flux in 500.00ms)
T9.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (106860 flux in 500.00ms)
T10.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (106374 flux in 500.00ms)
T11.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (107148 flux in 500.00ms)
T12.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (106872 flux in 500.00ms)
T13.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (107274 flux in 500.00ms)
T14.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (106476 flux in 500.00ms)
T15.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (107220 flux in 500.00ms)
T16.0: Commodore 1541 GCR (21/21 sectors) from Raw Flux (106026 flux in 500.00ms)
T17.0: Commodore 1541 GCR (19/19 sectors) from Raw Flux (72090 flux in 500.00ms)
T18.0: Commodore 1541 GCR (14/19 sectors) from Raw Flux (95370 flux in 500.00ms)
T19.0: Commodore 1541 GCR (2/19 sectors) from Raw Flux (95058 flux in 500.00ms)
T20.0: Commodore 1541 GCR (2/19 sectors) from Raw Flux (95754 flux in 500.00ms)
T21.0: Commodore 1541 GCR (5/19 sectors) from Raw Flux (96054 flux in 500.00ms)
T22.0: Commodore 1541 GCR (6/19 sectors) from Raw Flux (96162 flux in 500.00ms)
T23.0: Commodore 1541 GCR (12/19 sectors) from Raw Flux (94212 flux in 500.00ms)
T24.0: Commodore 1541 GCR (2/18 sectors) from Raw Flux (90000 flux in 500.00ms)
T25.0: Commodore 1541 GCR (11/18 sectors) from Raw Flux (90900 flux in 500.00ms)
T26.0: Commodore 1541 GCR (5/18 sectors) from Raw Flux (91848 flux in 500.00ms)
T27.0: Commodore 1541 GCR (7/18 sectors) from Raw Flux (91722 flux in 500.00ms)
T28.0: Commodore 1541 GCR (10/18 sectors) from Raw Flux (90450 flux in 500.00ms)
T29.0: Commodore 1541 GCR (10/18 sectors) from Raw Flux (90642 flux in 500.00ms)
T30.0: Commodore 1541 GCR (6/17 sectors) from Raw Flux (86088 flux in 500.00ms)
T31.0: Commodore 1541 GCR (6/17 sectors) from Raw Flux (86376 flux in 500.00ms)
T32.0: Commodore 1541 GCR (9/17 sectors) from Raw Flux (88104 flux in 500.00ms)
T33.0: Commodore 1541 GCR (10/17 sectors) from Raw Flux (86178 flux in 500.00ms)
T34.0: Commodore 1541 GCR (14/17 sectors) from Raw Flux (85632 flux in 500.00ms)
Cyl-> 0         1         2         3
H. S: 0123456789012345678901234567890123456789
0. 0: ..................XXX.X.X..XXXXX.X.
0. 1: ...................XX.XXX..X..XXXX.
0. 2: X..................XXXXXX.X..XX.XXX
0. 3: X.................XXX..XXXX..XX...X
0. 4: ...................XXXX.X.XXX...X..
0. 5: X.................X.XXX.XXXX..XX..X
0. 6: ...................XX.X.X.....XX...
0. 7: ...................XXXX.X..XXXXX.X.
0. 8: ...XX...............XX..X.XXXXXX.X.
0. 9: .X...X.............XX...XXX...XX...
0.10: ...................XXX..X.XXXXXXXX.
0.11: ...................XXX.XXXXXX...X..
0.12: ...................XXXX.X.XXXX.XXX.
0.13: X..X..............XXXX...XX..X.X...
0.14: .X................XXXXXX.XXX.......
0.15: ..XX...............X.XXXX.X...X.X..
0.16: ...................XXXXXXX..X..XX..
0.17: ...................XXXX.X.XX..
0.18: X..X...............X.XX.
0.19: .................
0.20: .................
Found 493 sectors of 683 (72%)

Oh well, good decoders can decode all 683 sectors of that image.

Some background: That disk has been read from a real disk multiple times and then the best rotation for each track has been combined into this image.

markusC64, 2023-06-03:
I hope I have translated the description properly.

[Decoding flux with a 1541.pdf](https://github.com/keirf/greaseweazle/files/11643305/Decoding.flux.with.a.1541.pdf)

keirf, 2023-06-03:
Thanks. Best guess is that the Greaseweazle PLL (or flux to bitcell converter) could be better tuned for GCR or even specifically for 1541 GCR. It is working okay for a range of 1541 cbmdos disks. Is this one just a tricky / marginal cbmdos disk or is it otherwise special in any other way (different speedzones etc)?

markusC64, 2023-06-03:
All standard speedzones, no copy protection at all.

No idea of those disks are mastered that way or if that is an efefct of its age.

The "preservation64.de" project has observed that on multiple disks. I decided that the public domain disk is a good example to post. Yes, that's a disk of commodore's public domain series - so no copyright problem at all.

Knowing how a correct decoder works (see the above pdf), I have created 2 artifical test flux images, which are successively more difficult to decode. The first one is already an unsolvable problem for GreaseWeazle - not a single sector is decoded. The second one makes even higher demands on the correctness of the decoder.

I can post them if you wish.

markusC64, 2023-06-03:
And of cause I can provide reference D64 (and G64) for Science III.

keirf, 2023-06-03:
Artificial images could be interesting too. But I suppose to start with the real provided image should yield interesting info by itself.

I do doubt that Greaseweazle will ever match third party tools for C64 analysis such as g64conc or DTC. A realistic aim for Greaseweazle is to support writing g64 images to disk with verification.

That said, cbmdos disks should certainly be fixable!

markusC64, 2023-06-03:
For sure.
[sc3.hardtodecode.zip](https://github.com/keirf/greaseweazle/files/11643521/sc3.hardtodecode.zip)

Note that this is from speedzones and decoded bitstream a completely standard disk. All is done to add some (hard) or even more (hardest) flux that the flux to bitstream decoding hardware on a 1541, 1570 and 1571 will ignore at all.

Thus the images test the flux to bitcell converter without using non standard speed and without using non standard bitcell content.

keirf, 2023-06-03:
Okay, perhaps to handle some copy protections weird flux reading/writing would be required (or should I say, a layer which closely emulates a 1541 could be required). However, I'm pretty confident from a quick look that the problems with `_scienceIII` are not at the flux level. If I look at the stream view in HxC software's track analyzer, the bit bands are pretty tight on where they should be (looking at eg, Track 0). Yet 5 sectors are inexplicably missed by `gw`.

Or track 19, where only 2 of 9 sectors are found. Again the bit bands are on point and tight (enough).

So.... it must be a bug or weakness in my sync detection or sector parsing, I reckon. I will have a closer look tomorrow!

markusC64, 2023-06-03:
If you cannot find something, you can save the output of the flux to bitcell decoder and then we can compare with a good decoding.

Its the same with that real image as with my "hard" from hardToDecode samples, but far less often that extra flux is there. That image applies the pattern observed in the pdf and in the science III image at every possible position. And the "hardest" image applies another pattern - so that I do not see any possibility to insert more flux without changing bitstream.

BTW:
rem to save time for testing, I have done it with only few tracks:
gw convert _scienceIII.scp _sc.scp --tracks c=0-2
g64conv.pl _sc.scp 1.txt rawUnpadded

will give you the output of the flux to bitstream decoder:
0 is a zero bit,
1 is a one bit,
_ is where the decoder has consumed a flux (just informational, not for the bitstream)

so you can compare.

keirf, 2023-06-04:
Ah yes, I see the noise in `_scienceIII` now. The pixels were too small for me to see in HxC tools on a hidpi display.

It doesn't look mastered that way though, right? The out-of-band fluxes look random. The kind of thing you see when the drive read-side logic doesn't like the bit timings. Perhaps these have a pattern though which the 1541 logic does happen to filter out.

keirf, 2023-06-04:
Okay a higher-level view of this which is not mired in modelling 1541 flip flops: I get a perfect conversion if I merge all short pulses into their adjacents. That is, short pulses break up longer pulses. Quite what the threshold should be for "short" in general I'm not sure: But obviously 2.5us works well for C64 (at 300rpm, though the dumps you provided are at 360rpm, but I resample them to 300rpm before I do anything else).

This does seem like something that may be more generally applicable too. I'll have to have a think about that!

markusC64, 2023-06-04:
Yes, occording to all avaiable sources, this is the pattern that a real 1541 logic filters out.
The length is not a valid citeria: More short flux aren't removed completely, see the algorithm the 1541 uses in hardware. That is applied in the hardest artificial example... 
Open question: How does a Commodore 8050 resp. 8250 resp SFD 1001 does that "filtering"? I guess the same with another fequency (not 16 MHz) and another Threshold... The bitcell sizes of those floppies doesn't match 16 MHz, so the clock has to be different, that's quite clear.

keirf, 2023-06-04:
Can you please test the `c64_lowpass` branch? There is an automated build also in progress here: https://github.com/keirf/greaseweazle/actions/runs/5168146338

...but I assume you prefer to run from source. This branch works on all your images including hardest. It could do with some more testing on difficult real images. :)

markusC64, 2023-06-04:
Congratulations. On the special discipline of decoding flux to bitstream with additional flux that a 1541/70/71 ignores, you've overtaken dtc.exe. Thus greaseweazle is not better in reading non protected standard 1541 disks into a d64 file compared to dtc.exe.

I can confirm that science III not only decodes without any error, the decoded image also matches the reference image.
And I can confirm the same is true for the original kryoflux image that has some revolutions with reading errors - in that case other revolutions of the same track are okay. It also decodes correctly.

I think I'll test some images that have a bitcell size with certain properties: A 1541/70/71 will read it using standard speedzones correctly, but the real bitcell size is only nearby.

keirf, 2023-06-04:
Reading protected 1541 disks *well* will need protection codecs in Greaseweazle, presumably. Which is possible. It's been suggested that at least Vorpal is nice to have.

markusC64, 2023-06-04:
It depends. Some of the copy protections have every single sector 1541 readable, so building a d64 is possible. The otrher question is for what that d64 is good for. Well, it can at least tell you that the read was successfull with high probabilty - and then use the flux stream for anything else.

In dtc.exe you can add a d64 decoder without saving the d64 even for those copyprotected images. It has the advantage that if decodes fails, a reread of that track is done. A nice feature...

So depending on the copy protection even d64 decoding can help..

markusC64, 2023-06-04:
While testing existing flux images, one question arises: Can I decode the back side of a kryoflux dump with greaseweazle for commodore.1541 format? The dumps have been made with a modified drive, so the back side contains all tracks.

hswap will not do the trick, because the flux also has to be reversed...

keirf, 2023-06-04:
Currently you cannot, because GW has no option to reverse the flux. I could add this.

keirf, 2023-06-04:
By the way, the dumps with short pulses needing to be filtered out: Is it possible this was due to incorrect density selection on the disk drive? 5.25 inch drives often need the correct setting at pin 2, which can be set via `gw pin set 2 L`. Perhaps that would have made better dumps that would not need filtering? Or was this considered already?

markusC64, 2023-06-04:
Not impossible, but improbable. But cannot tell further, I haven't made all the dumps I use for testing. But yoú see, from dozens of dumps (no, even more) just a minority have this issue - and often only on positions where the 1541 filters them out.

Okay, Heureca disks (known to vary speed) and german Geos 2.5 disks (front side) decode also correctly. You see, the mastered original Geos 2.5 disks vary speed within the tolerance area of the 1541 - probably just another copy protection so that burstnibbler does not really know where to short the track in order to be able to write it.

keirf, 2023-06-04:
Can we ask those who made the dump(s) whether pin 2 setting was considered? And whether they would do a new dump of a problematic disk with `gw pin set 2 L`?

keirf, 2023-06-04:
Also, are the dumps widely sourced, or are they actually from one or two people consistently using the same drive setup and configuration for every disk?

markusC64, 2023-06-04:
> Can we ask those who made the dump(s) whether pin 2 setting was considered? And whether they would do a new dump of a problematic disk with `gw pin set 2 L`?
> 
> Because if the real problematic dumps are a symptom of incorrect setup, I would not bother to implement the 2.5us lowpass filter. Or at least not enable it by default. Better to have the drive set up correctly.

I have asked friends. They should know more.

> Also, are the dumps widely sourced, or are they actually from one or two people consistently using the same drive setup and configuration for every disk?

Well, not too many poeople haing a modified drive together with flux hardware (if they have kryoflux, greaseweazle or supercard pro should not make a differenmce).

https://preservation64.de/index.php?title=The_Mask_of_the_Sun (you cannot download due to missing password) is another example. Only 1 sector of the backside of the disk fails with greaseweazle 1.12, and it decodes well with the snapshot) - for that to test, I had to convert the image to a p64 image and that to scp in order to reverse the flux... So greaseweazle 1.12 resp. snapshot shows the same difference like dtc.exe resp. g64conv.

And yes, both sources have dumped several disks with identical configuration, most of the disks are fully okay and decode with dtc.exe fine (you see, the problem we're speaking about makes dtc.exe fail).
Of cause the track with the copy protection cannot be decoded to d64...

keirf, 2023-06-04:
They may make dtc.exe fail if the actual problem is that pin 2 is not configured correctly. This is just something to check. Maybe changing the pin 2 setting here makes no difference, or even makes things worse! We don't know unless we try.

markusC64, 2023-06-04:
Well, I have thought about

> Currently you cannot, because GW has no option to reverse the flux. I could add this.

Pros:
+ On the one hand, greaseweazle supports modified drives. 
+ In case of reading errors, it might be sucessfull to try the other head. Flipping a 1541 disk means that the front-side (which is on the head for the backside) has an offset of +8 (Halt-) Tracks, and since the 1541 does only use 35 Tracks, any Floppy Drive should be able to reach that. So even with a non modified drive, you may benefit for 1541 disks.

Cons:
- We know that they are not wide spread, therefore probably not used that much.

In Summary I tend to say: Well, my concolusion is that supporting it would be better.

markusC64, 2023-06-04:
> They may make dtc.exe fail if the actual problem is that pin 2 is not configured correctly. This is just something to check. Maybe changing the pin 2 setting here makes no difference, or even makes things worse! We don't know unless we try.

dd0 is the default in kryoflux :-)
Checking it is dealed with "I have asked friends. They should know more." from above... I hope they know the ones that dumped it.

keirf, 2023-06-04:
Huh ok. Well maybe both dd0 and dd1 have been covered then. :) But apparently it does vary by drive whether 0 or 1 is best.

MichaelHuth, 2023-06-04:
I am the one Markus refers to enigma in his summary PDF.

We run preservation64 and have also dumped the Science III disk.

Regarding the decoding of flux times to read bits (i.e. the bits shifted into the VIA chip in a 1541 drive):
One can do the decoding by the book, ignoring any drive electronics. This will work for about 80% of the flux images, which turns out to be impractical. Often there are data errors in the decoded stream, where an actual drive just reads fine.

In reality the drives have a "kind of low-pass" filter for flux changes. That prevents detecting short burst of fluxes. 
I think that the reason is to be more resilient to aged disks or bad disk material. If in the middle of a bit cell you have some magnetic pinning or small area where the magnetization direction is changed, you get two magnetic phase boundaries. If these are very close together the drive detects that as a single flux change. Such fluxes at the wrong time should ideally be ignored by the drive.

The 1541 drive has no actual low pass filter but uses a very simple approach where the initial flip flop has an output pulse width of 2.5 µs. In the PDF that is indicated by the graph with the read/green markings. Simulating the resulting behavior just by joining together flux bursts is not going to work.

I think that the algorithm shown in the PDF in Igor Pro and Perl code **can be taken as reference** for correct simulation of the electronics of a 1541 between the heads flux detection and the digital bit shifted. I has proven in thousands of converted flux based images to work reliable.

For the Science III image it is hard to say if the small flux intervals were there already on creation or appeared later when the disk aged. Though, a typical modified PC drive for archiving has a different drive electronics and it does see these small flux intervals.
To be precise: The 1541 sees these flux changes as well, but does not shift a bit for them.

Science III was archived by Aki Sivula, most likley with a calibrated and modified Mitsumi drive.

Some notes on the algorithm:
The decoding of the current flux depends obviously on the decoding state taken over from the previous flux change.  Thus, after decoding a full rotation I continue to decode until a certain number of flux changes lead to the same (already) decoded bits.
Currently I use as threshold `min(25, num_fluxes)`.

Here is the complete code:
[Flux_Module.zip](https://github.com/keirf/greaseweazle/files/11646351/Flux_Module.zip)

keirf, 2023-06-04:
Hi @MichaelHuth and thanks for your detailed response. I will take some time here to lay out my thoughts.

Firstly, I will limit the scope of this discussion to the kinds of artifacts seen in Science III. That is, a standard DOS disk which has unexpected extra samples in the raw flux stream from disk. It should be easy to agree that these were not deliberately introduced, and that this would have been mastered originally as a totally standard disk with standard speedzone bitcell timings.

So the question is, where do these artifacts come from? Based on my experience with other disk formats I would say I have seen this kind of noise added in two cases:
1. Double-density disk read by a drive expecting high-density data. Personally I have seen this writing DD to a 3.5" HD disk without covering the HD hole. I expect that incorrect setting of floppy interface on pin 2 of many 5.25" HD drives could have similar effect.
2. Longer than usual MFM 8us time between flux transitions causes drive to ramp the gain and introduce noise even in as short a gap as ~10us. I have seen this with Mac GCR, and I have also seen this with poorly chosen Amiga sync marks (Amiga can set a custom sync mark and some loaders use marks which disobey MFM DD bit timings).

If the cause here were (1) then the fix would be to toggle pin 2. It seems to me worth confirming this, though I suspect that this may be something you have already considered and have optimally configured anyway.

If the cause is (2) then indeed that should be handled by software. This would also explain why the 1541 flip-flop based filter was implemented. However I would disagree that this means software needs to ape the 1541 hardware implementation. There is more than one way to fix the symptoms of (2), and 1541 chose a simple flip-flop circuit because that made sense in 1980s-era hardware. We can be more straightforward in our approach in modern software, no need to emulate a 1541!

So that's my working assumption: it's (1) or (2). I lean towards (2) but would like to eliminate (1) as a possibility.

markusC64, 2023-06-05:
I think it is option (2). You see, When we analyzed what went wrong in decoding Science III, I found that decoding a one flowwoed by twozeroes went wrong, there are the extra flux inserted (of cause tweo flux, you know that the number of flux on the track has to be even since every flux is a polarity change).

Two zeroes means a noticable gap. And Science III confirms that by the fact that on higher tracks, where the speed is lower and therefore the gap is longer, more extra flux are observed. Note that proper GCR is encoded such that that there are never more than two zeros in a row, so that's already the worst case to expect for a standard disk without copy protection.

I remember this because I still have the modified encoder that emits that extra flux to create the "hard ro decode" sample which just employs the result of the examination what went wrong.

If we have a disk with only very very few extra flux, the position of the extra flux does not tell much, because the law of large numbers does not apply.

keirf, 2023-06-05:
Thank you for the explanation of why two flux are inserted! That was puzzling me but of course it's because polarity reversals come in pairs. D'oh.

Well, then we agree on the symptom. But both (1) and (2) have that symptom. Although this feels less severe than (1), but I'm sure that can vary by drive and disk!

keirf, 2023-06-05:
Thanks everyone for the interesting discussion. No other community has engaged at this level.

closed 2023-06-20 by keirf

keirf, 2024-03-09:
I want to attach another, non-C64 image, which happens to exhibit this issue on one sector (sector 02, track 0.1). In this case a flux of about 8.7us gets split into (2.95, 2.675, 3.10) microseconds. Note that a fairly specific lowpass thresh value is required with our current algorithm: `2.675<thresh<2.95`. If the algorithm looked only for groups-of-three, that might work better.

[stev.zip](https://github.com/keirf/greaseweazle/files/14546734/stev.zip)

keirf, 2024-03-09:
@markusC64 I believe the above commit is an improvement to the flux-merging algorithm. But it might be worth running it with a few of your images which exhibit this problem? Just to make sure I haven't introduced a regression for C64 dumps.