keirf/greaseweazle #326: Is it possible to add support for reading/writing Atari 8 bit disks?

guytaylor, 2023-06-06:
Is it possible to add support for reading/writing Atari 8 bit disks?
i.e. ones from the the Atari 400/800 written with 810/1050 drives?

keirf, 2023-06-07:
Yes ATR image support is planned. Is that the main format you want supported?

smesgr9000, 2023-06-16:
I also would like to archive my Atari stuff. For Verifikation converting to ATR would be great. For copy protected disks ATX would be afaik better.

guytaylor, 2023-06-19:
Great! Yes, I want to archive some of my old personal Atari disks, so ATRs is all I would need. Thanks

avolle, 2024-09-08:
Hello,
Is there any news about support for the Atari ATR format for Atari 8 bit computers?
I would also be very interested in implementing it.
By the way, I think the Greaseweasel adapter is great. I already use it on the Amiga... and yes, I know that you can also create Atari disks with the sio2pc and a 1050 floppy, but the support in Greaseweasel would still be great.
greetings, Alex

notwork-2000, 2026-09-24:
Hi, gw writing directly to atr would be so useful.  Since HxC Floppy Emulator wont export to atr either, i know of no way to read a floppy with gw and convert it to atr.

Please please add atr support the gw.

Thank you in advance!

M1kerochip, 2026-09-24:
A8rawconv converts supercard pro images to atr or atx.

notwork-2000, 2026-09-30:
Thank you so much for pointing me to A8rawconv!