keirf/greaseweazle #329: Ggreaseweazle v4 and chinon fb-354

premiero, 2023-06-21:
Hi there.  I just tried to use my fully working Chinon FB-354 drive from real Amiga 500 and i've got stuck.
I am using original straight cable from Amiga as well as originall power cable from Amiga. 

gw info gives me:

Host Tools: 1.13
Device:
  Port:     COM3
  Model:    Greaseweazle V4
  MCU:      AT32F403A, 216MHz, 96kB SRAM
  Firmware: 1.3
  Serial:   GW0045C5927048000007A5070F
  USB Rate: Full Speed (12 Mbit/s)

so gw device is recognized. But every time i try to read anything from floppy a green diode lights up and then an error:

Reading c=0-79:h=0-1 revs=1.1
Format amiga.amigados
Command Failed: GetFluxStatus: No Index

What am I doing wrong? Is this some power related issue?

premiero, 2023-06-21:
Well i tried all --drive A/B/0/1/2 and nothing works. I am not sure how this fdd is powered, but measuring output with my multimeter on Amiga motherboard gives me +5V/GND/GND/+12V. So maybe it's really missing +12V. Thanks @keirf

keirf, 2023-06-21:
If it's an old full height Chinon drive then it does require 12v

premiero, 2023-06-21:
[image]

keirf, 2023-06-21:
I'm almost certain that drive needs 12v. Plausible methods for powering include a Molex PSU plus Molex to floppy adapter. Or use an ATX psu. Advantage is low watt ATX PSUs are dirt cheap and good quality (eg Delta PSUs on eBay) and you just need to bridge the green wire to ground to turn them on

keirf, 2023-06-21:
Or get yourself a PC 3.5 drive. They are basically always needing only 5v.

premiero, 2023-06-21:
> Or get yourself a PC 3.5 drive

That is the primary goal + printing the case, but atm i just wanted to test the device. :)

closed 2023-07-31 by keirf