keirf/greaseweazle #330: Add ZX Spectrum Opus Discovery disk config

gnulnulf, 2023-06-24:
The img file is normally called "OPD".
The single sided drive: 40T/18S(0-17)/256ss -> 180k
The double sided drive: 80T/18S(0-17)/256ss -> 720k
I'm not sure if I got the gaps right.

Main difference from other formats is the usage of sector 0.
My config is:

disk opus1.ss
    cyls = 40
    heads = 1
    tracks * ibm.mfm 
	    rate = 250
        secs = 18
        bps = 256
        id = 0
        gap2 = 22
        gap3 = 24
        gap4a = 32
        interleave = 7
        iam = no
    end
end

disk opus1.dd
    cyls = 80
    heads = 2
    tracks * ibm.mfm 
	    rate = 250
        secs = 18
        bps = 256
        id = 0
        gap2 = 22
        gap3 = 24
        gap4a = 32
        interleave = 7
        iam = no
    end
end

gnulnulf, 2023-08-17:
I know there is Opus1 and Opus2 hardware, we can expect it to work alike. zx.opus.sd and zx.opus.dd sound great.

mitchelln, 2024-01-31:
Hi. Does this work yet please, or does it need code changes to be implemented? Thanks.

gnulnulf, 2024-02-01:
You only need the profile in your diskdefs.cfg

mitchelln, 2024-02-01:
Thanks. Where do I place the diskdefs.cfg file? I can't see a parameter to specify it.  I'm running under linux.

mitchelln, 2024-02-01:
Thanks. That's sorted it. I tried to add an opd.py file as well, but it didn't pick that up.

gnulnulf, 2024-02-02:
@mitchelln , it now works for you? I think the diskdefs on linux is src/greaseweazle/data/diskdefs.cfg 

I editted the initial comment to be more readable.

Michielhimself, 2024-05-13:
How do I use this without an opd.py file? gw.exe complains that it doesn't recognise the OPD input file I am trying to write to a physical disk.

keirf, 2024-05-13:
Rename *.opd file to *.img and try again.

Michielhimself, 2024-05-13:
That worked and using the single sided format, delivers something on to disk that gives a readable directory entry on Opus but it doesn't lead to a loadable program (Wrong File Type error). Are the gaps certain to be correct for a stock Opus with a single-sided single density drive?

keirf, 2024-05-13:
Gaps don't really need to be specified. Try removing them all from your diskdefs.

keirf, 2024-05-13:
If that doesn't work, only then additionally try gap3=12

Michielhimself, 2024-05-13:
> If that doesn't work, only then additionally try gap3=12

still no, including with interleave = 13 and cskew =13. The Disk index "CAT 1's" on Opus / ZX Spectrum, but no files load. "m Wrong file type, 0:1"

keirf, 2024-05-13:
I don't know then I'm afraid. I assume the OPD is known good? It sounds more like a filesystem error than a sector/track level issue.

Michielhimself, 2024-05-13:
Yeah, all the OPD I'm using are the original 180K format and mount + load fine in an actual Opus via the GoTek, including copying them to physical disk (in the ZX Spectrum / Opus) and then running from the physical disk. Only factor I haven't been able to exclude yet is the floppy drive I'm using for the imaging. It runs pretty much error-free on the greaseweazle but that's still a drive verifying it's own sectors - if it's out of whack a bit it's not going to produce something the Opus can read.
I have another way of writing OPD to physical media, using a Spectrum HW accurate emulator on older PC hardware, but really wanted to have the Greaseweazle option working to not have to rely on older hardware anymore, so I'll look around for another 3.5" drive to use here in my cave-o-parts and see if that changes things.

Many thanks for the help with the attempts so far!

gnulnulf, 2024-05-13:
The reason I played with the gaps was to keep it within the maximum tracksize. 
full track = 6250
index ->gap4->trackheader-> gap1 -> ( sectorheader -> gap2 -> sector data -> gap3 ) * 18 ->gap4b

Are you using a 40 track drive or a 80 track with double steps? You can try to create a scp and look at it with hxc-> track analyzer. 

I used this to write:
gw.exe write --drive=1 --tracks=step=1:c=0-39:h=0 "D:\zxspectrum\d6.img" --format=opus1.ss --diskdefs=diskdefs.cfg

If you like I can try to reproduce it on my side.

keirf, 2024-05-13:
Greaseweazle will automatically mess with all gaps (except gap2) to fit sectors in. Of course if you do know the correct gap values from a real disk, that is best of all.

Michielhimself, 2024-05-14:
Trying to use an 80-track drive.
Should I use step = 1? I guess that's not handled automagically.
The 40-track I have has been modded (I keep as a spare for when the one in Opus dies) so doesn't take signal or voltage using standard cabling from the Greaseweazle.

Michielhimself, 2024-05-14:
Using same command options and the settings from gnulnulf's original post at the top of this thread in diskdefs.cfg, and using a different OPD image, I made another succesfully readable opus phsyical disk (used blank media to start, so not an accidental residue from an earlier attempt).

Michielhimself, 2024-05-14:
Update: one one of the disks I imaged with gnulnulf's settngs, some games loaded and others (higher numbered tracks) did not. I now imaged it again using the settings recommended by Keirf without the gap3=12 and that rewritten disk (from same image) now works fine for all programs on the image.

`disk zx.opus.ss
    cyls = 40
    heads = 1
    tracks * ibm.mfm 
	    rate = 250
        secs = 18
        bps = 256
		id = 0
		cskew = 13
        interleave = 13
        iam = no
    end
end

disk zx.opus.dd
    cyls = 80
    heads = 2
    tracks * ibm.mfm 
	    rate = 250
        secs = 18
        bps = 256
		id = 0
		cskew = 13
        interleave = 13
        iam = no
    end
end
`