keirf/greaseweazle #333: `pip install` on Debian requires python3-dev package

0cjs, 2023-07-08:
While trying to install Greaseweazle on Debian 11 (in a Python virtualenv) using 
```
pip install git+https://github.com/keirf/greaseweazle@latest
```
per the install instructions, I was given some error output, which ended with
```
      src/greaseweazle/optimised/optimised.c:3:10: fatal error: Python.h: No such file or directory
          3 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
...
Successfully built crcmod
Failed to build greaseweazle
ERROR: Could not build wheels for greaseweazle, which is required to install pyproject.toml-based projects
```
As it turns out, on Debian (and no doubt Ubuntu) you need to have the `python3-dev` package installed in order to install Greaseweazle. After a `sudo apt-get install python3-dev`, re-running the `pip install` command appeared to work fine, and I can run `gw info`.

This is something that should probably be mentioned in the Linux install instructions.

keirf, 2023-07-08:
Interesting. I don't think I'd expect a virtualenv to depend on a distro package like that. The install instructions actually say to use `python3 -m pip install ...`. Using `pip` directly doesn't always direct to the correct python installation, although that said I do think the virtualenv implementation does try to do the right thing.

0cjs, 2023-07-08:
`pip` was almost certainly doing the right thing; like `python` it's also run out of `virtualenv/bin/` and all the files there are hard-coded to use the virtualenv Python, not any other. (I run programs using `virtualenv/bin/` paths to run them in the virtualenv all the time.) But it's not particularly relevant for this case anyway, since a) I had the virtualenv activated, so the various paths were set up, and b) it's using the system Python anyway.

Which gets to why this is really happening: it's using the system Python, and the Debian `python3` package is _not_ everything you'd get if you built Python yourself and installed it. They've taken parts of the "standard" install and split them out into separate packages, such as the development tools into `python3-dev`. I've seen this also with the need to install `python3-distutils` for one of my applications where I didn't have to do this when running from one of my own builds of Python.

keirf, 2023-07-08:
Ah ok that makes sense. I will update the wiki appropriately.

keirf, 2023-07-08:
I always test with a pyenv-installed version of Python so I would never see this myself.

0cjs, 2023-07-08:
> I always test with a pyenv-installed version of Python so I would never see this myself.

Right. It's a bit insidious because developers are moderately unlikely to be testing on systems with `python3` but without the other packages that "complete" the Python distribution. I didn't notice the issue with my own code until I was bringing up a new personal Linux system and happened to run the tests in one of my repos before I'd installed most of my development tools on it.

I ended up working around this for my application by setting up automated tests that ran in Docker containers with minimal installs of various OSes. (That would probably be more trouble than it's worth in this particular case, though.)

keirf, 2023-07-30:
Finally updated the wiki!

closed 2023-07-30 by keirf