keirf/greaseweazle #338: gw fails to load after firmware update

yuv422, 2023-07-21:
Hi,
I plugged my unit into a windows 10 machine and it identified as COM3 I then ran 

and got this
```
$ ./gw.exe info
Host Tools: 1.14
Device:
  Port:     COM3
  Model:    Greaseweazle V4
  MCU:      AT32F403A, 144MHz, 96kB SRAM
  Firmware: 1.2
  Serial:   GW0045C5A270480000077D070F
  USB Rate: Full Speed (12 Mbit/s)

*** New firmware version 1.3 is available
To perform an Update:
 - Run "gw update" to download and install latest firmware
```

I then tried to update the firmware and now the unit isn't being recognized!

```
$ ./gw.exe update
Downloading latest firmware: greaseweazle-firmware-1.3.upd
Updating Main Firmware to version 1.3...
Done.
** FATAL ERROR:
Could not reopen port after mode switch
```

```
$ ./gw.exe info
Host Tools: 1.14
Device:
  Not found
```

Any idea how I recover from this?

I haven't actually gotten to use the unit at all yet. :(

keirf, 2023-07-21:
You have a couple of options. The first is to place a jumper across RXI and TXO pins. Then connect by USB. Now see if the device appears in `gw info`. If so it should be in bootloader mode and you should be able to retry `gw update`.

yuv422, 2023-07-21:
It looks like it did flash ok. I originally had it connected through a hub. Once I moved it on to a direct usb port it started to work.

It's strange it worked the first time before the flashing on the hub.

```
$ ./gw.exe info
Host Tools: 1.14
Device:
  Port:     COM3
  Model:    Greaseweazle V4
  MCU:      AT32F403A, 216MHz, 96kB SRAM
  Firmware: 1.3
  Serial:   GW0045C5A270480000077D070F
  USB Rate: Full Speed (12 Mbit/s)
```

It's good to know that you can flash in the bootloader too.

keirf, 2023-07-21:
That is weird. Usually will work fine on a hub. I guess as long as you found a port that does work, that's probably okay.

yuv422, 2023-07-21:
I'll try downgrading to 1.2 and seeing if it works on the hub again.

keirf, 2023-07-21:
To downgrade: `gw update --force --tag v1.2`

yuv422, 2023-07-22:
I went to run these tests but it's now working on the hub with the 1.3 firmware. So I think there must be some other issue with my hub / setup. So happy to close this issue.

SelfRef, 2024-12-02:
I can confirm that the latest version still has problem with update 1.5->1.6 through USB hub. Works fine when connected directly.

keirf, 2024-12-02:
So 1.5 worked but 1.6 doesn't, via hub? Does 1.5 work again if you downgrade?

SelfRef, 2024-12-03:
> So 1.5 worked but 1.6 doesn't, via hub? Does 1.5 work again if you downgrade?

No, I bought greaseweazle with stock firmware 1.5. I got banner in `gw` "New firmware version 1.6 is available". When I tried to upgrade, I've got "Could not reopen port after mode switch" (same as first post).

I was getting this error until I reconnected greaseweazle directly to PC, without a hub. Other features work fine through hub, so it's not a big deal, but maybe an additional message to try that in the error message would be helpful.

keirf, 2024-12-03:
Ahhhh ok I see. Yes suggesting to try direct connection or another port is a good idea. I will add that to the help text.

By the way are you on the latest tools, 1.21? Greaseweazle detection did get changed for Windows a release or so ago.

SelfRef, 2024-12-03:
> Ahhhh ok I see. Yes suggesting to try direct connection or another port is a good idea. I will add that to the help text.
> 
> By the way are you on the latest tools, 1.21? Greaseweazle detection did get changed for Windows a release or so ago.

First I tried on 1.21, then I tried git version and got the same issue, so it's for sure still exists on `master` branch.

keirf, 2024-12-03:
My best theory is that the hub simply doesn't detect the device disconnection, and thus never initiates re-enumeration. But then the device would still be detected, as it would never have been disconnected and torn down in the OS. So that doesn't make sense really.

*Something* about re-connection is obviously weird anyway. Stating the obvious.

What sort of hub do you have exactly? Maybe I can reproduce this if I have the same hub.

SelfRef, 2024-12-03:
I did some tests. Normally I use [TP-LINK UH720](https://www.tp-link.com/us/home-networking/usb-hub/uh720/), but I doubted it is related with the problem.

I've tried with TP-LINK and the cheapest generic hub I have, and it behaves the same

When connected directly to PC, the update always works:
```sh
❯ gw update --file greaseweazle-firmware-1.6.upd --force
*** TEST/PRE-RELEASE: 1.22.dev3+gf0d43b7
*** Use these tools ONLY for test and development!!
Updating Main Firmware to version 1.6...
Done.
```
When connected through (any) USB hub it either output this
```sh
❯ gw update --file greaseweazle-firmware-1.6.upd --force
*** TEST/PRE-RELEASE: 1.22.dev3+gf0d43b7
*** Use these tools ONLY for test and development!!
Updating Main Firmware to version 1.6...
** FATAL ERROR:
Could not reopen port after mode switch
```
Or this (notice the "Done" line)
```sh
❯ gw update --file greaseweazle-firmware-1.6.upd --force
*** TEST/PRE-RELEASE: 1.22.dev3+gf0d43b7
*** Use these tools ONLY for test and development!!
Updating Main Firmware to version 1.6...
Done.
** FATAL ERROR:
Could not reopen port after mode switch
```

I see a small difference in wait time just before the FATAL ERROR in case I connect Greaseweazle after connecting a hub to PC vs first connecting Greaseweazle to hub and then hub to PC. Not much in final effect, but it may be important.

keirf, 2024-12-03:
Seeing `Done` means that the mode switch and re-enumeration doesn't always fail: In those cases the mode switched to update, performed the update, but then failed to re-enumerate back in normal mode.

Have you tried more than one cable?

I often `gw update` via hub in Linux but I could give this a go on my Windows 11 laptop.

SelfRef, 2024-12-03:
I tried different cables now, but it doesn't change anything. Updating using direct connection with any cable works 100% of the time. Updating through USB hub always throws FATAL ERROR at the end. Without a doubt, there is some issue with re-enumerating USB devices when hub is used.

keirf, 2024-12-03:
Okay, I reproduced the same issue on a Linux box using a basic bus-powered USB 3.0 hub (Inatek HBU3VL2-4).

Totally weird. Sometimes it takes ages for the device to be logged as disconnected. Usually it then re-connects quickly, but sometimes it doesn't come back.

Anyway, this gives me something to experiment with!

keirf, 2024-12-04:
Current firmware waits around 500ms between disconnect and re-connect. That is actually not needed by the USB spec *and* seems to really piss off some hubs, for reasons I don't fully understand. 

So far I had great success simply removing this delay. It is implemented on branch `usb-disc` and in this automated build: https://github.com/keirf/greaseweazle-firmware/actions/runs/12158351570

Please can you download the above artifact (you will need to be logged in to GitHub to see and download the artifact). And then update to the new firmware for both bootloader and main firmware (not via hub, of course):
```
$ gw update --file greaseweazle-firmware-80a79cd/greaseweazle-firmware-1.6.upd --force --bootloader
$ gw update --file greaseweazle-firmware-80a79cd/greaseweazle-firmware-1.6.upd --force
```

Now you can connect via the problematic hub and test mode switches as follows:
```
$ gw info --bootloader
```

You should hopefully find the above command works after your update to fixed firmware, and is unreliable with the current latest released firmware v1.6.

SelfRef, 2024-12-04:
Indeed, I can confirm that after flashing the test firmware there are no errors through any hub:

```sh
❯ gw info --bootloader
*** TEST/PRE-RELEASE: 1.22.dev3+gf0d43b7
*** Use these tools ONLY for test and development!!
Host Tools: 1.22.dev3+gf0d43b7
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4.1
  MCU:      AT32F403A, 144MHz, 224kB SRAM
  Firmware: 1.6 (Bootloader)
  Serial:   GWB01431335976C01007281705
  USB:      Full Speed (12 Mbit/s)

❯ gw update --file greaseweazle-firmware-1.6.upd --force
*** TEST/PRE-RELEASE: 1.22.dev3+gf0d43b7
*** Use these tools ONLY for test and development!!
Updating Main Firmware to version 1.6...
Done.
```

Both `gw info --bootloader` and subsequent `gw update` commands are working correctly, no matter if I connect Greaseweazle through hub or not.

I think I can call it as resolved. Thank you very much :smiley:

keirf, 2024-12-04:
I'll try and get some broader testing and get this one pushed out, thanks! Unfortunately since it needs a bootloader update *as well*, I guess I must also add warning help to the `Could not reopen port[...]` message. I will leave the ticket open for tracking this work.

keirf, 2024-12-05:
My own testing indicates this isn't a panacea; I still find things are slow/unreliable through hubs on my home desktop Linux box. It seems to be related somehow to the Linux serial subsystem. I'm not sure I can make this 100% good, but certainly reducing the delay in the firmware doesn't do any apparent *harm*. So that can stay, and I've put a straightforward help hint in the tools error message.

Not sure this suffices to make a v1.7 firmware release. I'll think about it.

closed 2024-12-05 by keirf