keirf/greaseweazle #339: Hard sector support

xolod79, 2023-07-21:
I have a number of 8" hard sector floppies.  I would like to have support for them in gw.

keirf, 2023-07-30:
Happy to support. But I need help from someone (such as you) to provide raw dumps of floppies and hopefully docs about track formats (unless we can work that out, or find some online resources). I also need new features to be tested, for "gw read" and optionally "gw write" as well.

xolod79, 2023-08-16:
How can I make a raw image of a disk with hard sectors? I have Greaseweazle and Fluxengine. Judging by this, I can try Fluxengine for this. https://github.com/davidgiven/fluxengine/blob/master/doc/faq.md

keirf, 2023-08-17:
You can use greaseweazle but specify lots of revolutions, as each index pulse is judged to be a full revolution.

For example, `gw read --revs=74 --tracks=c=0-39:h=0-1 mydump/00.0.raw`

This would work if your disk has 36 sectors, it has 37 index holes, so the above will dump two actual full revolutions. The above assumes 40 cylinders and 2 heads. Adjust in the obvious way if this is incorrect.

xolod79, 2023-10-19:
Unfortunately, for some reason it is not possible to read the hard sector of the disk.
gw read --drive 1 --revs 74 --tracks c=0-77:h=0-1 ..\Pluremat3\plremat00.0.raw

https://mega.nz/file/GkQEmCjb#IXp6NUxNbZYhuDTUWmK5GkRRtR41ybM7lLThGVoR37w

keirf, 2023-10-25:
How many flux per track is reported during `gw read`? These files are basically empty. It's like no flux is found at all, as if RDATA line isn't working. They also onluy have one index pulse, but this could plausibly be a bug in my Krryoflux stream generator when presented with no flux.

xolod79, 2023-10-25:
Yes, fluex when reading were equal to zero.  There is an assumption that the NEC FD1165 drive on which I read this floppy disk is too modern and does not support hard sectors.  Since there are many holes in floppy disks, the drive cannot correctly determine the rotation speed of the floppy disk and therefore does not output data from it.

keirf, 2023-10-25:
That seems quite plausible.

sweproj, 2024-07-05:
Hi

Just to continue here. I have some hard sectors disc 5-1/4 for Northstar, it has 10 sectors. But it has 10 index holes for marking the sectors and there is one extra holes at the start of the disc. The disc is a Scotch 744-10 RH. Is there anyway to change how the index is counting?

keirf, 2024-07-05:
For dumping purposes it is sufficient for now to try dumping lots of "revolutions". Software can then decode that back to sectors later. Question is do you have a drive that will spin up a hard-sectored disk?

sweproj, 2024-07-05:
Hi

No I dont have a drive that take cares of the hard sectors, My idea would be to do somethings in fw software to detect the index pulse instead

keirf, 2024-07-05:
For dumping it really doesn't matter so long as the drive doesn't choke on more than expected number of index pulses.

sweproj, 2024-07-05:
Yes and it is choked since its 11 index holes. Manage to change the firmware side to find the start index and just send it over to the tool, but I also need the others to make the hfe correct

keirf, 2024-07-05:
I wouldn't change the firmware. Fix it up in software. I'm going to guess you want to make a raw HFEv3, so that generator will need fixing too?

Send me a raw plain old Kryoflux/SCP dump with lots of "revolutions" and I can take a look at the host side.

keirf, 2024-07-07:
Some hard-sector conversion support, and a North Star codec, are now included in the latest 1.18 release.

**Please** if you have hard-sectored disks and/or systems, and want to see further features, get in touch! These improvements can't happen without user feedback and testing.

closed 2024-08-22 by keirf