keirf/greaseweazle #346: wirte hfev3 error

emilen72, 2023-08-10:
Hi, I'm trying to write hfev3 to real disk... but I got two king of error:

./gw write /Users/bigsur/Downloads/Back_To_The_Future_2_Image_Works_AtariST_DiskImage/BackToTheFutureII_ImageWorks_Disk1of1_AtariST_Dump/BackToTheFutureII_ImageWorks_Disk1of1_AtariST_Dump0.scp.HxC_HFEv3.hfe 
Writing c=0-81:h=0-1
T0.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T0.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T1.0: Writing Track (Flux: 199.8ms period, 220.1 ms total, Terminate at index)
T1.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T2.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T2.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T3.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T3.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T4.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T4.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T5.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T5.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T6.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T6.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T7.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T7.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T8.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T8.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T9.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T9.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T10.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T10.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T11.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T11.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T12.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T12.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T13.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T13.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T14.0: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T14.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T15.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T15.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T16.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T16.1: Writing Track (Flux: 199.8ms period, 220.1 ms total, Terminate at index)
T17.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T17.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T18.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T18.1: Writing Track (Flux: 199.8ms period, 219.8 ms total, Terminate at index)
T19.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T19.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T20.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T20.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T21.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T21.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T22.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T22.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T23.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T23.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T24.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T24.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T25.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T25.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T26.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T26.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T27.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T27.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T28.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T28.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T29.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T29.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T30.0: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T30.1: Writing Track (Flux: 199.8ms period, 220.0 ms total, Terminate at index)
T31.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T31.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T32.0: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
T32.1: Writing Track (Flux: 199.8ms period, 219.9 ms total, Terminate at index)
[... 111 lines cut ...]
AssertionError

or if I set --version=3 optionI got
./gw write /Users/bigsur/Downloads/Back_To_The_Future_2_Image_Works_AtariST_DiskImage/BackToTheFutureII_ImageWorks_Disk1of1_AtariST_Dump/BackToTheFutureII_ImageWorks_Disk1of1_AtariST_Dump0.scp.HxC_HFEv3.hfe::version=3
** FATAL ERROR:
/Users/bigsur/Downloads/Back_To_The_Future_2_Image_Works_AtariST_DiskImage/BackToTheFutureII_ImageWorks_Disk1of1_AtariST_Dump/BackToTheFutureII_ImageWorks_Disk1of1_AtariST_Dump0.scp.HxC_HFEv3.hfe::version=3: Unrecognised file suffix '.hfe::version=3'
Known suffixes:
.a2r   .adf   .ads   .adm   .adl   .d64   .d71   .d81   .d88   .dcp   .dim   
.dsd   .dsk   .edsk  .fdi   .hdm   .hfe   .ima   .img   .imd   .ipf   .mgt   
.msa   .raw   .sf7   .scp   .ssd   .st    .td0   .xdf          

this on macOS Bigsure
./gw info
Host Tools: 1.15.1
Device:
  Port:     /dev/cu.usbmodem144301
  Model:    Greaseweazle F1 Plus (Unbuffered)
  MCU:      72MHz, 20kB SRAM
  Firmware: 1.4
  Serial:   GW066BFF303538524D43052323
  USB Rate: Full Speed (12 Mbit/s)

but same if I use gui GreaseweazleGUI 2.116 on windows with host tools 1.15

maybe I misunderstand and hfev3 isn't to be useful?

keirf, 2023-08-10:
Can you please upload or attach the problematic hfev3 image file? Hfev3 is mainly intended for emulation (Goteks) but it should not cause a crash in gw write! So I will reproduce this and fix the bug. Thank you.

emilen72, 2023-08-10:
[hfev3.zip](https://github.com/keirf/greaseweazle/files/12313879/hfev3.zip)
in the zip file the hfe disk images (2xBTTFII, R-Type, SOTB) that give Assertion Error....

the other one, gives another error but maybe a faulty image
./gw write /Users/bigsur/Downloads/bombuzal-image-works-disk-1of-1-atari-st.scp.-02-data-hx-c/Bombuzal/Bombuzal_ImageWorks_Disk1of1_AtariST.scp.HxC_HFEv3.hfe
** FATAL ERROR:
HFEv3: Short track data

hope can be helpful... TNX

keirf, 2023-08-10:
By the way what is the source of the hfev3 files? HxC tools?

commit bf23c6aa (2023-09-05): HFEv3: Improve error logging and make truncated bitrate opcode non-fatal

This is observed in some real HFEv3 images. We should not barf on them.

Fixes #346
closed 2023-09-05 by keirf