keirf/greaseweazle #347: TSC FLEX - no data on track 0 head 1 on Double Sided diskette

mickecamino, 2023-08-12:
I am trying to recover some old TSC FLEX diskettes and have stumbled on a quirk.
If a disk is formatted 40 tracks Double Sided, Single Density there is no data on Cylinder 0 Head 1.
The following custom setup works, but with bad data on C0, H1:
```
# TSC FLEX: DS/SD with SD (FM) cylinder 0.
disk flex.40.ds
  cyls = 40
  heads = 2
  tracks 0.0 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 4
    id = 1
  end
  tracks 0.1 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 4
    id = 1
  end
  tracks 1-39.0 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    id = 1
    h = 0
  end
  tracks 1-39.1 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    h = 0
    id = 11
  end
end
```
If I remove the definition for tracks 0.1 ibm.fm gw crashes (gw.exe on W10 x64)
```
gw info
Host Tools: 1.15.1
Device:
  Port:     COM11
  Model:    Greaseweazle F1
  MCU:      72MHz, 20kB SRAM
  Firmware: 1.4
  Serial:   GW0669FF534952754967072228
  USB Rate: Full Speed (12 Mbit/s)
```
If I leave the entry in I get:
```
T0.0: IBM FM (10/10 sectors) from Raw Flux (57133 flux in 401.03ms)
T0.1: Ignoring unexpected sector C:0 H:0 R:11 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:17 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:12 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:18 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:13 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:19 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:14 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:20 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:15 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:16 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:11 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:17 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:12 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:18 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:13 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:19 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:14 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:20 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:15 N:1
T0.1: Ignoring unexpected sector C:0 H:0 R:16 N:1
```
I got an image with some bad data, but I salvaged the diskette.
Question, how do I skip C0,H1 in the config file?

mickecamino, 2023-08-12:
And I solved it, the ticket can be closed.
Use this config:
```
# TSC FLEX: DS/SD with SD (FM) cylinder 0.
disk flex.40.ds
  cyls = 40
  heads = 2
  tracks 0.0 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 4
    id = 1
  end
  tracks 0.1 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 4
    h = 0
    id = 11
  end
  tracks 1-39.0 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    id = 1
    h = 0
  end
  tracks 1-39.1 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    h = 0
    id = 11
  end
end
```

keirf, 2023-08-17:
Is the interleave *really* different between cylinder 0 and cylinders 1 upward? It seems weird, and if interleave is actually the same on all cylinders, the format can be simplified. If you can confirm one way or the other, I can add this format to the diskdefs.cfg file.

mickecamino, 2023-08-17:
I see your point, so I tested the following:
FLEX 40 tracks, Double Side, Single Density
```
disk flex.40.dssd
  cyls = 40
  heads = 2
  tracks 0-39.0 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    id = 1
  end
  tracks 0-39.1 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    h = 0
    id = 11
  end
end
```
And it worked.
The following is also working:
FLEX 35 tracks, Single Side, Single Density
```
disk flex.35.sssd
  cyls = 35
  heads = 1
  tracks 0-34.0 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    id = 1
    h = 0
  end
end
``` 
And FLEX 40 track Single Side Single Density
```
disk flex.40.sssd
  cyls = 40
  heads = 1
  tracks 0-39.0 ibm.fm
    bps = 256
    iam = no
    gap3 = 16
    gap4a = 16
    secs = 10
    interleave = 6
    id = 1
    h = 0
  end
end
```
Unfortunately my 80-track drive broke, I am looking for a new one, once I get it I will test FLEX 80 track diskettes. I have two of them.

mickecamino, 2023-08-17:
This is the FLEX layout:
Track 0 does not have a sector 1. 
All of the other tracks don't have a sector 0.

```
track 0   sector  0   boot sector
      0           2   extended boot sector
      0           3   System Information Record (SIR)
      0           4   not used
      0           5   start of the directory area.
     ...         ...  the rest of track 0 is used for
                      directory storage
      1           1   start of the data and program
                      storage area of diskette
```

keirf, 2023-08-17:
Is that true? I haven't seen any Flex layouts defined with a sector 0 for track 0. Including yours.

mickecamino, 2023-08-17:
I got the information from the Flex user Group, and they are wrong.
The information I got is from their FAQ: [http://www.flexusergroup.com/flexusergroup/fug10.htm](url)
I downloaded the TSC 6809 Flex Adaption Guide and read it, and here is the information_
First, track 0 is always written in Single Density:
```
Double-density disks are usually not really different from single-density disks with the exception 
of the fact that there are more sectors per track. Technical Systems Consultants has altered this 
concept slightly. In our specifications, a "double-density disk" actually has track #0 written in 
single-density while all other tracks are written in double-density
```
Standard definition:
```
8" SINGLE-SIDED, SINGLE-DENSITY, SOFT-SECTORED DISK
This disk should be comprised of 77 tracks (numbered 0 thru 76) with 15 sectors per track (numbered 1 thru 15).

5 1/4" SINGLE-SIDED, SINGLE-DENSITY, SOFT-SECTORED DISK
This disk should be comprised of 35 tracks (numbered 0 thru 34) with 10 sectors per track (numbered 1 thru 10)
```
And here is the NOTES:
```
NOTES:
1) On double-density disks, track #0 is formatted in single-density to facilitate automatic density selection.
2) Side #0 is the bottom of the disk (opposite the label).
3) Sector size is 256 bytes.
4) Track numbers always begin with #0 and sector numbers always begin with #1.
```

mickecamino, 2023-08-28:
I looked in the Flex Adaption Guide for 6800 and found why some old FLEX disk has sector #0 on track #0. This is from the Notes Section:

_Some systems have ROM monitors with boots which look for a sector #0 on track #0. Disks for these systems may have a sector #0 instead of a sector #1 on track #0._

I guess that is why some got confused about this.

closed 2023-10-05 by keirf