keirf/greaseweazle #355: Use existing header for external LED?

evan, 2023-08-26:
I have a Greaseweazle V4. I installed it in an external drive enclosure with drives, so I can't see the activity LED. It would be nice if the software could support the use of the existing jumper block to run a panel-mounted LED, or if future revisions of the PCB had a dedicated LED header.

keirf, 2023-08-26:
You can connect an LED between pins 3v3 and TXO. You will need a series resistor of at least a few hundred ohms. If the LED isn't bright enough connect to 5v instead of 3v3.

evan, 2023-08-26:
Oh, excellent, I will go give this a try. Sorry if I missed it in the docs.

keirf, 2023-08-26:
I don't think it's documented. I should add it to the V4 Setup wiki page.

closed 2023-10-05 by keirf