keirf/greaseweazle #356: flippy disks

smesgr9000, 2023-08-26:
It would be nice to have some more documentation about flippies aka flippy disks aka flippy floppies in the wiki.

My suggestion under usage:

### Flippy Disks

Several 8bit home computers used 5.25" drives with a single read head but used both side of the disk to store data. IBM PC compatible drives with two heads can't read the other side of the disk. If greaseweazle uses both heads to read the other side no useful data is recorded. If the physically flipped disk is an error occurs `Command Failed: GetFluxStatus: No Index.

Those drives require a hardware modification to the drive to provide an alternative index sensor.
An alternative solution is to fake the index pulse signal by greaseweazel.

#### Measure your drive
The rotation speed of your drive is required. 
1. Put your a disk in your drive  
2. measure your speed by `gw rpm --nr 10`

The values should be relative constant for every measurement. If the rate spread is to high the chances are high wrong data is read with an fake index.

#### Read flip side
Now flip the floppy in your drive as you would do on the original hardware. And use the following command: `gw read --fake-index <speed> --tracks h=0" with your other options see (link to tracks). Example:

- `gw read --fake-index 360 --tracks h=0 --drive 0 --raw test.scp`  reads a RAW single sided image with 360rpm  for shugart drive 0.

Afaik it would be also useful to link this new wiki page in the FAQ @ "Track 0 not found"

jedahan, 2023-08-27:
Thank you so much for starting this discussion, adding a table for models and hardware people worked with, would be an excellent step as well.

Yserra, 2023-08-31:
If I may say so, most of the 360 RPM drives will refuse to communicate with the host computer (aka Greaseweazle board) until they find the index. So the --fake-index option may often not work. It's more for very old drives (300 RPM) that don't manage themselves the index hole. The alternative index sensor is more difficult to make, but it works.

smesgr9000, 2023-08-31:
yeah good point:
- Mitsubishi MF504A-368U  :white_check_mark:
- Ye-Data YD-380B :x:

keirf, 2023-10-01:
I have made a wiki page here https://github.com/keirf/greaseweazle/wiki/Flippy-Disks

Comments are welcome.

qbarnes, 2023-10-02:
I have a suggestion for clarifying the first method for reading flippy disks and adding a second.

I'd make the first bullet:
```
1. Punch two holes to make a second index hole in the disk jacket in
the correct flipped position.  An ordinary hand hole punch can be used, but
ensure that the punch is protected from scratching or damaging the media,
often using plastic tape to cover any exposed metal.
```

There is no need to open the jacket for this approach.  I do it all the time the way I describe.  I had an old flippy drive that got stolen, so to recover my old media, I did the above most of the time.

New method:
```
2. Read the normal side first.  For the flippy side, carefully slice open the
jacket on the label end not to nick or damage the media, remove the media,
flip it over, and reinsert it.
```

A modification to method #2 is to cut open the jacket, remove its media, and insert it into another jacket that's already been prepped with double index holes.  I use this method when the original jacket has problems such as mold.

Please fix any of the verbiage to make it clearer.  If you have any questions about my changes, please ask.

keirf, 2023-10-03:
@qbarnes I'm unclear exactly how you perform suggestion (1). This is one of this style of hole punch? https://uk.rs-online.com/web/p/pliers/4458841

Do you slip the anvil side inside the jacket? Is this easy without warping the jacket or scratching the media? How do you get the positioning correct?

qbarnes, 2023-10-03:
> This is one of this style of hole punch?

Nothing so sophisticated.  I just used a cheap, single hole punch from Staples for under $3.  I think this was it: https://www.staples.com/Staples-10573-CC-1-Hole-Punch-5-Sheet-Capacity-Silver/product_146308

The 1/4" hole punch punched a slightly larger hole than the original, but it didn't matter for all the diskettes I punched with the drive I was using.

> Do you slip the anvil side inside the jacket? Is this easy without warping the jacket or scratching the media?

Generally correct.  I do have to make a gap between the jacket and media to slip the taped up, bottom side of the punch inside, but the bending to accommodate the punch is minor and does not do any damage.

> How do you get the positioning correct?

With a template I made.

My paper template isn't a perfect technique.  I use it with a felt tip pen to draw where I want the index hole circle on the opposite side.  Then I compare that pen mark to its symmetrically matching hole already in the jacket and ever so slightly adjust where I need punch through the jacket to make a more accurate hole.

I only did about a 20-25 disks this way.  If I was doing more, I might make the template out of a more durable, less flexible material to more accurately position the pen mark (or maybe a jig), so I didn't have to manually check it and adjust.

Template and taped up punch: https://drive.google.com/file/d/1poNi4z_KxyJCfps-Rm_ZRv5RRCCdwzHp/view?usp=sharing

I realize I'm still hand-waving here and there, so if further explanation or pictures are needed, just let me know.

I originally started using approach 2 where I slice off a bit of the jacket on the outside edge (side furthest from the drive head hole) to extract and flip the media, but once I made the template, I found the above technique quicker, easier, and less intrusive.

smesgr9000, 2023-10-12:
> I have made a wiki page here https://github.com/keirf/greaseweazle/wiki/Flippy-Disks
> 
> Comments are welcome.

Thanks looks good to me

closed 2024-02-04 by smesgr9000