keirf/greaseweazle #358: Official SPS IPF #3158 (Kick Off 2, Atari ST) cannot be written to disk

keirf, 2023-09-03:
Problem is that the SPS IPF library returns `CapsTrackInfo.overlap` field as `-1`, which appears to mean "unknown". Since the Anco protection includes sectors crossing the index mark, we cannot correctly write them without knowing the correct write splice point.

This needs some investigation:
* Can DTC/Kryoflux write this IPF?
* Is there a heuristic we can use to sensibly guess a splice point?
* Should Greaseweazle explicitly reject such IPF images with a clear error message?

keirf, 2023-09-04:
This also applies to
* #3159 Kick Off 2: The Final Whistle
* #3183 Kick Off 2: Winning Tactics
* #3215 Kick Off 2: Return to Europe

keirf, 2023-09-04:
Looking at an original dump, compared with IPF, the original has two aspects to the NFA region missing from the IPF:
* It is longer (but presumably the protection check doesn't care)
* There is irregularity near the end of the NFA, which is probably a write splice.

From this I may be able to implement a heuristic to drop a write splice at the end of a NFA or fuzzy region if no splice is specified.

keirf, 2023-09-04:
See also #231 which can probably be fixed at the same time (fixing verification of NFA regions).

keirf, 2023-09-04:
Temporary link to original dump: https://drive.google.com/file/d/1yVEskEdX_NbyX_5Zv2CR7t61zScvTMtH/view?usp=sharing

Temporary link to IPF generated from this dump: https://drive.google.com/file/d/1t1sgxzxwFNKduSqfYAVsAqkt1t_Il0UJ/view?usp=sharing

keirf, 2024-08-11:
This needs testing now that handling of unspecified write splice has been committed to master. A user test in #464 indicates that there's probably a bit more work to do here. I believe actually that KO2 might be a special case for mastering.

DaveHarrison1968, 2024-08-22:
Hi Keir, I have left a message on the fb page also, but wanted to say that with the latest release of version 1.19 I have been finally able to write my Kick Off 2 (Atari St) IPF back to disk and loaded perfectly on my ST. I didn't get any errors with verification of any tracks while writing either. Thank You for all your hard work.

keirf, 2024-08-23:
Thanks for the report!

closed 2024-08-23 by keirf