keirf/greaseweazle #360: IBM format verification not always detecting CRC errors?

Litude, 2023-09-23:
I have dumped some floppies where I have noticed that Greaseweazle does not report any read errors, but investigating the floppy with HxC later reveals that there were actually some read errors.

I use the following command for dumping 3.5" 1440k IBM floppies: `gw read --tracks=c=0-81 --format ibm.1440 --revs 5 --raw NAME_00.0.raw` (I always dump a few extra tracks to capture possible duplicator information).

This is an example output from one dump that had one read error but Greaseweazle reported no errors:
```
Reading c=0-81:h=0-1 revs=5
Format ibm.1440
T0.0: IBM MFM (18/18 sectors) from Raw Flux (416987 flux in 998.24ms)
T0.1: IBM MFM (18/18 sectors) from Raw Flux (457358 flux in 998.16ms)
T1.0: IBM MFM (18/18 sectors) from Raw Flux (386199 flux in 998.15ms)
T1.1: IBM MFM (18/18 sectors) from Raw Flux (386400 flux in 998.12ms)
T2.0: IBM MFM (18/18 sectors) from Raw Flux (389595 flux in 998.13ms)
T2.1: IBM MFM (18/18 sectors) from Raw Flux (386130 flux in 998.12ms)
T3.0: IBM MFM (18/18 sectors) from Raw Flux (385404 flux in 998.11ms)
T3.1: IBM MFM (18/18 sectors) from Raw Flux (384687 flux in 998.11ms)
T4.0: IBM MFM (18/18 sectors) from Raw Flux (387504 flux in 998.11ms)
T4.1: IBM MFM (18/18 sectors) from Raw Flux (386674 flux in 998.08ms)
T5.0: IBM MFM (18/18 sectors) from Raw Flux (385885 flux in 998.12ms)
T5.1: IBM MFM (18/18 sectors) from Raw Flux (387185 flux in 998.10ms)
T6.0: IBM MFM (18/18 sectors) from Raw Flux (385975 flux in 998.12ms)
T6.1: IBM MFM (18/18 sectors) from Raw Flux (378726 flux in 998.10ms)
T7.0: IBM MFM (18/18 sectors) from Raw Flux (386348 flux in 998.09ms)
T7.1: IBM MFM (18/18 sectors) from Raw Flux (386328 flux in 998.07ms)
T8.0: IBM MFM (18/18 sectors) from Raw Flux (387309 flux in 998.08ms)
T8.1: IBM MFM (18/18 sectors) from Raw Flux (386710 flux in 998.05ms)
T9.0: IBM MFM (18/18 sectors) from Raw Flux (385292 flux in 998.07ms)
T9.1: IBM MFM (18/18 sectors) from Raw Flux (384690 flux in 998.07ms)
T10.0: IBM MFM (18/18 sectors) from Raw Flux (387190 flux in 998.07ms)
T10.1: IBM MFM (18/18 sectors) from Raw Flux (380896 flux in 998.05ms)
T11.0: IBM MFM (18/18 sectors) from Raw Flux (380278 flux in 998.07ms)
T11.1: IBM MFM (18/18 sectors) from Raw Flux (382750 flux in 998.04ms)
T12.0: IBM MFM (18/18 sectors) from Raw Flux (381716 flux in 998.06ms)
T12.1: IBM MFM (18/18 sectors) from Raw Flux (380599 flux in 998.05ms)
T13.0: IBM MFM (18/18 sectors) from Raw Flux (389386 flux in 998.09ms)
T13.1: IBM MFM (18/18 sectors) from Raw Flux (386876 flux in 998.04ms)
T14.0: IBM MFM (18/18 sectors) from Raw Flux (387989 flux in 998.08ms)
T14.1: IBM MFM (18/18 sectors) from Raw Flux (386080 flux in 998.09ms)
T15.0: IBM MFM (18/18 sectors) from Raw Flux (386706 flux in 998.08ms)
T15.1: IBM MFM (18/18 sectors) from Raw Flux (390364 flux in 998.07ms)
T16.0: IBM MFM (18/18 sectors) from Raw Flux (392447 flux in 998.07ms)
T16.1: IBM MFM (18/18 sectors) from Raw Flux (394498 flux in 998.05ms)
T17.0: IBM MFM (18/18 sectors) from Raw Flux (394382 flux in 998.08ms)
T17.1: IBM MFM (18/18 sectors) from Raw Flux (394316 flux in 998.06ms)
T18.0: IBM MFM (18/18 sectors) from Raw Flux (389336 flux in 998.08ms)
T18.1: IBM MFM (18/18 sectors) from Raw Flux (385719 flux in 998.07ms)
T19.0: IBM MFM (18/18 sectors) from Raw Flux (389536 flux in 998.08ms)
T19.1: IBM MFM (18/18 sectors) from Raw Flux (382036 flux in 998.07ms)
T20.0: IBM MFM (18/18 sectors) from Raw Flux (383723 flux in 998.07ms)
T20.1: IBM MFM (18/18 sectors) from Raw Flux (382264 flux in 998.08ms)
T21.0: IBM MFM (18/18 sectors) from Raw Flux (383122 flux in 998.09ms)
T21.1: IBM MFM (18/18 sectors) from Raw Flux (383795 flux in 998.09ms)
T22.0: IBM MFM (18/18 sectors) from Raw Flux (385161 flux in 998.10ms)
T22.1: IBM MFM (18/18 sectors) from Raw Flux (384113 flux in 998.06ms)
T23.0: IBM MFM (18/18 sectors) from Raw Flux (383202 flux in 998.06ms)
T23.1: IBM MFM (18/18 sectors) from Raw Flux (386633 flux in 998.06ms)
T24.0: IBM MFM (18/18 sectors) from Raw Flux (380251 flux in 998.09ms)
T24.1: IBM MFM (18/18 sectors) from Raw Flux (384956 flux in 998.10ms)
T25.0: IBM MFM (18/18 sectors) from Raw Flux (381862 flux in 998.06ms)
T25.1: IBM MFM (18/18 sectors) from Raw Flux (383042 flux in 998.04ms)
T26.0: IBM MFM (18/18 sectors) from Raw Flux (385109 flux in 998.06ms)
T26.1: IBM MFM (18/18 sectors) from Raw Flux (385736 flux in 998.03ms)
T27.0: IBM MFM (18/18 sectors) from Raw Flux (386260 flux in 998.08ms)
T27.1: IBM MFM (18/18 sectors) from Raw Flux (387422 flux in 998.05ms)
T28.0: IBM MFM (18/18 sectors) from Raw Flux (386420 flux in 998.08ms)
T28.1: IBM MFM (18/18 sectors) from Raw Flux (382143 flux in 998.08ms)
T29.0: IBM MFM (18/18 sectors) from Raw Flux (383100 flux in 998.07ms)
T29.1: IBM MFM (18/18 sectors) from Raw Flux (385334 flux in 998.06ms)
T30.0: IBM MFM (18/18 sectors) from Raw Flux (387172 flux in 998.07ms)
T30.1: IBM MFM (18/18 sectors) from Raw Flux (385051 flux in 998.05ms)
[... 124 lines cut ...]
1. 2: ................................................................................
1. 3: ................................................................................
1. 4: ................................................................................
1. 5: ................................................................................
1. 6: ................................................................................
1. 7: ................................................................................
1. 8: ................................................................................
1. 9: ................................................................................
1.10: ................................................................................
1.11: ................................................................................
1.12: ................................................................................
1.13: ................................................................................
1.14: ................................................................................
1.15: ................................................................................
1.16: ................................................................................
1.17: ................................................................................
Found 2880 sectors of 2880 (100%)
```

Greaseweazle reports no errors. Now opening the generated disk image in HxC and looking at the floppy analyzer shows this result:
[image]

As can be seen, track 79.1 sector 3 reports a CRC error and looking at the data it is apparent there was a read error (it should be filled with empty F6 bytes). Redumping it later with a different drive gave me an error free image. I would assume that the IBM format verification should be able to detect these kinds of errors? It certainly does detect some errors, but I have had a few images where Greaseweazle reported no errors but looking at the image data and opening the image in HxC reveals that there were one or more erroneous sectors. It is not limited to specific tracks or sectors either; in another dump, track 75.1 sector 1 had a CRC error according to HxC but Greaseweazle reported no errors. I have also had dumps where Greaseweazle reported one unreadable sector, but opening the image in HxC shows that there were actually several sectors with CRC errors.

What kind of verification is Greaseweazle actually performing when you specify an IBM format during dumping?

keirf, 2023-09-23:
Greaseweazle is converting the track to IBM sectors, and validating all CRCs. The situation here is that Greaseweazle's clock recovery is better than HxC's, and it can recover more data.

You can convert your `NAME_00.0.raw` Kryoflux-stream dump to an IMG later using Greaseweazle:
```
gw convert --format=ibm.1440 NAME_00.0.raw NAME.img
```

Take a look at the resulting `NAME.img` file using a hex dumper/editor and you should be able to confirm that for example track 79.1 sector 3 is as expected.

Litude, 2023-09-23:
I converted the raw image that HxC considered faulty with gw and got an IMG image that matched the other "better" dump byte-for-byte just like you explained! Had no idea HxC and gw did the conversion differently; will use only gw in the future. Thank you!

closed 2023-09-23 by Litude