keirf/greaseweazle #363: Trouble writing early PC98 games to disk

sdsnatcher73, 2023-10-06:
So I am trying to write these early PC98 games back to disk but after track 40 writes fail but with some retries it gets through but after 50 it alwyas fails. Not sure if I am doing something wrong but if so I can't figure out what. I've added a D88 image of the boot disk of Ys 2. Oddly the PC98 boots the written disk and is able to play through the demo and then asks for Disk A. 

HxC Track analyzer says the disks are 300RPM but have a 500kB datarate (which is odd, or is it not?). Any help would be appreciated. I am using MD2D media (MDHD media don't seem to work)..

[Ys II (Program disk).zip](https://github.com/keirf/greaseweazle/files/12832392/Ys.II.Program.disk.zip)

sdsnatcher73, 2023-10-06:
Hmm I seem to have trouble writing HD disks anyhow…

keirf, 2023-10-06:
HD disks? HD 3.5 inch drive I assume? Could be something like the density sensor switch is sticky?

sdsnatcher73, 2023-10-06:
No 5.25”

keirf, 2023-10-06:
I can confirm your attached D88 image writes fine for me to a HD 3.5-inch disk. Your problem is almost certainly density selection at the drive. 5.25-inch drives can't auto-detect disk density and need to be informed via pin 2. The default voltage at pin 2 is HIGH, so it makes sense to try setting it LOW instead and see what happens!

sdsnatcher73, 2023-10-06:
Tried it with the setting but it would not work. I switched to another drive and now it is working (pin2 set back to high)

case closed ;)

sdsnatcher73, 2023-10-06:
thanks for thinking along, should have tried if stuffed that worked before still worked before thinking something else was going on...

closed 2023-10-06 by sdsnatcher73