keirf/greaseweazle #364: After an erase I can still read the sectors of the disk

sdsnatcher73, 2023-10-08:
I am using a YD-380 at the moment as my main 5.25" HD drive. I am testing some other drives I have lying around and 1 drive was writing but nod reading (so I though, does not really matter). What is happening now and what seems odd. I wrote a disk on my YD-380, I run an erase on the disk and then I try to read the disk and all the content is still there... That defeats what erase should be doing, I'd imagined all magnetic flux would be gone...

d235j, 2023-10-08:
This can happen when e.g. attempting to erase an HD disk with a DD drive, or an HD drive set to DD settings, due to the higher magnetic coercivity of the HD disk; or erasing a disk which was written on a wide-head DD drive on a narrow-head HD drive, as the narrow head can't erase the entire track.

The only sure way to completely eliminate all magnetic flux from a disk is with a bulk eraser.

sdsnatcher73, 2023-10-08:
I was using an HD drive with HD media. Writing HD image works fine, writing DD images as well but I do have a DD drive for that. What is it the erase process sends to the drive? Could it be this drive is ignoring the commands?

keirf, 2023-10-08:
Greaseweazle tells the drive to write but doesn't send any flux transitions at all. This should result in random garbage transitions being read back.

Another thing to try is the `--hfreq` option to gw erase. This instead erases with a high frequency of flux transitions. It may work better in your drive. It usually reads back as no flux transitions whatsoever.

sdsnatcher73, 2023-10-08:
Okay using --hfreq works on the HD drive. Regular erase works on the DD drive but using --hfreq results in sectors still being readable on that drive. --hfreq is not available when writing with --pre-erase. Could this be added? Maybe it is possible to set an erase strategy per drive somehow inside the greaseweazle?

keirf, 2023-10-08:
You shouldn't actually need to erase or pre-erase at all. Do you see problems with these drives if you don't?

sdsnatcher73, 2023-10-14:
No not really seeing any issues when just writing. But I know some game disks have incomplete tracks (so let's say only a few sectors where more would fit) or even empty tracks. It's the reason I reckon --erase-empty exists. So it would still make sense to allow setting an erase strategy for --drive a and --drive b (or --drive 0 and --drive 1 when using shugart).

An example of such disk is the User DIsk of Ys, currently gw host tools have issues with the D88 although I believe it is valid (SAMdisk accepts it, but I have not tried actually writing it as my old PC is not hooked up since I got the gw ;)
[Ys (User disk).zip](https://github.com/keirf/greaseweazle/files/12906227/Ys.User.disk.zip)

keirf, 2023-10-14:
Thanks for the offending D88 image file: This is now fixed in the latest build and I will shortly make a new release of the Greaseweazle tools.

If you are really wanting to be sure that a disk is erased, a simple but effective method is to wipe across the disk jacket systematically with a strong magnet (eg. neodymium as found in old HDDs). It works! And as a bonus, it's quicker than running `gw erase`.

To prove this to yourself, try it, then `gw read name.scp`. Import the SCP image file into HxC and then use HxC disk visualiser to observe a sea of read (random/illegal flux).

closed 2023-10-14 by keirf