keirf/greaseweazle #365: can't write disk

Telev78, 2023-10-12:
Hello,

I recently got a Panasonic JU-455-5 5.25" 360 KB disk drive.
In my 386 with this drive, I can read (disk created on another drive) and write well.
I am testing now with GW F1 "bluepill", I am able to read the disk, but I cannot write to any disk...
I tested with 3 different cables (2 homemade & 1 commercial), with the same result.
Could there be an incompatibility?

Regards

keirf, 2023-10-12:
The F1 device is known good, tested with another drive?

Telev78, 2023-10-12:
yes tested good with another 3.5 drive and 5.25 360k drive.
tested with 2 different F1

Telev78, 2023-10-12:
tested now with gw V4 and the disk work fine.
so this drive don't work with the F1

keirf, 2023-10-12:
Ohhh it's probably because the drive has strong pull-ups and the signal lines can't be driven hard enough by the F1 microcontroller. The V4 board has separate driver chips.

closed 2023-10-12 by keirf