keirf/greaseweazle #366: Cannot read or write: GetFluxStatus: No Index

mhoeben, 2023-10-12:
Installed all software without any problems on Pop OS 22.04 LTS, attached an Amiga drive with a straight cable, attached power cable (both correctly oriented), drive has jumper on D0 (D1 and D2 are not bridged), successfully upgraded as directed, and `gw info` says:
```
Host Tools: 1.15.1
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4
  MCU:      AT32F403A, 216MHz, 96kB SRAM
  Firmware: 1.4
  Serial:   GWB0B563A85976C01007B41705
  USB Rate: Full Speed (12 Mbit/s)
```
Board shows a non-blinking red status LED. `gw read --drive 0 image.adf` says:
```
Reading c=0-79:h=0-1 revs=1.1
Format amiga.amigados
Command Failed: GetFluxStatus: No Index
```
While trying to read, another LED comes on as green. Drive does not show any sign of life.

Tried `--drive=a` and `b`, and `1` and `2` for good measure. Nothing, always the same issue. Tried different drive, other flat cables, etc...

Read all documentation, Readit posts, forum posts, etc..., but non seems to cover this particular issue that driver `--drive=?` settings don't resolve the issue.

_Please advice on steps how to resolve this issue._

keirf, 2023-10-12:
Some original full height Amiga drives need 12v power. Greaseweazle supplies only 5v power. You will need a separate power supply for the drive. Eg molex PSU and molex to floppy tail.

mhoeben, 2023-10-13:
Ok, understood. Will power the drive from the Amiga and retry. Thanks.

closed 2023-10-13 by mhoeben