keirf/greaseweazle #368: Amiga SCP writing and converting don't works on version 1.16

merlinkv, 2023-10-18:
Hello.

Writing SCP Amiga files to real DD disks or converting ipf's to scp's don't work on Host Tools version 1.16, however on version 1.15.1 everything works fine.

Any idea?

keirf, 2023-10-18:
What happens?

keirf, 2023-10-18:
I tried a quick conversion of an IPF to SCP and the result looked okay. So please give a specific IPF-to-SCP conversion which has regressed with 1.16: 1. Name the offending IF (eg SPS IPF number); 2. Your exact command line; 3. Expected vs actual result on 1.16 gw tools.

merlinkv, 2023-10-18:
Hi Keirf.

OK I will do it this evening after job.

Thanks & Best Regards

merlinkv, 2023-10-18:
Hi keirf.

I make a clean install of 1.15.1 and 1.16 on separate directories with sps dll x64 on Host Tools root dirs.

Now all apear to be working fine with both versions, but I have found big differences on the size of converted files.

Command lines:

**v 1.16**

gw.exe convert --tracks=c=0-83:h=0-1 "F:\Amiga_Preservation\Games_Test2\Elf_Disk2.ipf" "F:\Amiga_Preservation\Games_Test2\Elf_D2_sps473_GW_1_15_1.scp::disktype=amiga"

Size of file: 12.987.762 bytes

**v 1.15.1**

gw.exe convert --tracks=c=0-83:h=0-1 "F:\Amiga_Preservation\Games_Test2\Elf_Disk2.ipf" "F:\Amiga_Preservation\Games_Test2\Elf_D2_sps473_GW_1_15_1.scp::disktype=amiga"

Size of file: 25.973.438 bytes

The size of all files converted with 1.15.1 are the double (+/-) in size compared with the files converted with v 1.16, tested with Elf_sps473, Awesome_sps431, Hook_sps887 and Transarctica_sps744.

Any idea? I'm doing something wrong?

Thank you and sorry for bothering you with my troubles.

Best regards.

keirf, 2023-10-18:
The difference is that v1.16 emits only one revolution per index-aligned track, whereas v1.15.1 always emits two revolutions.

Now I think about it, this is probably broken when generating SCP, as SCP format declares in its header a number of revolutions that applies for every track in the image, and gw sets this as min(revolutions across all tracks). So the tracks that need two revolutions (becase there is data across the index) will get clipped to just one if there is even just one index-aligned track.

merlinkv, 2023-10-18:
SCP files of Hook Disk1 made with GW and readed with SuperCard Pro software..

v 1.16

[image]

v 1.15.1

[image]

Index is the "standard" mode. Splice is the mode used for preservation, from 1 to 5 revolutions.

keirf, 2023-10-19:
Okay I reverted to old 1.15.1 behaviour by default, which can be overridden by specifying number of revolutions explicitly for example: `gw convert name.ipf name.scp::revs=1`

closed 2023-10-19 by keirf